ros/meta-ros #1039: OSError: Environment variable 'AMENT_PREFIX_PATH' is not set or empty on  target bord

alexleel, 2023-03-17:
**Describe the bug**
I have encountered the error as following 
root@intel-x86-64:~# ros2 pkg list
Traceback (most recent call last):
  File "/usr/bin/ros2", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.10/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/usr/lib/python3.10/site-packages/ros2pkg/command/pkg.py", line 37, in main
    return extension.main(args=args)
  File "/usr/lib/python3.10/site-packages/ros2pkg/verb/list.py", line 23, in main
    for pkg_name in sorted(get_package_names()):
  File "/usr/lib/python3.10/site-packages/ros2pkg/api/__init__.py", line 29, in get_package_names
    return get_packages_with_prefixes().keys()
  File "/usr/lib/python3.10/site-packages/ament_index_python/packages.py", line 34, in get_packages_with_prefixes
    return get_resources('packages')
  File "/usr/lib/python3.10/site-packages/ament_index_python/resources.py", line 63, in get_resources
    for path in get_search_paths():
  File "/usr/lib/python3.10/site-packages/ament_index_python/search_paths.py", line 29, in get_search_paths
    raise EnvironmentError(
OSError: Environment variable 'AMENT_PREFIX_PATH' is not set or empty

**To Reproduce**
1. I build the image with bitbake 'ros-image-core' successfully, 
2. I have writed an image file to a storage device on a target board.
3. login target board and run some ros2 CLI such as ''ros2 pkg list'/ 'ros2 run xx'
4. error occurs :

The "Build Configuration:" section shown by bitbake at the beginning of the build is very good start.
1. conf/bblayers.conf
  ../layers/meta-ros/meta-ros2-foxy \
  ..layers/meta-ros/meta-ros2 \
. ./meta-ros/meta-ros-common \
2 conf/local.conf:
IMAGE_INSTALL:append = " \
                     sros2 \
                     packagegroup-ros2-demos ros2service  examples-rclcpp-minimal-service \
                     quality-of-service-demo-cpp \
                     "
**Expected behavior**
Can run CLI without error.

kulkarni-raunak, 2023-10-20:
Any update on this?

Jinsun-Lee, 2023-11-20:
`source /opt/ros/humble/share/ros_environment/local_setup.bash`

This worked for me.

robwoolley, 2024-11-21:
Closing as resolved.

closed 2024-11-21 by robwoolley