ros/meta-ros #1040: librclcpp.so: undefined reference to `__atomic_exchange_1

Jpgig, 2023-03-30:
Issue :
I am trying to integrate "ROS 2 Humble in Microchip PolarFire® SoC FPGA Icicle Kit with Yocto".
When I run bitbake rclcpp-components I am getting the below error:
librclcpp. so: undefined reference to `__atomic_exchange_1

**To Reproduce**
Followed steps noted in this link:
https://news.accelerationrobotics.com/ros-2-humble-in-microchip-polarfire-icicle-kit-with-yocto/

**Expected behavior**
Should generate a Linux image without any error.

robwoolley, 2024-11-21:
Could you please try to reproduce this with the latest updates?

robwoolley, 2025-01-21:
Thanks for reporting this.  This has been fixed in kirkstone with https://github.com/ros/meta-ros/commit/e4af3cf867c3fe30ac233f08f1aead777a975a47

I also provided feedback in the issue you filed upstream as I believe they should be adding libatomic in rclcpp itself: https://github.com/ros2/rclcpp/issues/2150