ros/meta-ros #1048: Request for Humble on Yocto LTS Kirkstone

Bentis, 2023-09-08:
Now that both Foxy and Yocto Honister has reached End-of-Life;
is there any community effort underway to get Humble working with the current Yocto LTS, Kirkstone?

Looking to start a new project using ROS2 on Yocto, and a bit worried if this approach is making us to use old unsupported versions.

Looking at the unfinished #1031, and their [fork](https://github.com/RethinkRobotics-opensource/meta-ros/tree/experimental/humble-mickledore) it seems @RethinkMatze might be close?

RethinkMatze, 2023-09-08:
Hi @Bentis ,

since the development here seemed to unfortunately have ceased, we have forked meta-ros and although it is far from thoroughly tested, it suites our needs.  There is a [branch for humble/kirkstone](https://github.com/RethinkRobotics-opensource/meta-ros/tree/experimental/humble-kirkstone) and one for [humble/mickledore](https://github.com/RethinkRobotics-opensource/meta-ros/tree/experimental/humble-mickledore). The later one, I am currently working on and I will probably be able to update to the latest humble / noetic. 

If you check it out and give feedback would be just great.

Regards,

   Matthias

rcwoolley, 2023-11-13:
Hi @Bentis,  I just pushed updates to the repository.

The main README shows which combination of ROS and Yocto are supported.  For a preview of kirkstone check out the kirkstone-next branch.  I have built ros-image-core.  I encourage you to file issues for any problems you find.

Regards,
Rob

closed 2023-11-13 by robwoolley