ros/meta-ros #1056: Build error with ros-image-world

bchoineubility, 2023-11-15:
Hi,

I try to build kirkstone-humble as [instruction.](https://github.com/ros/meta-ros#getting-started) by the way, following error is occurred

```
bchoi@nvidia-build:~/work/raspberrypi/build$ bitbake ros-image-world
Loading cache: 100% |#########################################################################################################################################################################################################################################| Time: 0:00:00
Loaded 569 entries from dependency cache.
WARNING: /home/bchoi/work/raspberrypi/meta-ros/meta-ros2-humble/generated-recipes/rmf-ros2/rmf-websocket_2.1.7-1.bb: ROS_EXEC_DEPENDS:remove += is not a recommended operator combination, please replace it.                                                 | ETA:  0:00:21
WARNING: /home/bchoi/work/raspberrypi/meta-ros/meta-ros2-humble/generated-recipes/ur-description/ur-description_2.1.1-1.bb: QA Issue: ur-description: LICENSE value "BSD-3-Clause & Universal-Robots-A-S’-Terms-and-Conditions-for-Use-of-Graphical-Documentation" has an invalid separator "Universal-Robots-A-S’-Terms-and-Conditions-for-Use-of-Graphical-Documentation" that is not in the valid list of separators (&|() ) [license-format]
WARNING: /home/bchoi/work/raspberrypi/openembedded-core/meta/recipes-kernel/lttng/babeltrace_1.5.11.bb: distutils-common-base.bbclass is deprecated, please use setuptools3-base.bbclass instead###############################                               | ETA:  0:00:03
Parsing recipes: 100% |#######################################################################################################################################################################################################################################| Time: 0:00:28
Parsing of 3717 .bb files complete (402 cached, 3315 parsed). 5378 targets, 158 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'ROS_UNRESOLVED_DEP-python3-distro' (but /home/bchoi/work/raspberrypi/meta-ros/meta-ros2-humble/generated-recipes/flir-camera-driver/spinnaker-camera-driver_2.0.7-1.bb DEPENDS on or otherwise requires it)
NOTE: Runtime target 'spinnaker-camera-driver' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['spinnaker-camera-driver', 'ROS_UNRESOLVED_DEP-python3-distro']
NOTE: Runtime target 'packagegroup-ros-world-humble' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-ros-world-humble', 'spinnaker-camera-driver', 'ROS_UNRESOLVED_DEP-python3-distro']
NOTE: Runtime target 'packagegroup-ros-world' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-ros-world', 'packagegroup-ros-world-humble', 'spinnaker-camera-driver', 'ROS_UNRESOLVED_DEP-python3-distro']
ERROR: Required build target 'ros-image-world' has no buildable providers.
Missing or unbuildable dependency chain was: ['ros-image-world', 'packagegroup-ros-world', 'packagegroup-ros-world-humble', 'spinnaker-camera-driver', 'ROS_UNRESOLVED_DEP-python3-distro']

Summary: There were 3 WARNING messages.
Summary: There were 2 ERROR messages, returning a non-zero exit code.
bchoi@nvidia-build:~/work/raspberrypi/build$
```

Do I need any modification for build as [instruction](https://github.com/ros/meta-ros#getting-started) ?

BR,
Mark

robwoolley, 2023-11-16:
Hi Mark,

Thanks for trying the latest changes.  My initial testing was just focused on ros-image-core for qemux86-64 so that we could get all the new Yocto releases and ROS distributions up-to-date.  The next focus is to get ros-image-world to start building.

I noticed that are building for raspberrypi.  I followed the instructions and set MACHINE = "raspberrypi4-64" in my conf/local.conf.  I was able to successfully build ros-image-core.

Regards,
Rob

bchoineubility, 2023-11-16:
@robwoolley  Thanks for reply and sharing time to see my issue.

I will try to build ros-core-image with your instruction too.

I am trying to build ros and it is challenge for me. it's a pleasant for me. 

I am looking forward to update for ros-image-world also.

 BR,
Mark

closed 2023-11-16 by bchoineubility