ros/meta-ros #1058: ModuleNotFoundError: No module named 'generate_parameter_library_py'

bchoineubility, 2023-11-16:
Hi, 

I try to build kirkstone-humble with meta-tegra.  by the way, I faced configure error at first.
```
| CMake Error at /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/moveit-core/2.5.5-1-r0/recipe-sysroot/usr/share/generate_parameter_library/cmake/generate_parameter_library.cmake:33 (message):
|   generate_parameter_library_cpp() variable
|   'generate_parameter_library_cpp_BIN' must not be empty
| Call Stack (most recent call first):
|   online_signal_smoothing/CMakeLists.txt:31 (generate_parameter_library)
```

I copied generate_parameter_lib_cpp python script to sysroot manually and configure error would be fixed at the time.
But, the new compile error is occurred : 

```
| Traceback (most recent call last):
|   File "/home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/moveit-core/2.5.5-1-r0/recipe-sysroot/usr/bin/generate_parameter_library_cpp", line 5, in <module>
|     from generate_parameter_library_py.generate_cpp_header import main
| ModuleNotFoundError: No module named 'generate_parameter_library_py'

```

I don't understand why moveit-core recipe could not find dependency of generate_parameter_library_py  even though it added to DEPEND in the its recipe.

BR,
Mark

robwoolley, 2023-11-16:
Hi Mark,

The challenge was that generate_parameter_library_py is a tool that needs to run, so you need to add generate-parameter-library-py-native instead.

See my pull request #1060 for the fix.  If it works for you, I'd be happy to merge it.

Regards,
Rob

bchoineubility, 2023-11-16:
@robwoolley Many thanks. my issue is cleared with your solution.  I can see build error is gone.

by the way, I had created new bbappend file for moveit-core to my layer directory with the same name in the meta-ros2-humble/moveit and I had appended generate-parameter-library-py-native to ROS_BUILDTOOL_DEPENDS as you did, but, issue was remain. 

I have created bbappend file with the same name in the ros layer to my layer if I encounter build error, then I could resolve my issue. 

I am afraid my solution would not be correct. 

Can you advise for me?

BR,
Mark

closed 2023-11-16 by bchoineubility