ros/meta-ros #1061: Need to replace PNBLACKLIST with SKIP_RECIPE

moto-timo, 2023-11-16:
**Describe the bug**
A very large number of recipes are skipped via PNBLACKLIST, but this variable has been renamed SKIP_RECIPE.

**To Reproduce**
Layer Index Update [Nov. 16, 2023, 6:23 p.m.] (https://layers.openembedded.org/layerindex/updates/38472/)

Layer Index Update [Nov. 13, 2023, 7:30 p.m.](https://layers.openembedded.org/layerindex/updates/38449/)

These should be somewhat trivial to fix by using [scripts/contrib/convert-variable-renames.py](https://git.yoctoproject.org/poky/tree/scripts/contrib/convert-variable-renames.py)

**Expected behavior**
Any branch after the PNBLACKLIST change (kirkstone 4.0+) should not have the old variable usage.

grsandeep85, 2023-11-16:
Mark Hatle has already fixed this issue. @robwoolley as mentioned in meeting can you cherry pick these patches https://github.com/Xilinx/meta-ros/commit/fb6cc73c4013a41cbbeb763213efe62e0d2c08f3

moto-timo, 2023-11-16:
#1062 #1063 #1064 #1065

robwoolley, 2023-11-17:
Thanks, guys.  I reviewed the PRs and merged them.

closed 2023-11-17 by robwoolley