ros/meta-ros #1067: task fail zstd-vendor_0.15.8-1.bb:do_patch

nirhenn, 2023-11-17:
I am trying to build Kirkstone build with meta-ros2 humble layer and the build fails with the following error:

ERROR: zstd-vendor-0.15.8-1-r0 do_patch: Applying patch '0001-CMakeLists.txt-prevent-building-zstd-with-ExternalPr.patch' on target directory '/home/osboxes/yocto/sources/build/tmp/BUILD-poky-humble-kirkstone/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/zstd-vendor/0.15.8-1-r0/git'
CmdError('quilt --quiltrc /home/osboxes/yocto/sources/build/tmp/BUILD-poky-humble-kirkstone/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/zstd-vendor/0.15.8-1-r0/recipe-sysroot-native/etc/quiltrc push', 0, 'stdout: Applying patch 0001-CMakeLists.txt-prevent-building-zstd-with-ExternalPr.patch
patching file CMakeLists.txt
patching file cmake_minimum_required_2.8.12.patch
patching file no_internal_headers.patch
Hunk #1 FAILED at 1.
Not deleting file no_internal_headers.patch as content differs from patch
1 out of 1 hunk FAILED -- rejects in file no_internal_headers.patch
Patch 0001-CMakeLists.txt-prevent-building-zstd-with-ExternalPr.patch does not apply (enforce with -f)

stderr: ')
ERROR: Logfile of failure stored in: /home/osboxes/yocto/sources/build/tmp/BUILD-poky-humble-kirkstone/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/zstd-vendor/0.15.8-1-r0/temp/log.do_patch.10781
ERROR: Task (/home/osboxes/yocto/sources/ROS/meta-ros/meta-ros2-humble/generated-recipes/rosbag2/zstd-vendor_0.15.8-1.bb:do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6482 tasks of which 6450 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/osboxes/yocto/sources/ROS/meta-ros/meta-ros2-humble/generated-recipes/rosbag2/zstd-vendor_0.15.8-1.bb:do_patch
Summary: There was 1 ERROR message, returning a non-zero exit code.

Any tips on how to resolve it will be appreciated.

robwoolley, 2023-11-19:
Hi @nirhenn,

Thanks for reporting this.  I am still working on getting ros-image-world to build, but please keep filing these types of reports as it helps me prioritize the work.

I have pushed a commit to kirkstone-next that should help fix the problem.  Let me know if it fixes things for you: https://github.com/ros/meta-ros/commit/4f9e1da84dcd49694d80520df7c7b3fc2186ee42

Regards,
Rob

grsandeep85, 2023-11-22:
Hi @robwoolley,

Removing the patch will cause other packages to fail. I've rebased the patch in #1074 along with other fixes. Once it is applied to master we can cherry pick to kirkstone.

robwoolley, 2023-11-27:
Thanks @grsandeep85, I just pushed your fixes to kirkstone-next.

closed 2023-11-27 by robwoolley