ros/meta-ros #1071: Add GitHub Actions to meta-ros

robwoolley, 2023-11-17:
Adding GitHub Actions to meta-ros would be useful for automating checks and builds.

- @ajlennon Yocto Layer CI Build and Test with GitHub Action by Alex Lennon (https://youtu.be/TsAcxd_acJI?si=HqRCqZ9M-f0T-Y-3) his work is here: https://github.com/DynamicDevices/meta-mono/blob/master/.github/workflows/CI_github.yml
- @razr did some work integrating meta-ros CI with GitHub Actions (https://github.com/razr/meta-ros/blob/hardknott-ur5-gh-action/.github/workflows/two-jobs.yaml)
- @Allegra42 is also giving a talk at the Yocto Summit 2023 entitled "Building a Yocto CI Pipeline with KAS, GitHub Actions and AWS - What could possibly go wrong?" (https://summit.yoctoproject.org/yocto-project-summit-2023-11/talk/RNHAPH/)

robwoolley, 2023-11-18:
@moto-timo Also pointed me to the work done for CROPS: https://github.com/crops/poky-container/blob/master/.github/workflows/build-test-deploy.yml

rcwoolley, 2023-12-12:
See [ROS 2 Tooling Working Group](https://github.com/ros-tooling/community) for existing ROS tooling:

- [Design: GitHub Action Release Process](https://github.com/ros-tooling/design/blob/main/docs/action-release.md)
- [action-ros-ci](https://github.com/ros-tooling/action-ros-ci)
- [ROS 2 GitHub Action CI Template](https://github.com/ros-tooling/action-ros-ci-template)
- https://github.com/ros-tooling/setup-ros
- https://github.com/ros-tooling/setup-ros-docker/tree/master
- https://github.com/osrf/docker_images