ros/meta-ros #1072: ImportError: cannot import name 'QtCore' from 'PyQt5' (unknown location)

bchoineubility, 2023-11-20:
Hi,

I am trying to build qt-gui-cpp.  by the way, python3-pyqt5-native would not support qt core module.

```
| FAILED: sip/qt_gui_cpp_sip/Makefile /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/build/sip/qt_gui_cpp_sip/Makefile
| cd /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/git/src/qt_gui_cpp_sip && /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3 /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/share/python_qt_binding/cmake/sip_configure.py /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/build/sip/qt_gui_cpp_sip qt_gui_cpp.sip /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/build/src/qt_gui_cpp_sip "/home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/git/src/qt_gui_cpp_sip/../../include /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/include/pluginlib /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/include/ament_index_cpp /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/include/class_loader /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/include /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/include/rcpputils /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/include/rcutils /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/include/python3.10" "/home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/lib/libament_index_cpp.so /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/lib/libconsole_bridge.so.1.0 dl /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/lib/librcutils.so /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/lib/librcpputils.so /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/lib/libclass_loader.so /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/lib/libtinyxml2.so.9.0.0 /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/lib/libtinyxml2.so.9.0.0 /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/lib/libpython3.10.so /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/build/libqt_gui_cpp.so" "lib" ""
| /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/share/python_qt_binding/cmake/sip_configure.py:2: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
|   from distutils.spawn import find_executable
| Traceback (most recent call last):
|   File "/home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/qt-gui-cpp/2.2.2-1-r0/recipe-sysroot/usr/share/python_qt_binding/cmake/sip_configure.py", line 9, in <module>
|     from PyQt5 import QtCore
| ImportError: cannot import name 'QtCore' from 'PyQt5' (unknown location)
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/bchoi/work/official_ros_humble_build/sources/meta-ros/meta-ros2-humble/generated-recipes/qt-gui-core/qt-gui-cpp_2.2.2-1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2479 tasks of which 2429 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/bchoi/work/official_ros_humble_build/sources/meta-ros/meta-ros2-humble/generated-recipes/qt-gui-core/qt-gui-cpp_2.2.2-1.bb:do_compile
Summary: There was 1 WARNING message.
Summary: There was 1 ERROR message, returning a non-zero exit code.
bchoi@nvidia-build:~/work/official_ros_humble_build/bchoi-build$
```

I tried to add python3-pyqt5 dependency to qt-gui-cpp recipe. but, it doesn't work now. 

I hope to know avoid this issue.

BR,
Mark

robwoolley, 2023-11-20:
Thanks for reporting this.  I added support for python3-pyqt5-native, 3 years ago but haven't submitted it for merging yet (see #674 )

I intend to submit it to meta-qt5 as part of getting ros-image-world to build.

bchoineubility, 2023-11-24:
Hi @robwoolley 

I am trying to build recipe python3-pyqt5-native-5.15.1.bb with your [efforts](https://github.com/ros/meta-ros/issues/674) for this issue. 

I changed python-pyqt5.inc as followings, because version do not match with kirkstone.

```
SUMMARY = "Python Qt5 Bindings"
AUTHOR = "Phil Thomson @ riverbank.co.uk"
HOMEPAGE = "http://riverbankcomputing.co.uk"
SECTION = "devel/python"
LICENSE = "GPLv3"
LIC_FILES_CHKSUM = "\
    file://LICENSE;md5=d32239bcb673463ab874e80d47fae504 \
"
SRC_URI = "\
    https://www.riverbankcomputing.com/static/Downloads/PyQt5/${PV}/PyQt5_gpl-${PV}.tar.gz \
"
SRC_URI[md5sum] = "67508b652098d2e05c4c2b5baeb170cc"
SRC_URI[sha256sum] = "3718ce847d824090fd5f95ff3f13847ee75c2507368d4cbaeb48338f506e59bf"
S = "${WORKDIR}/PyQt5_gpl-${PV}"

inherit qmake5
DEPENDS = "qtbase qtdeclarative qtquickcontrols2"

export BUILD_SYS
export HOST_SYS
export STAGING_INCDIR
export STAGING_LIBDIR

PARALLEL_MAKEINST = ""

DISABLED_FEATURES = "PyQt_Desktop_OpenGL PyQt_Accessibility PyQt_SessionManager ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', '', 'PyQt_OpenGL', d)}"

PYQT_MODULES = "QtCore QtGui QtNetwork QtXml QtNetwork QtQml ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'QtQuick QtWidgets QtQuickWidgets', '', d)}"

do_configure_prepend() {
    cd ${S}
    echo "py_platform = linux" > pyqt.cfg
    echo "py_inc_dir = %(sysroot)/$includedir/python%(py_major).%(py_minor)${PYTHON_ABI}" >> pyqt.cfg
    echo "py_pylib_dir = %(sysroot)/${libdir}/python%(py_major).%(py_minor)" >> pyqt.cfg
    echo "py_pylib_lib = python$%(py_major).%(py_minor)" >> pyqt.cfg
    echo "pyqt_module_dir = ${D}/${libdir}/python%(py_major).%(py_minor)/site-packages" >> pyqt.cfg
    echo "pyqt_bin_dir = ${D}/${bindir}" >> pyqt.cfg
    echo "pyqt_sip_dir = ${D}/${datadir}/sip/PyQt5" >> pyqt.cfg
    echo "pyuic_interpreter = ${D}/${bindir}/python%(py_major).%(py_minor)" >> pyqt.cfg
    echo "pyqt_disabled_features = ${DISABLED_FEATURES}" >> pyqt.cfg
    echo "qt_shared = True" >> pyqt.cfg
    QT_VERSION=`${OE_QMAKE_QMAKE} -query QT_VERSION`
    echo "[Qt $QT_VERSION]" >> pyqt.cfg
    echo "pyqt_modules = ${PYQT_MODULES}" >> pyqt.cfg
    echo yes | ${PYTHON} configure.py --verbose --qmake  ${STAGING_BINDIR_NATIVE}/${QT_DIR_NAME}/qmake --configuration pyqt.cfg $(if [ -n "${STAGING_HOST_DIR}" ]; then echo --sysroot ${STAGING_HOST_DIR}; fi)
}

do_compile() {
    cd ${S}
    oe_runmake
}

do_install() {
    cd ${S}
    oe_runmake MAKEFLAGS='-j 1' install
}

FILES_${PN} += "${libdir}/${PYTHON_DIR}/site-packages ${datadir}/sip/PyQt5/"

RDEPENDS_${PN} = "qtbase qtdeclarative qtquickcontrols2 qtquickcontrols2-mkspecs"

``` 

By the way, this also reference Qt gui also. 

```
 Reading /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.1-r1/PyQt5-5.15.1/QtCore/QtCore.pro
| Reading /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.1-r1/PyQt5-5.15.1/QtGui/QtGui.pro
| Project ERROR: Unknown module(s) in QT: gui
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/bchoi/work/official_ros_humble_build/sources/meta-neubility/yocto-land/recipes-python/pyqt5/python3-pyqt5-native_5.15.1.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1644 tasks of which 1636 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/bchoi/work/official_ros_humble_build/sources/meta-neubility/yocto-land/recipes-python/pyqt5/python3-pyqt5-native_5.15.1.bb:do_configure
```
before doing it, I remove --sysroot option because STAGING_HOST_DIR is empty.

BR,
Mark

bchoineubility, 2023-11-30:
I managed to succeed in building python3-pyqt5-native with kirkstone qt-5.15.7.

Here, I leave my recipe for it.

```
SUMMARY = "Python Qt5 Bindings"
AUTHOR = "Phil Thomson @ riverbank.co.uk"
HOMEPAGE = "https://www.riverbankcomputing.com/software/pyqt"
SECTION = "devel/python"
LICENSE = "GPL-3.0-only"
LIC_FILES_CHKSUM = "\
    file://LICENSE;md5=d32239bcb673463ab874e80d47fae504 \
"

inherit pypi

PYPI_PACKAGE = "PyQt5"

SRC_URI[md5sum] = "b60f895e70d31d774e6cd374efd17c65"
SRC_URI[sha256sum] = "d9a76b850246d08da9863189ecb98f6c2aa9b4d97a3e85e29330a264aed0f9a1"

S = "${WORKDIR}/PyQt5-${PV}"

inherit qmake5
inherit python3native python3-dir

DEPENDS = "qtbase qtdeclarative qtquickcontrols2"
DEPENDS += "sip3-native python3-native"

PARALLEL_MAKEINST = ""

DISABLED_FEATURES = "PyQt_Desktop_OpenGL PyQt_Accessibility PyQt_SessionManager ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', '', 'PyQt_OpenGL', d)}"
#PYQT_MODULES = "QtCore QtGui QtNetwork QtQml ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'QtQuick QtWidgets QtQuickWidgets', '', d)}"
#PYQT_MODULES = "QtNetwork QtQml"
#PYQT_MODULES = "QtCore QtGui QtNetwork QtXml QtNetwork QtQml"

# $includedir is recipes-sysroot-native/usr/include
# can not inject dependenies of qmake with configure.py.
do_configure:prepend() {
    cd ${S}
    echo "py_platform = linux" > pyqt.cfg
    echo "py_inc_dir = $includedir/python%(py_major).%(py_minor)${PYTHON_ABI}" >> pyqt.cfg
    echo "py_pylib_dir = ${libdir}/python%(py_major).%(py_minor)" >> pyqt.cfg
    echo "py_pylib_lib = python$%(py_major).%(py_minor)" >> pyqt.cfg
#    echo "pyqt_module_dir = ${D}/usr/lib/python%(py_major).%(py_minor)/site-packages" >> pyqt.cfg
#    echo "pyqt_bin_dir = ${D}/usr/bin" >> pyqt.cfg
#    echo "pyqt_sip_dir = ${D}/usr/share/sip/PyQt5" >> pyqt.cfg
    echo "pyqt_module_dir = ${libdir}/python%(py_major).%(py_minor)/site-packages" >> pyqt.cfg
    echo "pyqt_bin_dir = ${bindir}" >> pyqt.cfg
    echo "pyqt_sip_dir = ${datadir}/sip/PyQt5" >> pyqt.cfg
    echo "pyuic_interpreter = ${bindir}/python%(py_major).%(py_minor)" >> pyqt.cfg
    echo "pyqt_disabled_features = ${DISABLED_FEATURES}" >> pyqt.cfg
    echo "qt_shared = True" >> pyqt.cfg
    QT_VERSION=`${OE_QMAKE_QMAKE} -query QT_VERSION`
    echo "[Qt $QT_VERSION]" >> pyqt.cfg
#    echo "pyqt_modules = ${PYQT_MODULES}" >> pyqt.cfg
    echo yes | ${PYTHON} configure.py --verbose --qmake  ${STAGING_BINDIR_NATIVE}/${QT_DIR_NAME}/qmake --configuration pyqt.cfg --sysroot ${STAGING_DIR_NATIVE}
}

do_configure:append() {
  modules="QtDBus QtNetwork QtXml"
  for module in $modules
    do
    sed -i'' -r -e "/INCPATH/a\INCPATH       += -I../../recipe-sysroot-native/usr/include/$module" ${S}/$module/Makefile
  done
  sed -i'' -r -e "/INCPATH/a\INCPATH       += -I../../recipe-sysroot-native/usr/include/QtXml" ${S}/pyrcc/Makefile
  sed -i'' -r -e "/INCPATH/a\INCPATH       += -I../../recipe-sysroot-native/usr/include/QtXml" ${S}/pylupdate/Makefile
}

EXTRA_QMAKEVARS_POST = "QT=core"

do_compile() {
    cd ${S}
    oe_runmake
}

do_install:class-native() {
    cd ${S}
    oe_runmake MAKEFLAGS='-j 1' install INSTALL_ROOT=${D}
}

FILES:${PN} += "${libdir}/${PYTHON_DIR}/site-packages ${datadir}/sip/PyQt5/"

RDEPENDS:${PN} = "qtbase qtdeclarative qtquickcontrols2 qtquickcontrols2-mkspecs"
RDEPENDS:${PN} += "python3-core python3-sip3"

inherit native
```

Can you check this recipe is correct and let me know I am missing?

BR,
Mark

robwoolley, 2023-11-30:
Hi Mark,

Thanks for looking into this.  I took a look at it last week as well.  Many of the problems I encountered only appeared when I started trying to use it to build qt-gui-cpp.  I had to add a few host tools (like python3-pyqt-build-native) and heavy work on sip3 to make qt-gui-cpp build.

There are 2 things I think may be incorrect.  It looks like the intent was to specify the Qt modules explicitly in PYQT_MODULES.  Commenting this out and using EXTRA_QMAKEVARS_POST seems to be against the original intent of the maintainer.  I think removing ${D} from pyqt_module_dir, pyqt_bin_dir, pyqt_sip_dir may cause problems when trying to use these tools in other recipes.  From memory, the ${D} variable is the full absolute path under the "image" directory for -native packages.   I suspect this means that they are getting installed back into one of the recipes-sysroot directories and not getting packaged.

I am planning on pushing out a developer branch with my changes this week.  Would you be interested in collaborating to verify that you can reproduce a build of qt-gui-cpp with it?

Regards,
Rob

robwoolley, 2023-12-02:
I have started to push my changes upstream starting with meta-openembedded.  You can see my cover letter and 2 patches to sip3 here:

- https://lists.openembedded.org/g/openembedded-devel/message/107194
- https://lists.openembedded.org/g/openembedded-devel/message/107193
- https://lists.openembedded.org/g/openembedded-devel/message/107195

After it is accepted, I can push my pyqt5 changes to meta-qt5.

robwoolley, 2023-12-02:
I have made a pull request to meta-qt5 with the changes I made to python3-pyqt5 to add -native: https://github.com/meta-qt5/meta-qt5/pull/546

robwoolley, 2023-12-02:
I have created a feature branch with the qt-gui-cpp changes: https://github.com/robwoolley/meta-ros/commits/qt-gui-cpp

I am waiting for the sip3 changes to meta-openembedded and the python3-pyqt5 changes to meta-qt5 to be accepted first before applying the changes to meta-ros.

bchoineubility, 2023-12-07:
Hi @robwoolley 

Can you share added-the-py_ssize_t_clean-argument-to-the-module-directive.patch you create? 
I can not find it to test it.

BR,
Mark

robwoolley, 2023-12-08:
It seems that "git am" stripped whitespace at the end of the lines when I reapplied the patches.  This causes the breakage you saw.  I revised the git commit and force pushed the changes to robwoolley/meta-openembedded.

The clean build I did with this seemed to work, hopefully it works for you now with your setup:

```
git clone -b kirkstone git://git.yoctoproject.org/poky

cd poky

git clone -b kirkstone https://github.com/robwoolley/meta-openembedded.git

 . oe-init-build-env

bitbake-layers add-layer ../meta-openembedded/meta-oe/

bitbake sip3 sip3-native
```

bchoineubility, 2023-12-11:
Thanks Rob, 

I could build sip3 with new patch. by the way, trying to build qt-gui-cpp have also error, it should be hard working as you mentioned.

```
| /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.7-r0/recipe-sysroot-native/usr/bin/sip -w -n PyQt5.sip -t WS_X11 -t Qt_5_15_0 -x PyQt_Desktop_OpenGL -x PyQt_Accessibility -x PyQt_SessionManager -x PyQt_OpenGL -f -P -o -y QtCore.pyi -c /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.7-r0/PyQt5-5.15.7/QtCore -I sip -I /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.7-r0/PyQt5-5.15.7/sip /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.7-r0/PyQt5-5.15.7/sip/QtCore/QtCoremod.sip
| sip: /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.7-r0/PyQt5-5.15.7/sip/QtCore/QtCoremod.sip:23: syntax error
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/sources/meta-qt5/recipes-python/pyqt5/python3-pyqt5-native_5.15.7.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2389 tasks of which 2386 didn't need to be rerun and 2 failed.

Summary: 2 tasks failed:
  /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/sources/meta-qt5/recipes-python/pyqt5/python3-pyqt5_5.15.7.bb:do_configure
  /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/sources/meta-qt5/recipes-python/pyqt5/python3-pyqt5-native_5.15.7.bb:do_configure
Summary: There were 2 ERROR messages, returning a non-zero exit code.
bchoi@nvidia-build:~/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build$
```

Can you look into it also?

BR,
Mark

bchoineubility, 2023-12-11:
Hi Rob,

I am looking into segment fault for python3-pyqt5-native and get a clue.

```
openat(AT_FDCWD, "/home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.7-r0/PyQt5-5.15.7/Qt/sipQtcmodule.c", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=0, ...}, AT_EMPTY_PATH) = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x29} ---
+++ killed by SIGSEGV (core dumped) +++
```

And I found size of `tmp/work/x86_64-linux/python3-pyqt5-native/5.15.7-r0/PyQt5-5.15.7/Qt/sipQtcmodule.c` is zero.

bchoineubility, 2023-12-13:
> It seems that "git am" stripped whitespace at the end of the lines when I reapplied the patches. This causes the breakage you saw. I revised the git commit and force pushed the changes to [...]

Hi @robwoolley 

This patch works fine.  I had an version problem for my meta-qt5 and it resolved for now by myself.

Thanks.

BR,
Mark

bchoineubility, 2023-12-14:
Hi @robwoolley ,

I could build sip3 and sip3-native after applying patch. but, I could not build python3-pyqt5-native.

I tried to build with :  
```
meta-qt5             = "kirkstone-next:f5dfcd1417fa01dc59b3fc1f28bbe7c986de0e9d"

bchoi@nvidia-build:~/work/third-party/meta-openembedded$ git remote -v
origin  https://github.com/robwoolley/meta-openembedded.git (fetch)
origin  https://github.com/robwoolley/meta-openembedded.git (push)
bchoi@nvidia-build:~/work/third-party/meta-openembedded$ git branch

* kirkstone
bchoi@nvidia-build:~/work/third-party/meta-openembedded$ git show
commit ad1494b75f6ece452bdaec88c32772a32cd9186b (HEAD -> kirkstone, origin/kirkstone)
```
  
l attached log and run file. [log_run_pyqt5-native.zip](https://github.com/ros/meta-ros/files/13670049/log_run_pyqt5-native.zip)

BR,
Mark

bchoineubility, 2023-12-18:
Hi @robwoolley ,

I tried to debug for segmentation fault and got followings.

```
Starting program: /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.7-r0/recipe-sysroot-native/usr/bin/sip_debug -w -n PyQt5.sip -t WS_X11 -t Qt_5_15_0 -x PyQt_Desktop_OpenGL -x PyQt_Accessibility -x PyQt_SessionManager -x PyQt_OpenGL -f -P -c /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.7-r0/PyQt5-5.15.7/Qt -I sip -I /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.7-r0/PyQt5-5.15.7/sip /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/work/x86_64-linux/python3-pyqt5-native/5.15.7-r0/PyQt5-5.15.7/sip/Qt/Qtmod.sip
warning: Error disabling address space randomization: 명령을 허용하지 않음
BFD: /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2: unknown type [0x13] section `.relr.dyn'
warning: `/home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2': Shared library architecture unknown is not compatible with target architecture i386:x86-64.
warning: `/home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2': Shared library architecture unknown is not compatible with target architecture i386:x86-64.
BFD: /home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/sysroots-uninative/x86_64-linux/lib/libc.so.6: unknown type [0x13] section `.relr.dyn'
warning: `/home/bchoi/nvidia-yocto-bsp/official_ros_humble_build/bchoi-build/tmp/sysroots-uninative/x86_64-linux/lib/libc.so.6': Shared library architecture unknown is not compatible with target architecture i386:x86-64.

Program received signal SIGSEGV, Segmentation fault.
generateCompositeCpp (pt=0x7ffc1bf7d4a0, pt=0x7ffc1bf7d4a0, py_debug=0, codeDir=<optimized out>) at gencode.c:1141
1141        if (isPY_SSIZE_T_CLEAN(mod))
(gdb)
```

It would be caused by referencing variable "mod" which is  NULL.  fuction generateCompositeCpp only declare it as a pointer and do not allocate memory now.

BR,
Mark

bchoineubility, 2023-12-18:
I think this patch need to change as following because code intent to use to generate PY_SSIZE_T_CLEAN :

```
 1140 #ifdef MOD_PY_SSIZE_T_CLEAN
 1141         prcode(fp,
 1142 "\n"
 1143 "#define PY_SSIZE_T_CLEAN\n"
 1144             );
 1145 #endif
```

bchoineubility, 2023-12-19:
And I just add to qtbase-native.bb in the meta-qt5 it blown away a problem "module not found QT : gui" . 

```
PACKAGECONFIG += "gui"
```

robwoolley, 2025-09-02:
This was merged into meta-qt5:
https://github.com/meta-qt5/meta-qt5/commit/1fa0490319be5d188aef2fc004fd8b0dba962377
https://github.com/meta-qt5/meta-qt5/commit/e7795708cd4aecb51547104b4aac8511b1dc7b08

closed 2025-09-02 by robwoolley