ros/meta-ros #1082: Python nodes not locating correctly [Kirkstone/Humble]

Bentis, 2023-12-18:
**Describe the bug**
Python nodes not running through ros2 command.

**To Reproduce**
Yocto version: Kirkstone
ROS Distro: Humble

I'm building a basic image which is in essence `ros-image-core` with additional packages `demo-nodes-cpp demo-nodes-py` for imx8 platform.

When listing the executables in the packages, only the cpp package works.
```
root@imx8mn-var-som:~# source ros_setup.bash
root@imx8mn-var-som:~# ros2 pkg executables demo_nodes_cpp
demo_nodes_cpp add_two_ints_client
(..snip..)
demo_nodes_cpp talker
demo_nodes_cpp talker_loaned_message
demo_nodes_cpp talker_serialized_message
root@imx8mn-var-som:~# ros2 pkg executables demo_nodes_py 
root@imx8mn-var-som:~# 
```

Using an invalid package name it will give an error, so the package name is validated, but it cannot find any executables.

If I run a strace on the ros2 command it seems to only check for binaries in `/usr/lib/<pkg-name>/`

> openat(AT_FDCWD, "/usr/lib/demo_nodes_py", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = -1 ENOENT (No such file or directory)
> write(2, "No executable found\n", 20No executable found

I can find the python module in `/usr/lib/python3.10/site-packages/demo_nodes_py/`  and running the scripts from there directly works.

**Expected behavior**
ros2 locating python packages, nodes, executables

---

Seems to affect all python ros packages.
Thinking it might be related to the ament_python build type installation task ?

Bentis, 2023-12-18:
actually it seems the entry_points are installed in `/usr/bin/`,
while ros expects them to be in `/usr/lib/<pkg-name>/`

> tmp/work/armv8a-fslc-linux/demo-nodes-py/0.20.3-1-r0/package/usr/bin/talker_qos
> tmp/work/armv8a-fslc-linux/demo-nodes-py/0.20.3-1-r0/package/usr/bin/talker

```
root@imx8mn-var-som:~# /usr/bin/talker
[INFO] [1702914104.194657681] [talker]: Publishing: "Hello World: 0"
[INFO] [1702914105.107813555] [talker]: Publishing: "Hello World: 1"
```

jiaxshi, 2024-05-21:
You can refer to my PR for fixing this issue:
https://github.com/ros/meta-ros/commit/9ac73da79c8134cf7b9da167ddb891dfc9909e8b

robwoolley, 2026-11-02:
Since, the ROS applications are now installed in /opt/ros/<DISTRO> the original problem should no longer be present.  Closing for now.  Please open a new issue if the problem persists.

closed 2026-11-02 by robwoolley