ros/meta-ros #1086: Build error for aerostack2

bchoineubility, 2024-01-03:
Hi,

Happy new year. :)

I have an build error for aerostack2 package.

```
Missing or unbuildable dependency chain was: ['as2-python-api', 'ROS_UNRESOLVED_DEP-python3-pydantic']
ERROR: Required build target 'aerostack2' has no buildable providers.
Missing or unbuildable dependency chain was: ['aerostack2', 'as2-python-api', 'ROS_UNRESOLVED_DEP-python3-pydantic']
```
aerostack2 package have dependency for pydantic which would be available for nanbield not kirkstone.

Perhaps, python system of nanbield would be backported to kirkstone.

BR,
Mark

robwoolley, 2024-01-03:
Hi Mark,

Thanks again for reporting build errors.  I fixed this as part of the work to get ros-image-world to build for Kirkstone+Humble.  I am still working on getting everything ready to push upstream, but I did push the fix for this to kirkstone-next: https://github.com/ros/meta-ros/commit/d8cae1cd72c2edbc228e20ad680c84aca0efb8df

Regards,
Rob

bchoineubility, 2024-01-05:
Hi Rob,

Thanks for your work.

I am looking forward pushing your job.

BR,
Mark

robwoolley, 2024-01-12:
This should be fixed now in the kirkstone-next branch for Kirkstone+Humble.   Closing for now, please reopen if you encounter any problems.  Thanks for reporting this!

closed 2024-01-12 by robwoolley