ros/meta-ros #1091: Bitbake parsing error due to undocumented dependency on meta-qt5 in kirkstone-next branch in meta-ros2-humble layer

nicklausd, 2024-01-16:
**Describe the bug**
Trying to build a distro with Humble based on the "kirkstone-next" without meta-qt5 results in a parsing failure in Bitbake due to a change to the file `meta-ros2-humble/recipes-bbappends/qt-gui-core/qt-gui-cpp_2.2.2-1.bbappend` in commit c28e1d1 that added a dependency to the class `qmake5_base` (note that I haven't checked if this is also in any other branches or ROS distributions as I am not currently working with them). The `qmake5_base` class is not a part of this repo nor any layer in the `LAYERDEPENDS` in the `layer.conf` for Humble - it comes from `meta-qt5`. The `SKIP_RECIPE` definitions in `ros-distro-recipe-blacklist.inc` for when qt5 isn't enabled are also not enough to prevent this issue as `SKIP_RECIPE` merely stops a recipe from _building_, but not _parsing_. The [BBMASK](https://docs.yoctoproject.org/dev/ref-manual/variables.html#term-BBMASK) variable is the only solution I know of to completely stop parsing of a recipe.

As a temporary workaround, I am adding `BBMASK = "qt-gui-cpp"` to my own distro config to prevent it from parsing and the rest of my build goes as it did before c28e1d1.

**To Reproduce**
Layers used:

- poky (with sublayers meta, meta-poky, and meta-yocto-bsp)
- meta-openembedded (with sublayers meta-oe, meta-perl, meta-python, meta-networking, and meta-webserver)
- meta-raspberrypi
- meta-ros (with sublayers meta-ros-common, meta-ros2, and meta-ros2-humble)

My image is fairly minimal, with the only ROS-related entries in my `IMAGE_INSTALL` being `ros-core` and `python3-colcon-common-extensions`.

My MACHINE is sort-of custom but is essentially just `raspberrypi4-64` with some tweaks to use a custom .wks file and some additional device tree overlays. It should not matter in any case - the problem should be reproducible for any machine so long as you are not pulling meta-qt5 in your build config.

I am not using mcf for build setup - I'm using kas with a config defining the above layers. I took a quick peak at the mcf defs for my configuration (kirkstone+humble) though and I suspect the problem will be reproducible using it if pointed to kirkstone-next since it also does not reference meta-qt5.

The error I get upon a parsing failure is the following:

`ERROR: ParseError at /mnt/data/build/../layers/meta-ros/meta-ros2-humble/recipes-bbappends/qt-gui-core/qt-gui-cpp_2.2.2-1.bbappend:6: Could not inherit file classes/qmake5_base.bbclass`

**Expected behavior**
If meta-qt5 is a hard requirement after c28e1d1 or related changes, I'd expect it to be documented somewhere so I know to include it in my build config before running into weird parsing errors. If meta-qt5 is not expected to be a hard requirement, then I'd expect parsing should be skipped entirely (likely via `BBMASK` as mentioned above) for recipes such as qt-gui-cpp that may depend on it.

rcwoolley, 2024-01-23:
I have pushed this commit to kirkstone-next: https://github.com/ros/meta-ros/commit/685215f9837d47089d2eaaa296c1dac574f44ddf

It should resolve your issue.  Thanks for reporting it!

robwoolley, 2024-01-23:
I have pushed this commit to kirkstone-next: https://github.com/ros/meta-ros/commit/685215f9837d47089d2eaaa296c1dac574f44ddf

It should resolve your issue. Thanks for reporting it!

nicklausd, 2024-01-23:
> I have pushed this commit to kirkstone-next: [685215f](https://github.com/ros/meta-ros/commit/685215f9837d47089d2eaaa296c1dac574f44ddf)
> 
> It should resolve your issue. Thanks for reporting it!

It does indeed resolve it, thanks for the quick fix!

closed 2024-01-23 by nicklausd