ros/meta-ros #1096: ROS2 Humble - nav2-costmap-2d build error

langung102, 2024-01-26:
**Describe the bug**
Hi, i'm trying build nav2-costmap-2d in Humble distro on "kirkstone". And i got this error:

```
[13/36] /media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ --sysroot=/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DTINYXML2_IMPORT -Dnav2_costmap_2d_core_EXPORTS -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/git/include -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/eigen3 -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/geometry_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/laser_geometry -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/map_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/message_filters -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/nav2_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/nav_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/pluginlib -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rclcpp -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rclcpp_lifecycle -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/sensor_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/std_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/std_srvs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/tf2 -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/tf2_geometry_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/tf2_ros -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/visualization_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/angles -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/lifecycle_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rclcpp_action -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/bond -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/action_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rcl_interfaces -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/builtin_interfaces -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rosidl_runtime_c -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rcutils -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_interface -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rosidl_runtime_cpp -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_fastrtps_cpp -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rmw -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_fastrtps_c -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_introspection_c -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_introspection_cpp -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/ament_index_cpp -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/libstatistics_collector -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rcl -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rcl_logging_interface -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rcl_yaml_param_parser -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/tracetools -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rcpputils -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/statistics_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rosgraph_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_cpp -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_c -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/unique_identifier_msgs -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/class_loader -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rcl_lifecycle -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/orocos_kdl -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rcl_action -I/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/tf2_msgs -mcpu=cortex-a53 -march=armv8-a+crc -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0=/usr/src/debug/nav2-costmap-2d/1.1.12-1-r0                      -fdebug-prefix-map=/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0=/usr/src/debug/nav2-costmap-2d/1.1.12-1-r0                      -fdebug-prefix-map=/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot=                      -fdebug-prefix-map=/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav [... cut ...]
/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/logging.hpp:535:5: note: in expansion of macro 'RCUTILS_LOG_INFO_NAMED'
  535 |     RCUTILS_LOG_INFO_NAMED( \
      |     ^~~~~~~~~~~~~~~~~~~~~~
/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/git/src/clear_costmap_service.cpp:89:3: note: in expansion of macro 'RCLCPP_INFO'
   89 |   RCLCPP_INFO(
      |   ^~~~~~~~~~~
cc1plus: all warnings being treated as errors
```

**To Reproduce**
Layers: poky, meta-openembedded, meta-raspberrypi
Machine: raspberrypi3-64

Can anyone help me?
Thank you in advance!

robwoolley, 2024-01-31:
Thanks for reporting this.  Did you use kirkstone or kirkstone-next for this?

I haven't seen this particular compiler warning before:
```
/media/minhlang102/Share/yocto/build/tmp/work/cortexa53-poky-linux/nav2-costmap-2d/1.1.12-1-r0/recipe-sysroot/usr/include/rcutils/rcutils/logging_macros.h:72:18: error: format not a string literal and no format arguments [-Werror=format-security]
```

I believe that it happened, but I am confused why I didn't see it when building for qemux86_64.

minhlang102, 2024-02-04:
Hello @robwoolley, thanks for your reply.

I think my project was not set up correctly. You're right, i have tried follow step in REAME and i built it successfully with qemux86_64 and raspberrypi3-64.

Thank you again.

langung102, 2024-02-04:
Hello @robwoolley, thanks for your reply.

I think my project was not set up correctly. You're right, i have tried follow step in REAME and i built it successfully with qemux86_64 and raspberrypi3-64.

Thank you again.

closed 2024-02-04 by langung102