ros/meta-ros #1098: The compilation of bond-core requires a license [humble]

henaiguo, 2024-01-30:
**Describe the bug**
When installing bond-core, it shows  
ERROR: bond-core-3.0.2-2r0 do_create_spdx: Cannot find any test for license BSD

rcwoolley, 2024-01-30:
Thanks for reporting this.  The license identifiers used by ROS are not SPDX identifiers that are required by Yocto.  (full details here: https://github.com/ros/meta-ros/issues/1078#issuecomment-1862982325 )

Until this is fixed upstream in ROS, the appropriate workaround is to add this to your local.conf:

`
INHERIT_DISTRO:remove = "create-spdx"
WARN_QA:remove = "license-exists"
`

We are revising the getting started instructions and will make sure to include that step.

Regards,
Rob

robwoolley, 2024-01-30:
Thanks for reporting this. The license identifiers used by ROS are not SPDX identifiers that are required by Yocto. (full details here: https://github.com/ros/meta-ros/issues/1078#issuecomment-1862982325 )

Until this is fixed upstream in ROS, the appropriate workaround is to add this to your local.conf:

INHERIT_DISTRO:remove = "create-spdx" WARN_QA:remove = "license-exists"

We are revising the getting started instructions and will make sure to include that step.

Regards,
Rob

closed 2024-01-30 by robwoolley