ros/meta-ros #1099: No 'rosidl_typesupport_c' found

Marshall-Wen-Amba, 2024-02-02:
When I tried to use "colcon build" to do native build for a topic node on the board with Yocto ROS humble, I met below issue.
It is about "**No 'rosidl_typesupport_c' found**".
Actually, rosidl_typesupport_c was built into the package.
It is appreciated that if any of you here has any good idea. Thanks a lot.

# find -name rosidl_typesupport_c
./ament_index/resource_index/rosidl_typesupport_c
./ament_index/resource_index/package_run_dependencies/rosidl_typesupport_c
./ament_index/resource_index/packages/rosidl_typesupport_c
./ament_index/resource_index/rosidl_runtime_packages/rosidl_typesupport_c
./ament_index/resource_index/parent_prefix_path/rosidl_typesupport_c
./rosidl_typesupport_c

# colcon build --base-paths ./ \
>    --build-base ./build \
>    --install-base ./install \
>    --packages-select topic_link \
>    --event-handlers console_direct+ \
>    --cmake-args -DCMAKE_VERBOSE_MAKEFILE=ON
Starting >>> topic_link
-- Build type: Release
-- Found ament_cmake: 1.3.5 (/usr/share/ament_cmake/cmake)
-- Found rclcpp: 16.0.6 (/usr/share/rclcpp/cmake)
-- Found rosidl_generator_c: 3.1.5 (/usr/share/rosidl_generator_c/cmake)
-- Found rosidl_adapter: 3.1.5 (/usr/share/rosidl_adapter/cmake)
-- Found rosidl_generator_cpp: 3.1.5 (/usr/share/rosidl_generator_cpp/cmake)
CMake Error at /usr/share/rosidl_typesupport_c/cmake/get_used_typesupports.cmake:35 (message):
  No 'rosidl_typesupport_c' found
Call Stack (most recent call first):
  /usr/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:8 (get_used_typesupports)
  /usr/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /usr/share/builtin_interfaces/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /usr/share/builtin_interfaces/cmake/builtin_interfacesConfig.cmake:41 (include)
  /usr/share/libstatistics_collector/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /usr/share/libstatistics_collector/cmake/libstatistics_collectorConfig.cmake:41 (include)
  /usr/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /usr/share/rclcpp/cmake/rclcppConfig.cmake:41 (include)
  CMakeLists.txt:16 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/root/build/topic_link/CMakeFiles/CMakeOutput.log".
--- stderr: topic_link
CMake Error at /usr/share/rosidl_typesupport_c/cmake/get_used_typesupports.cmake:35 (message):
  No 'rosidl_typesupport_c' found
Call Stack (most recent call first):
  /usr/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:8 (get_used_typesupports)
  /usr/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /usr/share/builtin_interfaces/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /usr/share/builtin_interfaces/cmake/builtin_interfacesConfig.cmake:41 (include)
  /usr/share/libstatistics_collector/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /usr/share/libstatistics_collector/cmake/libstatistics_collectorConfig.cmake:41 (include)
  /usr/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /usr/share/rclcpp/cmake/rclcppConfig.cmake:41 (include)
  CMakeLists.txt:16 (find_package)

---
Failed   <<< topic_link [0.76s, exited with code 1]

Summary: 0 packages finished [1.36s]
  1 package failed: topic_link
  1 package had stderr output: topic_link

Marshall-Wen-Amba, 2024-02-04:
Already got the answer. Thanks~~

Marshall-Wen-Amba, 2024-02-04:
Already know how to fix it.

closed 2024-02-04 by Marshall-Wen-Amba

RodBelaFarin, 2024-02-20:
Would you be so kind to share your fix? I just now hit the same error for a different package (openvino_wrapper_lib).

Marshall-Wen-Amba, 2024-02-20:
> Would you be so kind to share your fix? I just now hit the same error for a different package (openvino_wrapper_lib).

Hi Rod:

If rosidl_typesupport_c is confirmed to be included in your firmware, pls make sure you do "# source /usr/bin/ros_local_setup.bash" before colcon build.

RodBelaFarin, 2024-02-20:
Thanks for sharing!

Unfortunately this file does not exist on my system (Ubuntu 22.04). I installed rosidl_typesupport_c as a binary, so usually source /opt/ros/humble/setup.bash should the work, but it's not.

I will continue to try.

Marshall-Wen-Amba, 2024-02-20:
Hi Rod:

Let me clarify a bit: here is talking about how to make "colcon build" work on Yocto ROS2 world, where it runs in an embedded ARM Linux environment.

For PC ROS2, I think it should be more likely to be some dependency issues, which can be resolved by "sudo apt-get install XXX". :)