ros/meta-ros #1101: BBMASK exclusion for rpi-config_%.bbappend should use override :append syntax

jdiez17, 2024-02-09:
**Describe the bug**
In my build (for a `phyboard-pollux-imx8mp`), I get the following build error:

```
ERROR: No recipes in default available for:
  /workdir/meta-ros/meta-ros-common/recipes-bsp/bootfiles/rpi-config_%.bbappend
```

This appears to be because the [BBMASK append](https://github.com/ros/meta-ros/blob/master/meta-ros-common/conf/ros-distro/ros-distro.conf#L50) in meta-ros-common uses the `+=` append syntax, and presumably the value of `BBMASK` is reset by something else in my build configuration.

Unfortunately, I cannot get the variable history with `bitbake-getvar -r raccoon-image-minimal BBMASK` because of the error above.

However, changing the above linked line in `meta-ros-config` to be an `:append` does cause the build to work correctly.

```diff
diff --git a/meta-ros/meta-ros-common/conf/ros-distro/ros-distro.conf b/meta-ros/meta-ros-common/conf/ros-distro/ros-distro.conf
index 8e38ab60034..8a264f0724b 100644
--- a/meta-ros/meta-ros-common/conf/ros-distro/ros-distro.conf
+++ b/meta-ros/meta-ros-common/conf/ros-distro/ros-distro.conf
@@ -47,7 +47,7 @@ ROS_EXTRA_BUILDCFG_VARS ?= "DISTRO_NAME ROS_DISTRO ROS_VERSION ROS_PYTHON_VERSIO
 BUILDCFG_VARS:append = " ${ROS_EXTRA_BUILDCFG_VARS}"

 # Ignore metadata that requires meta-raspberrypi if it's not present.
-BBMASK +=  "${@bb.utils.contains('BBFILE_COLLECTIONS', 'raspberrypi', '', 'recipes-bsp/bootfiles/rpi-config_%.bbappend', d)}"
+BBMASK:append =  " ${@bb.utils.contains('BBFILE_COLLECTIONS', 'raspberrypi', '', 'recipes-bsp/bootfiles/rpi-config_%.bbappend', d)}"
```

**After** the change, this is the variable history of BBMASK:

```
#
# $BBMASK [3 operations]
#   :append /workdir/meta-ros/meta-ros-common/conf/ros-distro/ros-distro.conf:50
#     " ${@bb.utils.contains('BBFILE_COLLECTIONS', 'raspberrypi', '', 'recipes-bsp/bootfiles/rpi-config_%.bbappend', d)}"
#   set /workdir/meta/conf/documentation.conf:93
#     [doc] "Prevents BitBake from processing specific recipes or recipe append files."
#   override[mx8m-nxp-bsp]::append /workdir/meta-phytec/conf/layer.conf:38
#     " meta-imx/meta-bsp/recipes-graphics/imx-gpu-apitrace/imx-gpu-apitrace_10.0.0.bb"
# pre-expansion value:
#   " meta-imx/meta-bsp/recipes-graphics/imx-gpu-apitrace/imx-gpu-apitrace_10.0.0.bb ${@bb.utils.contains('BBFILE_COLLECTIONS', 'raspberrypi', '', 'recipes-bsp/bootfiles/rpi-config_%.bbappend', d)}"
BBMASK=" meta-imx/meta-bsp/recipes-graphics/imx-gpu-apitrace/imx-gpu-apitrace_10.0.0.bb recipes-bsp/bootfiles/rpi-config_%.bbappend"
```

Line 38 of `meta-phytec/conf/layer.conf` is 

```
BBMASK:mx8m-nxp-bsp:append = " meta-imx/meta-bsp/recipes-graphics/imx-gpu-apitrace/imx-gpu-apitrace_10.0.0.bb"
```
(I don't think this would resetting the variable value...)

**Expected behavior**
BBMASK contains `recipes-bsp/bootfiles/rpi-config_%.bbappend`.

If you need any more info, I'd be happy to answer.

moto-timo, 2024-02-09:
You/We can add a `dynamic-layers/raspberrypi` directory and add
`BBFILES_DYNAMIC += " \
    raspberrypi:${LAYERDIR}/dynamic-layers/raspberrypi/*/*/*.bb \
    raspberrypi:${LAYERDIR}/dynamic-layers/raspberrypi/*/*/*.bbappend \`

and 

`LAYERRECOMMENDS = "raspberrypi"`

to `conf/layer.conf`

moto-timo, 2024-02-09:
But probably you should just send a PR with the above _ to : change.

robwoolley, 2024-04-17:
This has been fixed on master and merged to other branches: https://github.com/ros/meta-ros/commit/0766fc4bd6b6dad0363d3c5659a2e880d0842656

closed 2024-04-17 by robwoolley