ros/meta-ros #1104: AMENT_PREFIX_PATH uses a semicolon instead of a colon

jdiez17, 2024-03-05:
**Describe the bug**
The two directories in `AMENT_PREFIX_PATH` set in [ros_ament_cmake](https://github.com/ros/meta-ros/blob/master/meta-ros2/classes/ros_ament_cmake.bbclass#L22) should be separated by a semicolon, not a colon.

I noticed this during my work on building workspaces using colcon. I had this error from `ament_cmake`:

```
| [5.673s] colcon.colcon_ros.prefix_path.ament WARNING The path '/build/tmp/work/cortexa57-poky-linux/ros-rust-example/1.0+gitAUTOINC+3e1126a711-r0/recipe-sysroot/usr;/build/t
mp/work/cortexa57-poky-linux/ros-rust-example/1.0+gitAUTOINC+3e1126a711-r0/recipe-sysroot-native/usr' in the environment variable AMENT_PREFIX_PATH doesn't exist 
```

robwoolley, 2024-03-09:
Thanks @jdiez17 for reporting this.

For future reference, we discovered that references in the code indicate that AMENT_PREFIX_PATH use the operating systems path delimiter, whereas CMAKE_PREFIX_PATH is meant to accept lists which in CMake are stored as strings using semicolon as the delimiter.

References:

- https://github.com/colcon/colcon-ros/blob/a0d62231efeb0e89a32bffa17ee3bab44dc485a3/colcon_ros/task/__init__.py#L56
- https://github.com/ament/ament_package/blob/35b15ae4d8c5e1dbb6ab622ff95d046aa3ca1b28/ament_package/template/prefix_level/local_setup.sh.in#L50
- https://github.com/ament/ament_cmake/blob/c1cf011746457c2dace5f5236b486af83030813b/ament_cmake_include_directories/cmake/ament_include_directories_order.cmake#L28

robwoolley, 2024-11-21:
Closing as solved

closed 2024-11-21 by robwoolley