ros/meta-ros #1105: Error for adding demo_nodes_py or demo_nodes_cpp to IMAGE_INSTALL

Jpgig, 2024-03-08:
I tried to add demo_nodes_py to IMAGE_INSTALL as below:
IMAGE_INSTALL:append = " demo-nodes-py"

and getting build error now as below:
ERROR: example-interfaces-0.9.3-2-r0 do_package: QA Issue: example-interfaces: Files/directories were installed but not shipped in any package:
  /usr/lib/python3.11/site-packages/example_interfaces/[example_interfaces_s__rosidl_typesupport_fastrtps_c.cpython-311-aarch64-linux-gnu.so](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2Fexample_interfaces_s__rosidl_typesupport_fastrtps_c.cpython-311-aarch64-linux-gnu.so__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCITES-lCg%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765315682%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=cXEFOfTCdHCP6VKXsrlt6va72%2BMAi4z73kg32d%2Bv6ug%3D&reserved=0)
  /usr/lib/python3.11/site-packages/example_interfaces/[example_interfaces_s__rosidl_typesupport_c.cpython-311-aarch64-linux-gnu.so](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2Fexample_interfaces_s__rosidl_typesupport_c.cpython-311-aarch64-linux-gnu.so__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCITehvVZQ%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765323090%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=ycsRlvk4Ekirn1jkygbCkghN4%2FpKPKpy4VBTW7IuyUU%3D&reserved=0)
  /usr/lib/python3.11/site-packages/example_interfaces/[example_interfaces_s__rosidl_typesupport_introspection_c.cpython-311-aarch64-linux-gnu.so](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2Fexample_interfaces_s__rosidl_typesupport_introspection_c.cpython-311-aarch64-linux-gnu.so__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCKdTtco2A%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765329321%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=32flueiSJZa4xEhGVXd1MQnsXxC1a5asTOsAcMJVEP0%3D&reserved=0)
  /usr/lib/python3.11/site-packages/example_interfaces/[__init__.py](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2F__init__.py__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCJ7csdhdA%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765335932%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=BgxTADElsRSCY8FfWju5%2FYjSveXrAiZuEZ2UQJNJG5c%3D&reserved=0)
  /usr/lib/python3.11/site-packages/example_interfaces/msg
  /usr/lib/python3.11/site-packages/example_interfaces/srv
  /usr/lib/python3.11/site-packages/example_interfaces/action
  /usr/lib/python3.11/site-packages/example_interfaces/msg/_u_int32_multi_array_s.c
  /usr/lib/python3.11/site-packages/example_interfaces/msg/_float32_multi_array_s.c
  /usr/lib/python3.11/site-packages/example_interfaces/msg/[_byte_multi_array.py](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2F_byte_multi_array.py__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCL94vCVeA%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765341328%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=mkNxfRFUo5ONdOl42Pu21Jcl0EQXYojcW6mf3WewPrk%3D&reserved=0)
  /usr/lib/python3.11/site-packages/example_interfaces/msg/[_empty.py](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2F_empty.py__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCIiMXvZaA%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765346990%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=fI%2B3UGHxIObcbE78YQSUHbxNBxGY9LlNjw9Pr9R2%2BIA%3D&reserved=0)
  /usr/lib/python3.11/site-packages/example_interfaces/msg/_byte_s.c
  /usr/lib/python3.11/site-packages/example_interfaces/msg/[_float32.py](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2F_float32.py__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCIpjJl59g%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765352151%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=a%2BVoenZ6ChOEo3D8d%2FUF4AB5%2F96TcnaWOfa2sYlL9cQ%3D&reserved=0)
  /usr/lib/python3.11/site-packages/example_interfaces/msg/[_int16_multi_array.py](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2F_int16_multi_array.py__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCJzUfQmpg%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765357423%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=fAOQEYdkRNA6eGV2uSy3hWKl0wjEcH6aPOxbHE%2Fy%2B08%3D&reserved=0)
  /usr/lib/python3.11/site-packages/example_interfaces/msg/_int32_s.c
  /usr/lib/python3.11/site-packages/example_interfaces/msg/_multi_array_dimension_s.c
  /usr/lib/python3.11/site-packages/example_interfaces/msg/[_float32_multi_array.py](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2F_float32_multi_array.py__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCIyoXoFCA%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765362950%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=hPZ1ull5UIZCQnICD4RxCLrfgg58JCUbMxcxPG%2FWSC8%3D&reserved=0)
  /usr/lib/python3.11/site-packages/example_interfaces/msg/_u_int32_s.c
  /usr/lib/python3.11/site-packages/example_interfaces/msg/_int8_multi_array_s.c
  /usr/lib/python3.11/site-packages/example_interfaces/msg/[_multi_array_dimension.py](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2F_multi_array_dimension.py__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCLjkz7_Qg%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765368332%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=%2F9GS7Dh3FpW6R32B327Ftc0OqkOsclWvFdZbr6X5ek0%3D&reserved=0)
  /usr/lib/python3.11/site-packages/example_interfaces/msg/_int64_multi_array_s.c
  /usr/lib/python3.11/site-packages/example_interfaces/msg/[_bool.py](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2F_bool.py__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCIwnZ0iqg%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765373996%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=ziKZsPf7X0tIO7MvtaHPkM41K%2BB0dWO7PU1PtpicU6o%3D&reserved=0)
  /usr/lib/python3.11/site-packages/example_interfaces/msg/[_int32.py](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Furldefense.com%2Fv3%2F__http%3A%2F%2F_int32.py__%3B!!AjveYdw8EvQ!YhfFeCRxND0I [... cut ...] rfaces_0.9.3-2.bb__%3B!!AjveYdw8EvQ!YhfFeCRxND0IknQ5wbkh_SQwgrHy10K5AjYB-27Iy3VvfPaxpkdwDaK8aho1g5VychGabvceKhAhWNi7PCIlh7B24A%24&data=05%7C02%7Cjiten.patel%40aptiv.com%7C910de6cb3fcc47f792d208dc3f7dcc5a%7C6b1311e5123f49dbacdf8847c2d00bed%7C1%7C0%7C638455056765546182%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=i%2Bf%2F97MpuRLDgKDgI1SdBiwA96jX0lacGT7OTea1%2BJY%3D&reserved=0):do_package) failed with exit code '1

Can you guide here?

robwoolley, 2024-03-09:
Thanks @Jpgig,  meta-ros sets PYTHON_SITEPACKAGES_DIR in this class (https://github.com/ros/meta-ros/blob/e45cd718ea2d319afa2c9a6b610e55397df7d0c7/meta-ros-common/classes/ros_opt_prefix.bbclass#L28) to use ros_libdir = "${ros_prefix}/${baselib}" which should be /usr/lib64 on your aarch64 target.

I took a look at the CMakeLists.txt for example_interfaces ( https://github.com/ros2/example_interfaces/blob/humble/CMakeLists.txt#L52 ).  I don't see a reason why it would use /usr/lib.  The install macro is relative and the recipe seems to inherit all the build system classes that would ensure that CMAKE_INSTALL_DIR or CMAKE_PREFIX_DIR is set appropriately.

I will have to do some test builds for mickledore+humble to see if I can reproduce the failure.

Jpgig, 2024-03-20:
Hi Rob, did you get some time to build this?

robwoolley, 2024-03-21:
Hi Jiten,

I noticed that the "mickledore" branch was missing some commits from "mickledore-next".  One of which specifically related to using the correct arch-specific libdir: https://github.com/ros/meta-ros/commit/689d33b7598a7a0dcfa9e777702cfbae586fac1f

There is a high-chance that this will solve you issue.   Please update your meta-ros layer to include the latest commits and do a rebuild.  Please let me know if this helps solve your issue.

Regards,
Rob

Jpgig, 2024-03-21:
Thanks Rob, I took the latest commit still getting same error.

rcwoolley, 2024-03-22:
Hi Jiten,

I did fresh test builds for both recipes (demo-nodes-py and demo-nodes-cpp) for Humble and Mickledore on both 

**qemux86_64:**

> Build Configuration:
> BB_VERSION           = "2.4.0"
> BUILD_SYS            = "x86_64-linux"
> NATIVELSBSTRING      = "debian-11"
> TARGET_SYS           = "aarch64-oe-linux"
> MACHINE              = "qemuarm64"
> DISTRO               = "nodistro"
> DISTRO_VERSION       = "nodistro.0"
> TUNE_FEATURES        = "aarch64 armv8a crc cortexa57"
> TARGET_FPU           = ""
> DISTRO_NAME          = "OpenEmbedded"
> ROS_DISTRO           = "humble"
> ROS_VERSION          = "2"
> ROS_PYTHON_VERSION   = "3"
> meta-raspberrypi     = "mickledore:5c2d60b3f50d9b89978b58db0bf8068acf7e7877"
> meta-ros2-humble     
> meta-ros2            
> meta-ros-common      = "mickledore:2b64ab2a209f689689af3774f2aca9d3f0d6db17"
> meta-python          
> meta-oe              = "mickledore:39968837196cb48209b71e8852dd04a2f8ccdca8"
> meta                 = "mickledore:7e3489c0c5970389c8a239dc7b367bcadf554eb5"

**and qemuarm64:**

> Build Configuration:
> BB_VERSION           = "2.4.0"
> BUILD_SYS            = "x86_64-linux"
> NATIVELSBSTRING      = "debian-11"
> TARGET_SYS           = "aarch64-oe-linux"
> MACHINE              = "qemuarm64"
> DISTRO               = "nodistro"
> DISTRO_VERSION       = "nodistro.0"
> TUNE_FEATURES        = "aarch64 armv8a crc cortexa57"
> TARGET_FPU           = ""
> DISTRO_NAME          = "OpenEmbedded"
> ROS_DISTRO           = "humble"
> ROS_VERSION          = "2"
> ROS_PYTHON_VERSION   = "3"
> meta-raspberrypi     = "mickledore:5c2d60b3f50d9b89978b58db0bf8068acf7e7877"
> meta-ros2-humble     
> meta-ros2            
> meta-ros-common      = "mickledore:2b64ab2a209f689689af3774f2aca9d3f0d6db17"
> meta-python          
> meta-oe              = "mickledore:39968837196cb48209b71e8852dd04a2f8ccdca8"
> meta                 = "mickledore:7e3489c0c5970389c8a239dc7b367bcadf554eb5"

This was done using the [mcf](https://github.com/ros/meta-ros/blob/build/scripts/mcf) tool with the [ros2-humble-mickledore.mcf](https://github.com/ros/meta-ros/blob/build/files-contrib/ros2-humble-mickledore.mcf) configuration.  Both builds were successful.

It is possible that there is something specific to your setup that I'm not seeing.  Can you share step-by-step instructions for creating your build environment and running the build?

Can you also share the Build Configuration information from the beginning of your build? (eg as shown above)

Regards,
Rob

robwoolley, 2024-03-22:
Hi Jiten,

I did fresh test builds for both recipes (demo-nodes-py and demo-nodes-cpp) for Humble and Mickledore on both

**qemux86_64:**

> Build Configuration:
> BB_VERSION = "2.4.0"
> BUILD_SYS = "x86_64-linux"
> NATIVELSBSTRING = "debian-11"
> TARGET_SYS = "aarch64-oe-linux"
> MACHINE = "qemuarm64"
> DISTRO = "nodistro"
> DISTRO_VERSION = [...]

**and qemuarm64:**

> Build Configuration:
> BB_VERSION = "2.4.0"
> BUILD_SYS = "x86_64-linux"
> NATIVELSBSTRING = "debian-11"
> TARGET_SYS = "aarch64-oe-linux"
> MACHINE = "qemuarm64"
> DISTRO = "nodistro"
> DISTRO_VERSION = [...]

This was done using the [mcf](https://github.com/ros/meta-ros/blob/build/scripts/mcf) tool with the [ros2-humble-mickledore.mcf](https://github.com/ros/meta-ros/blob/build/files-contrib/ros2-humble-mickledore.mcf) configuration. Both builds were successful.

It is possible that there is something specific to your setup that I'm not seeing. Can you share step-by-step instructions for creating your build environment and running the build?

Can you also share the Build Configuration information from the beginning of your build? (eg as shown above)

Regards,
Rob

Jpgig, 2024-03-22:
I'm following below steps:
1) Added the ROS layer as per the link below:
https://gist.github.com/robwoolley/2feef3cf763d371d9a0c534cae1b38b6
2) I also added below to local.conf:
IMAGE_INSTALL:append = " sros2 demo-nodes-py"
EXTRA_OECMAKE:pn- += " -DSECURITY=ON"
3) Then building using below command:
bitbake wrlinux-image-small

I see below for ros layer:
meta-ros-common      
meta-ros2            
meta-ros2-humble     = "mickledore-next:2b64ab2a209f689689af3774f2aca9d3f0d6db17"

rcwoolley, 2024-03-24:
I managed to reproduce the problem with example_interfaces Wind River Linux LTS 23.  The ament_cmake_python package in ROS automatically tries to detect the PYTHON_SITE_PACKAGE directory.

The code can be seen here:
https://github.com/ament/ament_cmake/blob/9743c9cd3e5aef9566acefea6e6e3724e0e5657e/ament_cmake_python/ament_cmake_python-extras.cmake#L43

When running the python code under devshell, I can see it returns the following:
`>>> print(os.path.relpath(sysconfig.get_path('purelib', **kwargs), start='${CMAKE_INSTALL_PREFIX}').replace(os.sep, '/'))            lib/python3.11/site-packages`

I haven't compared this to poky mickledore to determine why it seems different (ie lib vs lib64).  However, a simple workaround may be to just add EXTRA_OECMAKE += " -DPYTHON_INSTALL_DIR=${baselib}/${PYTHON_DIR}/site-packages" in any affected packages (eg example-interfaces and demo-nodes-*)

robwoolley, 2024-03-24:
I managed to reproduce the problem with example_interfaces Wind River Linux LTS 23. The ament_cmake_python package in ROS automatically tries to detect the PYTHON_INSTALL_DIR directory.

The code can be seen here: https://github.com/ament/ament_cmake/blob/9743c9cd3e5aef9566acefea6e6e3724e0e5657e/ament_cmake_python/ament_cmake_python-extras.cmake#L43

When running the python code under devshell, I can see it returns the following:
> \>\>\> print(os.path.relpath(sysconfig.get_path('purelib', **kwargs), start='${CMAKE_INSTALL_PREFIX}').replace(os.sep, '/'))
lib/python3.11/site-packages

I compared this to the mickledore release for the Poky reference distro:

`bitbake-getvar -r example-interfaces  'FILES:example-interfaces'

> \#
> \# $FILES:example-interfaces [5 operations]
> \#   set /builds/oe-ros-ci/build-ci/openembedded-core/meta/conf/bitbake.conf:340
> \#     "${bindir}/* ${sbindir}/* ${libexecdir}/* ${libdir}/lib*${SOLIBS}             ${sysconfdir} ${sharedstatedir} ${localstatedir}             ${base_bindir}/* ${base_sbindir}/*             ${base_libdir}/*${SOLIBS}             ${base_prefix}/lib/udev ${prefix}/lib/udev             ${base_libdir}/udev ${libdir}/udev             ${datadir}/${BPN} ${libdir}/${BPN}/*             ${datadir}/pixmaps ${datadir}/applications             ${datadir}/idl ${datadir}/omf ${datadir}/sounds             ${libdir}/bonobo/servers"
> \#   append /builds/oe-ros-ci/build-ci/meta-ros/meta-ros-common/classes/ros_faulty_solibs.bbclass:5
> \#     "${libdir}/lib*${SOLIBSDEV}"
> \#   :prepend /builds/oe-ros-ci/build-ci/meta-ros/meta-ros-common/classes/ros_component.bbclass:23
> \#     "     ${datadir}/${ROS_BPN}     ${libdir}/${PYTHON_DIR}     ${libdir}/${ROS_BPN} "
> \#   :prepend /builds/oe-ros-ci/build-ci/meta-ros/meta-ros2/classes/ros_ament_cmake.bbclass:28
> \#     "     ${datadir}/ament_index "
> \#   rename from FILES:${PN} data.py:97 [expandKeys]
> \#     "${bindir}/* ${sbindir}/* ${libexecdir}/* ${libdir}/lib*${SOLIBS}             ${sysconfdir} ${sharedstatedir} ${localstatedir}             ${base_bindir}/* ${base_sbindir}/*             ${base_libdir}/*${SOLIBS}             ${base_prefix}/lib/udev ${prefix}/lib/udev             ${base_libdir}/udev ${libdir}/udev             ${datadir}/${BPN} ${libdir}/${BPN}/*             ${datadir}/pixmaps ${datadir}/applications             ${datadir}/idl ${datadir}/omf ${datadir}/sounds             ${libdir}/bonobo/servers ${libdir}/lib*${SOLIBSDEV}"
> \# pre-expansion value:
> \#   "     ${datadir}/ament_index      ${datadir}/${ROS_BPN}     ${libdir}/${PYTHON_DIR}     ${libdir}/${ROS_BPN} ${bindir}/* ${sbindir}/* ${libexecdir}/* ${libdir}/lib*${SOLIBS}             ${sysconfdir} ${sharedstatedir} ${localstatedir}             ${base_bindir}/* ${base_sbindir}/*             ${base_libdir}/*${SOLIBS}             ${base_prefix}/lib/udev ${prefix}/lib/udev             ${base_libdir}/udev ${libdir}/udev             ${datadir}/${BPN} ${libdir}/${BPN}/*             ${datadir}/pixmaps ${datadir}/applications             ${datadir}/idl ${datadir}/omf ${datadir}/sounds             ${libdir}/bonobo/servers ${libdir}/lib*${SOLIBSDEV}"
> FILES:example-interfaces="     /usr/share/ament_index      /usr/share/example_interfaces     /usr/lib/python3.11     /usr/lib/example_interfaces /usr/bin/* /usr/sbin/* /usr/libexec/* /usr/lib/lib*.so.*             /etc /com /var             /bin/* /sbin/*             /lib/*.so.*             /lib/udev /usr/lib/udev             /lib/udev /usr/lib/udev             /usr/share/example-interfaces /usr/lib/example-interfaces/*             /usr/share/pixmaps /usr/share/applications             /usr/share/idl /usr/share/omf /usr/share/sounds             /usr/lib/bonobo/servers /usr/lib/lib*.so"
> 

You can see from above that /usr/lib/python3.11 is derived from ${libdir} so I check libdir to see where it is defined.  It is as expected:

> bitbake-getvar -r example-interfaces libdir
> \#
> \# $libdir [2 operations]
> \#   exported /builds/oe-ros-ci/build-ci/openembedded-core/meta/conf/bitbake.conf:46
> \#     [export] "1"
> \#   set /builds/oe-ros-ci/build-ci/openembedded-core/meta/conf/bitbake.conf:46
> \#     "${exec_prefix}/${baselib}"
> \# pre-expansion value:
> \#   "${exec_prefix}/${baselib}"
> export libdir="/usr/lib"
> 
> bitbake-getvar -r example-interfaces baselib
> \#
> \# $baselib [2 operations]
> \#   set /builds/oe-ros-ci/build-ci/openembedded-core/meta/conf/bitbake.conf:10
> \#     "${BASELIB}"
> \#   set /builds/oe-ros-ci/build-ci/openembedded-core/meta/conf/bitbake.conf:11
> \#     [vardepvalue] "${baselib}"
> \# pre-expansion value:
> \#   "${BASELIB}"
> baselib="lib"
> 

So, then I check Wind River Linux LTS 23 which is based on mickledore as well:

`bitbake-getvar -r example-interfaces  'FILES:example-int
`

> WARNING: You have included the meta-security layer, but 'security' has not been enabled in your DISTRO_FEATURES. Some bbappend files and preferred version setting may not take effect. See the meta-security README for details on enabling security support.
> WARNING: Host distribution "debian-11" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
> \#                                                                                                                                    
[... cut ...]  can refer to the commit I pointed to before to see examples of how this was done for other packages: https://github.com/ros/meta-ros/commit/689d33b7598a7a0dcfa9e777702cfbae586fac1f

That said, it will likely require a more permanent fix for multilib in the later releases.  You may wish to file a bug report against Wind River Linux LTS 23 explaining that the Python sysconfig may be reporting the wrong information on multilib systems.   Then they can work to get it fixed upstream if it is needed.

Jpgig, 2024-03-26:
So your suggestion to add below:
EXTRA_OECMAKE += " -DPYTHON_INSTALL_DIR=${nonarch_libdir}/${PYTHON_DIR}/site-packages"
instead of this:
EXTRA_OECMAKE += " -DPYTHON_INSTALL_DIR=${baselib}/${PYTHON_DIR}/site-packages"
Below is the screen of my file which is under **meta-ros/meta-ros2-humble/generated-recipes/demos folder:**

[image]

Jpgig, 2024-03-27:
I'm able to compile now after adding below line to example-interfaces_0.9.3-2.bbappend recipe file:
EXTRA_OECMAKE += " -DPYTHON_INSTALL_DIR=${baselib}/${PYTHON_DIR}/site-packages"
but still getting below error when trying to run :  **ros2 run demo_nodes_py talker**
[image]

jiaxshi, 2024-04-22:
Hi @Jpgig 
You can refer to my PR for fixing this issue:
https://github.com/ros/meta-ros/pull/1102/commits/9ac73da79c8134cf7b9da167ddb891dfc9909e8b

robwoolley, 2026-11-02:
Closing as fixed by https://github.com/ros/meta-ros/commit/9ac73da79c8134cf7b9da167ddb891dfc9909e8b

closed 2026-11-02 by robwoolley