ros/meta-ros #1106: Enable Security for ROS

Jpgig, 2024-03-08:
Can you also guide me how to enable security portion of the ROS?
As per ROS humble documentation need to enable DSECURITY=ON while building the ROS.
https://docs.ros.org/en/humble/Tutorials/Advanced/Security/Introducing-ros2-security.html

robwoolley, 2024-03-09:
Thanks @Jpgig, I think this is a really good feature to enable.

Based on the documentation, it looks like the CMake macro needs to be enabled for the DDS provider:

> Fast DDS requires an additional CMake flag to build the security plugins, so the colcon invocation needs to be modified to pass:
> 
> colcon build --symlink-install --cmake-args -DSECURITY=ON

This seems to be specific to FastDDS ( https://fast-dds.docs.eprosima.com/en/v2.13.3/fastdds/security/security.html )

Is that the DDS provider you wish to use?  Do you know if the security feature can be enabled on other DDS providers or on Zenoh?  I am curious because the default DDS provider changed to Cyclone DDS at one point, DDS middleware from RTI is also popular, and now Zenoh is being pushed as an alternative.

Jpgig, 2024-03-10:
Thanks for quick response Rob.
As per below ROS humble documentation, **The default DDS vendor is eProsima’s Fast DDS.**
https://docs.ros.org/en/humble/Installation/DDS-Implementations.html
Is that the DDS provider you wish to use?
As FastDDS is default DDS provider, I'm using that.
Do you know if the security feature can be enabled on other DDS providers or on Zenoh? 
I have not looked into other DDS provider.

rcwoolley, 2024-03-18:
Based on the documentation, it sounds like you would need to do one of 2 options:

Modify the recipe for Fast DDS that needs -DSECURITY=ON and append it to the cmake invocation by adding:

       EXTRA_OECMAKE:append = " -DSECURITY=ON"

The other option would be to make a quick and dirty change to conf/local.conf:

      EXTRA_OECMAKE:pn-<PACKAGE NAME> += " -DSECURITY=ON"

Let me know what results you get.  You mentioned using Wind River Linux LTS earlier, I will try kicking off a test build with Mickledore and Humble this week as well.

Jpgig, 2024-03-20:
I tried this option. Seems no compilation error but can't test without examples able to added to run it.
Its related to another issue I logged as below:
Error for adding demo_nodes_py or demo_nodes_cpp to IMAGE_INSTALL #1105
Are you able to get that added?

ctgulzh, 2025-08-30:
Hello Jpgig!

Is the issue https://github.com/ros/meta-ros/issues/1105 resolved? I noticed that the latest example-interfaces_%.bbappend file in the /petalinux/kr260_xsa_ros/components/yocto/layers/meta-ros/meta-ros2-humble/recipes-bbappends/example-interfaces/ path doesn't actually include the changes referenced by https://github.com/ros/meta-ros/issues/1105. However, when I add demo_nodes_py or demo_nodes_cpp via IMAGE_INSTALL, the issue you mentioned doesn't occur. Why is this?

robwoolley, 2026-11-02:
The resolution of #1105 seems to resolve this issue.  Please feel free to open a new issue if there are any outstanding issues.

closed 2026-11-02 by robwoolley