ros/meta-ros #1107: Create CI/CD pipeline and infrastructure for meta-ros

robwoolley, 2024-03-09:
Which CI platforms are of interest to the group?
* **Jenkins**, **GitLab CI**, Tekton, GitHub Actions, etc
* @moto-timo had previously endorsed Tekton and K8s but now advocates for simple tools for automation:
* He suggested https://laminar.ohwg.net/docs.html and https://concourse-ci.org/ 
* @robwoolley has used popper in the past (no longer maintained): https://github.com/getpopper/popper
* He has also used Act to run GitHub Actions locally: https://github.com/nektos/act/releases

Which workflows are of interest?

- **Classic**: ie. bitbake or SDKs (Containers optional)
- **On-target**: ie. Building with bitbake, SDK, or natively on OpenEmbedded (incl. QEMU)
- **Self-hosted**: ie. GitLab for a single user (gitaly, container/package registries) with sstate cache and download dir; (Containers optional)
- **On-premise**: ie. GitLab for multiple users using tools like Harbor and Ceph / MinIO on self-managed infrastructure with RBAC & certificates; (Containers optional)
- **Cloud**: ie. Using a cloud environment with cloud services for storage and orchestration (eg. Cloud Service Provider (CSP) or Cloud Development Environment (CDE))
- Others?

References

- See #1071 for issues specific to GitHub Actions
- [DRM-CI: A GitLab-CI pipeline for Linux kernel testing](https://www.collabora.com/news-and-blog/blog/2024/02/08/drm-ci-a-gitlab-ci-pipeline-for-linux-kernel-testing/) 
- Also see https://lore.kernel.org/lkml/20240228225527.1052240-1-helen.koike@collabora.com/
- https://docs.mesa3d.org/ci/index.html
- [FreeDesktop / Mesa CI Gantt Charts](https://gitlab.freedesktop.org/mesa/mesa/-/tree/main/bin/ci?ref_type=heads)

- [ROS CI](https://wiki.ros.org/CIs)
- [ROS Build Farm](https://wiki.ros.org/buildfarm) - Based on Jenkins, Puppet, Reprepro, GitHub, and Ubuntu
- [ROS GitLab CI](https://gitlab.com/VictorLamoine/ros_gitlab_ci ) - Uses OSRF Docker images to compile packages for ROS 1
- [ROS Industrial (industrial_ci)](https://github.com/ros-industrial/industrial_ci) - Supports GitLab

- Yocto Summit Talks related to CI/CD
- Highly Scalable Build Automation by Paul Barker, Yocto Project Dev Day Virtual, North America, 2020 ([Slides](https://wiki.yoctoproject.org/wiki/images/archive/0/02/20200702140427%21DD1_Highly_Scalable_Build_Automation_NA20.pdf))
- “Gitlab CI for dummies” by Jon Mason, Bruce Ashfield, Yocto Project Summit 2021.05 ([Slides](https://elinux.org/images/c/c1/013-1845-SLIDES-gitlab_ci_for_dummies.pdf), [Video](https://www.youtube.com/watch?v=3EqYkhAaiN0))
- Automating Yocto image testing with Gitlab CI/CD, Clay Timmons, Yocto Project Summit 2022.11 ([Slides](https://summit.yoctoproject.org/media/yocto-project-summit-2022-11/submissions/ZGD8TF/resources/yps2022_gitlabCICD_pxTTWco.pdf), [Video](https://www.youtube.com/watch?v=CvLm5Ftv6XI))
- Building a Yocto CI Pipeline with KAS, GHActions and AWS by Anna-Lena Marx, Building YPS 2023.11 - 2023/11/29 ([Slides](https://marx.engineer/content/talks/2023_YoctoSummit_Building-a-Yocto-CI-GitHubActions.pdf), [Video](https://www.youtube.com/watch?v=2MAMCI8xvY8))
- Yocto Layer CI Build and Test with GitHub Actions, Alex Lennon ([Video](https://www.youtube.com/watch?v=TsAcxd_acJI))
- CI/CD for Yocto Project Maintainers With Kubernetes and Tekton Pipelines, Trevor Gamblin,Tim Orling ([Video](https://www.youtube.com/watch?v=luxMUcOB_JM))
- Yocto Continuous Integration in a Kube - Joshua Watt, Garmin ([Video](https://www.youtube.com/watch?v=GAkpFUmoMT0))

robwoolley, 2024-03-09:
@jonmason and @rossburton have an advanced gitlab-ci.yml for Yocto in meta-arm: https://github.com/jonmason/meta-arm/blob/master/.gitlab-ci.yml

moto-timo, 2024-03-10:
I have some older, "not-how-I-would-design-it-now" kas/gitlab-ci.yml usage on https://gitlab.com/moto-timo/yp-dev-day_virtualization (the home server that the jobs run on lost some spinning-rust back in June/July and is now offline).

Highlights there are usage of multiconfig and a bunch of different targets for specific functional reasons.

I am happy to explain how I would do it now if I was starting over again.

moto-timo, 2024-03-10:
Also, fairly extensive usage of GitHub Actions on the various crops projects https://github.com/crops which I am also more than happy to explain/clarify.

khem and cliff have done some good work with GitHub Actions for https://github.com/Yoedistro (and you can ask khem about self-hosted GitHub Actions runners.

moto-timo, 2024-03-10:
Pascal has had some limited but promising success with GitHub Actions for https://github.com/meta-homeassistant/meta-homeassistant which I am also experimenting with in other areas.

moto-timo, 2024-03-10:
I can also share my design/architecture for my ["Living on Master"](https://youtu.be/l6NwYGbWO5s?si=N63A7uethpKHeYlD) talk, but sadly I have no access to any of the code for that project. It never made it through the Open Sourcing process.

bchoineubility, 2024-03-19:
What are many CI/CD for ROS and Yocto !

I aware them in now.

robwoolley, 2024-04-17:
Support for GitLab CI has been added: https://github.com/ros/meta-ros/blob/build/.gitlab-ci.yml

GitHub Actions for superflore has been added: https://github.com/ros/meta-ros/blob/master/.github/workflows/generate_recipes.yml

Work will continue, however we've hit the MVP.

closed 2024-04-17 by robwoolley