ros/meta-ros #1108: Build-errors due to format-security

Bentis, 2024-03-11:
**Describe the bug**
Same build error as in #1096 
string-format warning in upstream code causes compilation error with default compiler flag.

**To Reproduce**
Using latest `meta-ros` on branch `kirkstone`:
`meta-ros2-humble     = "HEAD:685215f9837d47089d2eaaa296c1dac574f44ddf"`

Running:
`bitbake teleop-twist-joy` or  `bitbake nav2-costmap-2d`

```
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fslc-linux"
MACHINE              = "imx8mn-var-som"
DISTRO               = "fslc-xwayland"
DISTRO_VERSION       = "4.0"
TUNE_FEATURES        = "aarch64 armv8a crc crypto"
TARGET_FPU           = ""
DISTRO_NAME          = "FSLC Wayland with XWayland"
ROS_DISTRO           = "humble"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"
```
Log:
```
| FAILED: CMakeFiles/teleop_twist_joy.dir/src/teleop_twist_joy.cpp.o
| /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot-native/usr/bin/aarch64-fslc-linux/aarch64-fslc-linux-g++ -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -Dteleop_twist_joy_EXPORTS -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/build -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/git/include -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/geometry_msgs -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/std_msgs -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/builtin_interfaces -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rosidl_runtime_c -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rcutils -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_interface -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rosidl_runtime_cpp -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_fastrtps_cpp -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rmw -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_fastrtps_c -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_introspection_c -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_introspection_cpp -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/ament_index_cpp -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/libstatistics_collector -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rcl -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rcl_interfaces -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rcl_logging_interface -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rcl_yaml_param_parser -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/tracetools -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rcpputils -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/statistics_msgs -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rosgraph_msgs -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_cpp -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rosidl_typesupport_c -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp_components -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/class_loader -I/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/sensor_msgs -march=armv8-a+crc+crypto -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0=/usr/src/debug/teleop-twist-joy/2.4.5-1-r0                      -fdebug-prefix-map=/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0=/usr/src/debug/teleop-twist-joy/2.4.5-1-r0                      -fdebug-prefix-map=/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot=                      -fdebug-prefix-map=/workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden -fPIC -Wall -Wextra -Wpedantic -std=gnu++17 -MD -MT CMakeFiles/teleop_twist_joy.dir/src/teleop_twist_joy.cpp.o -MF CMakeFiles/teleop_twist_joy.dir/src/teleop_twist_joy.cpp.o.d -o CMakeFiles/teleop_twist_joy.dir/src/teleop_twist_joy.cpp.o -c /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/git/src/teleop_twist_joy.cpp
| In file included from /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/logging.hpp:24,
|                  from /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/client.hpp:40,
|                  from /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/callback_group.hpp:24,
|                  from /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/any_executable.hpp:20,
|                  from /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/memory_strategy.hpp:25,
|                  from /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/memory_strategies.hpp:18,
|                  from /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/executor_options.hpp:20,
|                  from /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/executor.hpp:37,
|                  from /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/executors/multi_threaded_executor.hpp:25,
|                  from /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/executors.hpp:21,
|                  from /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rclcpp/rclcpp/rclcpp.hpp:155,
[... cut ...]
| /workdir/yocto/build_fslc-xwayland/tmp/work/armv8a-fslc-linux/teleop-twist-joy/2.4.5-1-r0/recipe-sysroot/usr/include/rcutils/rcutils/logging_macros.h:72:18: error: format not a string literal and no format arguments [-Werror=format-security]
|    72 |       rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
```

**Expected behavior**
Builds without error (but with warnings?)

Seems the problem is upstream code which relies on printf without format-strings. So the warning is valid, but the handling as error causes issues for us.

Bentis, 2024-03-11:
Seems to originate from:
```
# $SECURITY_STRINGFORMAT [3 operations]
#   set? /workdir/yocto/sources/poky/meta/conf/distro/include/security_flags.inc:18
#     "-Wformat -Wformat-security -Werror=format-security"
```

The security_flags.inc file even contains a few exceptions already:

```
# Recipes which fail to compile when elevating -Wformat-security to an error
SECURITY_STRINGFORMAT:pn-busybox = ""
SECURITY_STRINGFORMAT:pn-gcc = ""
```

**Suggested fix:** 
Add exceptions for the problematic packages somewhere in `meta-ros2-humble`:
```
# Recipes which fail to compile when elevating -Wformat-security to an error
SECURITY_STRINGFORMAT:pn-teleop-twist-joy = ""
SECURITY_STRINGFORMAT:pn-nav2-costmap-2d = ""
```

Is `meta-ros2-humble/conf/ros-distro/include/humble/ros-distro.inc` the correct location for this?

robwoolley, 2024-03-14:
Thanks for reporting this.  I would suggest keeping it localized to the recipe.

https://github.com/ros/meta-ros/blob/e45cd718ea2d319afa2c9a6b610e55397df7d0c7/meta-ros1-melodic/recipes-bbappends/microstrain-mips/microstrain-mips_0.0.3-1.bbappend#L13

There doesn't appear to be a bbappend for teleop-twist-joy but there is one for nav2-costmap-2d:
https://github.com/ros/meta-ros/blob/kirkstone/meta-ros2-humble/recipes-bbappends/navigation2/nav2-costmap-2d_1.1.12-1.bbappend

You could set SECURITY_STRINGFORMAT = "" as one possible fix.

Another route would be to demote the error back to a warning with:
CXXFLAGS += "-Wno-error=format-security"

That may be more suitable as it shows the intent and may be more obvious to remove when no longer needed.