ros/meta-ros #1110: Build error dues to failed patch on rviz-orge-vendor

xchg-dot-ca, 2024-03-13:
Building 
ros2-humble for Yocto mickledore
when processing layer for rviz2 getting the error in colsole:
`
ERROR: rviz-ogre-vendor-11.2.8-1-r0 do_patch: Applying patch '0001-CMakeLists-remove-all-ExternalProject_Add-calls-and-.patch' on target directory '/media/mlytvynyuk/2TB/workspace/yocto/build/tmp/work/cortexa57-poky-linux/rviz-ogre-vendor/11.2.8-1-r0/git'
`
In patch log:
`
NOTE: Applying patch '0001-CMakeLists-remove-all-ExternalProject_Add-calls-and-.patch' (../poky/meta-ros/meta-ros2-humble/recipes-bbappends/rviz/rviz-ogre-vendor/0001-CMakeLists-remove-all-ExternalProject_Add-calls-and-.patch)
ERROR: Applying patch '0001-CMakeLists-remove-all-ExternalProject_Add-calls-and-.patch' on target directory '/media/mlytvynyuk/2TB/workspace/yocto/build/tmp/work/cortexa57-poky-linux/rviz-ogre-vendor/11.2.8-1-r0/git'
CmdError('quilt --quiltrc /2TB/yocto/build/tmp/work/cortexa57-poky-linux/rviz-ogre-vendor/11.2.8-1-r0/recipe-sysroot-native/etc/quiltrc push', 0, 'stdout: Applying patch 0001-CMakeLists-remove-all-ExternalProject_Add-calls-and-.patch
patching file CMakeLists.txt
Hunk #1 FAILED at 6.
`

Here is my ROS related chunk from local.conf

```bitbake
# ROS2 Humble core
CORE_IMAGE_EXTRA_INSTALL += " \
  ros-core \
  ignition-math6-vendor \
  ignition-cmake2-vendor \
  angles \
  dummy-sensors \
  demo-nodes-cpp \
  rviz2 \
  "
```

xchg-dot-ca, 2024-03-14:
Thank you for looking into this.
My idea is to run all this on aarch64le based board, including visualization.

xchg-dot-ca, 2024-03-21:
I will add more details:
This is an attempt to run Ros2/Rviz in yocto linux which runs under QNX hypervisor on aarch64 architecture.
Also we have solved yocto and qt and 3d acceleration
Using this repo I have successfully build image with yocto+ros2.

Addid rviz layer causing all sort of unmet dependencies.
namely ignition math6 ( which I tried to hack-around ) as well as other ignition packages

robwoolley, 2024-03-22:
I do have some unfinished work where I was adding 37+ new recipes to introduce ignition and some of its dependencies.  The challenge is all the optional graphics libraries and other optional dependencies that aren't already in Yocto/OpenEmbedded.  You can use recipetool to make it easier, but it is still significant work to get going.

The work I did was for Humble+Kirkstone, however it should provide a path to getting it working on Humble+Mickledore.   I estimate that it could be either days or weeks of work depending on your familiarity with Linux and/or Yocto.   Does that help?  If that's a challenge you're willing to take on, let me know.

whni, 2024-04-24:
Hi @robwoolley 

I am also doing a side project using Yocto + ROS2 (humble kirkstone) on Turtlebot3, and seeing that lots of ignition or gazebo dependencies are missing as reported on my machine.

My host machine: ubuntu 22.04
ROS setup steps: https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions
Ignition package install: https://gazebosim.org/api/math/6.4/install.html

Here are my project configurations
[local.conf.txt](https://github.com/ros/meta-ros/files/15097535/local.conf.txt)
[bblayers.conf.txt](https://github.com/ros/meta-ros/files/15097536/bblayers.conf.txt)

When trying to build `turtlebot3`
> NOTE: Runtime target 'rviz2' is unbuildable, removing...
> Missing or unbuildable dependency chain was: ['rviz2']
> NOTE: Runtime target 'turtlebot3-bringup' is unbuildable, removing...
> Missing or unbuildable dependency chain was: ['turtlebot3-bringup', 'rviz2']
> ERROR: Required build target 'turtlebot3' has no buildable providers.
> Missing or unbuildable dependency chain was: ['turtlebot3', 'turtlebot3-bringup', 'rviz2']

When trying to build `rviz2`
> ERROR: Nothing PROVIDES 'rviz2'
> rviz2 was skipped: Recipe will be skipped because: qt5: depends on qtbase; opengl: depends on rviz-ogre-vendor which depends on mesa which is not available because of missing opengl or vulkan in DISTRO_FEATURES; x11: depends on rviz-rendering which depends on rviz-ogre-vendor which depends on libx11,libxrandr,libxaw which require x11 in DISTRO_FEATURES; ignition: depends on rviz-default-plugins which depends on unavailable ROS_UNRESOLVED_DEP-ignition-math6

These error also occurs whenever I am trying to build any graphics things

robwoolley, 2024-04-24:
Hi Weiheng,

Glad to hear from you.  I have had the simulation tools like RViz and Gazebo running on Yocto in past releases (foxy or galactic if memory serves).  It is on my list of things to look at next.

I would be interested in hearing more about the use case because running a desktop on the robot itself seems unusual to me.  I would have assumed that you might run rviz2 and Gazebo on a laptop with a GPU running Ubuntu and connect over the network to your robot.  Can you tell me more about why running the simulation directly on the robot supports your work?

As far as build instructions are concerned, I just recently added support for using kas.  You may find it easier to automatically set up an environment: https://github.com/ros/meta-ros/blob/build/kas/README.md

Regards,
Rob

whni, 2024-04-24:
Hi @robwoolley

Sorry for the confusion. Ubuntu 22.04 is just my host machine to build the Yocto ROS2 project (humble + kirkstone). Actually I can install rviz2 and gazebo directly on ubuntu via apt. I didn't intend to build it into my robot image.

What I tried was just to build some default image like `turtlebot3` or `ros-image-world` according to the instruction here: https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions. 

However, when building these targets, it threw some dependency chain error like
```
Missing or unbuildable dependency chain was: ['turtlebot3', 'turtlebot3-bringup', 'rviz2']
```
That's why I also tried `bitbake rviz2` then reached the ignition dependency error.

robwoolley, 2024-04-24:
I have a Turtlebot3 myself and was planning on trying to get it working again for the upcoming Jazzy release.

May I suggest that you file a new issue for addressing the missing support for Turtlebot3?  We can work towards adding the missing packages there.

In the mean-time, if you just want to get ROS 2 Kirkstone + Humble running on the board, I suggest following these build instructions instead: https://github.com/ros/meta-ros/blob/build/kas/README.md

It will build ros-image-core which I have recently verified will work.  The ros-image-world literally builds everything and is more intended for nightly builds to look for regressions.  We are still working on fixing some outstanding packages to get it working 100%.

whni, 2024-04-24:
@robwoolley Thank you so much! I will file the issue. At the same time, could you suggest a previous version that works on turtlebot3? We are actively working on a project so want to bring up the bot asap