ros/meta-ros #1113: cmake*.bb causes issues with master

mhatle, 2024-04-04:
cmake*.bb causes issues with unrelated packages.  The older version of cmake (3.22.3) is being used instead of the standard Yocto Project version.

The files:
./meta-ros2-iron/recipes-devtools/cmake/cmake-native_3.22.3.bb
./meta-ros2-iron/recipes-devtools/cmake/cmake_3.22.3.bb
./meta-ros2-rolling/recipes-devtools/cmake/cmake-native_3.22.3.bb
./meta-ros2-rolling/recipes-devtools/cmake/cmake_3.22.3.bb
./meta-ros2-humble/recipes-devtools/cmake/cmake-native_3.22.3.bb
./meta-ros2-humble/recipes-devtools/cmake/cmake_3.22.3.bb

When a layer (I'm using humble w/ scarthgap) contains these, for some reason this version becomes the default cmake version.  I'm guessing it's a priority issue.  I've had to mask these out using BBMASK in order for unrelated software to correctly function with cmake.

Can thesse be removed from master (scarthgap)?

robwoolley, 2024-04-07:
Yes, I think we can remove it.   It's strange that the newer versions don't supersede this one.

I included CMake originally because the older Yocto releases required a newer CMake.  However, with Dunfell going EOL this month (and no one seemingly using it) we can simply remove the copy of the recipe.   The version provided by kirkstone should be sufficient going forward: https://layers.openembedded.org/layerindex/recipe/5305/

robwoolley, 2024-04-07:
I pushed a commit to master-next that removes all the copies of the cmake recipe.

Thanks for pointing this out, Mark.

I hope to merge to master after I have done some additional coverage builds.

robwoolley, 2024-04-08:
I have merged it to master.   The ros-image-core builds 100% for Raspberry Pi 4.   There are remaining issues for ros-image-world.  A build with --continue has 77 failures, but completes 19858 tasks out of 22099.

closed 2024-04-08 by robwoolley