ros/meta-ros #1138: Turtlebot3 support is missing on ROS2 humble kirkstone due to ignition/gazebo dependency errors

whni, 2024-04-24:
Turtlebot3 support is missing on ROS2 humble kirkstone due to ignition/gazebo dependency errors

Build host: ubuntu 22.04
ROS setup steps: https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions
Ignition package install: https://gazebosim.org/api/math/6.4/install.html

Project configurations:
[local.conf.txt](https://github.com/ros/meta-ros/files/15097535/local.conf.txt)
[bblayers.conf.txt](https://github.com/ros/meta-ros/files/15097536/bblayers.conf.txt)

When trying to build `turtlebot3`
```
NOTE: Runtime target 'rviz2' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['rviz2']
NOTE: Runtime target 'turtlebot3-bringup' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['turtlebot3-bringup', 'rviz2']
ERROR: Required build target 'turtlebot3' has no buildable providers.
Missing or unbuildable dependency chain was: ['turtlebot3', 'turtlebot3-bringup', 'rviz2']
```

When trying to build `rviz2`
```
ERROR: Nothing PROVIDES 'rviz2'
rviz2 was skipped: Recipe will be skipped because: qt5: depends on qtbase; opengl: depends on rviz-ogre-vendor which depends on mesa which is not available because of missing opengl or vulkan in DISTRO_FEATURES; x11: depends on rviz-rendering which depends on rviz-ogre-vendor which depends on libx11,libxrandr,libxaw which require x11 in DISTRO_FEATURES; ignition: depends on rviz-default-plugins which depends on unavailable ROS_UNRESOLVED_DEP-ignition-math6
```

The problematic recipes are `turtlebot3-navigation2` and `turtlebot3-bringup`. It looks like `gazebo` and `rviz2` fail to build due to some missing dependency about `ignition` libraries on ubuntu 22.04, which can not be resolved by installing the package via https://gazebosim.org/api/math/6.4/install.html

whni, 2024-04-24:
@robwoolley Could you check this issue? Is there any estimated time that the support might arrive? Thank you so much!

robwoolley, 2024-04-25:
Hi @whni,

From a high-level the problem is that the turtlebot3 recipes were only supported for ROS 1 melodic. (according to here: https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions)

That said it should be possible to get them working on ROS 2 as it seems that there is still active work in the past year being done related to Humble (https://github.com/ROBOTIS-GIT/turtlebot3/branches/all)

What I see is that rviz2 is listed as a runtime dependency for turtlebot3_bringup: https://github.com/ROBOTIS-GIT/turtlebot3/blob/ros2-devel/turtlebot3_bringup/package.xml

However, it only seems to be required because of the rviz2 launch file: rviz2.launch.py

I think a quick hack would be to create a bbappend for turtlebot3-bringup that did:
ROS_EXEC_DEPENDS:remove = "rviz2"

This isn't a fix I can add to the tree.  The better alternative would be to either break out rviz2.launch.py into its own package to contain the runtime dependency or make it a build-time configuration option whether to include rviz2 in the image.

Let me know how that goes.  It should help unblock you without needing to wait for ignition and rviz2 to be fixed.

Regards,
Rob

whni, 2024-04-25:
Hi @robwoolley 

As for the whole turtlebot3 support, I think `rviz2` is a must since both `turtlebot3-bringup` and `turtlebot3-navigation2` require it.

Thank you so much for this hack. Now `turtlebot3-bringup` is good to build while I don't have a chance to test it on the real hw. As `turtlebot3-navigation2`, it shows the similar error:
```
ERROR: Nothing PROVIDES 'rviz-default-plugins' (but /home/weihengni/Projects/robotpi/meta-ros/meta-ros2-humble/generated-recipes/slam-toolbox/slam-toolbox_2.6.8-1.bb, /home/weihengni/Projects/robotpi/meta-ros/meta-ros2-humble/generated-recipes/navigation2/nav2-rviz-plugins_1.1.14-1.bb DEPENDS on or otherwise requires it)
rviz-default-plugins was skipped: Recipe will be skipped because: qt5: depends on qtbase; opengl: depends on rviz-common which depends on mesa which is not available because of missing opengl or vulkan in DISTRO_FEATURES; x11: depends on rviz-rendering which depends on rviz-ogre-vendor which depends on libx11,libxrandr,libxaw which require x11 in DISTRO_FEATURES; ignition: depends on unavailable ROS_UNRESOLVED_DEP-ignition-math6
ERROR: Required build target 'turtlebot3-navigation2' has no buildable providers.
Missing or unbuildable dependency chain was: ['turtlebot3-navigation2', 'nav2-bringup', 'slam-toolbox', 'rviz-default-plugins']
```

Other than this, I also found another build error from `ogre_1.12.12.bb` in another dependency chain related to 'rviz-default-plugins':
```
|     /home/weihengni/Projects/robotpi/ros_common_artifacts/BUILD-poky-humble-kirkstone/work/core2-64-poky-linux/ogre/1.12.12-r0/build/imgui-1.79/imgui.cpp
|
|   Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
|   .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
|
|
| CMake Error at Components/Overlay/CMakeLists.txt:42 (add_library):
|   No SOURCES given to target: OgreOverlay
```

In a word, there are at least two dependency errors that we need to fix to enable rviz2 on humble: `rviz2`, `rviz-common`, `rviz-ogre-vendor`, `rviz-rendering` <- `rviz-default-plugins` 
1. <- `ignition-math6-vendor` <- `ROS_UNRESOLVED_DEP-ignition-math6`
2. <- `ogre_1.12.12.bb`

Really appreciate that you could soon fix turtlebot3 for the community!

robwoolley, 2024-04-25:
Hi @whni,

Thanks for reporting back.  Glad that it is successful for you.

It seems that a similar problem exists with nav2_bringup where it has some scripts that use rviz2 at runtime so the whole package has an exec depends on it.  You could use the same hack again for nav2_bringup.

However, based on the instructions here: https://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/ it appears that turtlebot3_navigation is only required on the Ubuntu system in the tutorial.  If you use bitbake to build the individual turtlebot3-* recipes that you need instead of using the turtlebot3 metapackage, you should be able to get what you need.

Regards,
Rob

whni, 2024-04-25:
@robwoolley Yeah, now I am able to work on the hacky build :) Thank you so much for the help.

But anyway, it should still be great that Humble officially supports rviz packages to unblock the top level turtlebot3 recipe, which could help more noob ROS developers, like me, avoid the pain time at the beginning. Also rviz should be very useful for the development of some non-mobile devices with a local display. Here is the list that I've realized:
`rviz2`, `rviz-common`, `rviz-ogre-vendor`, `rviz-rendering`, `rviz-default-plugins`.

Thank you again!

whni, 2024-04-26:
Hi @robwoolley 

Another topic about another common dependency for rviz packages: https://github.com/ros/meta-ros/blob/e4bacc5540c1077cc4ac98947a4c56a65cc33cbb/meta-ros-common/recipes-devtools/ogre/ogre_1.12.12.bb

It's weird that it fails to build with bitbake:
```
| -- Configuring done
| CMake Error at Components/Overlay/CMakeLists.txt:42 (add_library):
|   Cannot find source file:
|
|     /home/weihengni/Projects/robotproject/ros_common_artifacts/BUILD-poky-humble-kirkstone/work/core2-64-poky-linux/ogre/1.12.12-r0/build/imgui-1.79/imgui.cpp
|
|   Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
|   .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
|
|
| CMake Error at Components/Overlay/CMakeLists.txt:42 (add_library):
|   No SOURCES given to target: OgreOverlay
```

I checked the build folder and found the downloaded imgui.tar.gz is 0 size (by Components/Overlay/CMakeLists.txt). However, if I manually clone the repo and build it on my host machine, this file was downloaded correctly. Is there any difference between bitbake system and the host?

Hope this should be a minor fix we can do to enable some more package

whni, 2024-04-26:
After hacking ogre package, I again encountered a weird issue when doing `bitbake rviz-rendering`:
```
| NOTE: VERBOSE=1 cmake --build /home/weihengni/Projects/robotproject/ros_common_artifacts/BUILD-poky-humble-kirkstone/work/core2-64-poky-linux/rviz-rendering/11.2.12-1-r0/build --target all --
| ninja: error: '/home/weihengni/Projects/robotproject/ros_common_artifacts/BUILD-poky-humble-kirkstone/work/core2-64-poky-linux/assimp/5.2.5-r0/recipe-sysroot/usr/lib/libz.so', needed by 'librviz_rendering.so', missing and no known rule to make it
```

Cannot figure out why `librviz_rendering.so` needs a libz.so file from another package `assimp/5.2.5-r0/recipe-sysroot/usr/lib/libz.so`

whni, 2024-04-26:
Hi @robwoolley 

So far I have derived few hacky solution to enabled most recipes for turtlebot3, except those depending on `rviz-default-plugins` (`ignition-math6-vendor` <- `ROS_UNRESOLVED_DEP-ignition-math6`)

Could you please help resolve this last one? Thank you so much!

I also attached my patches here
[turtlebot3_humble_hacks.zip](https://github.com/ros/meta-ros/files/15127034/turtlebot3_humble_hacks.zip)

robwoolley, 2024-04-28:
Hi @whni,

I spent some time this weekend dusting off my old attempt of getting ogre, rviz, and gazebo to build.   I have pushed a preview to a feature branch here: https://github.com/robwoolley/meta-ros/tree/feature/turtlebot3

I was able to successfully build the turtlebot3 recipe for ROS Humble and Yocto Kirkstone for a Raspberry Pi 4 with those changes.   Take note of the commit message which mentions the extra layers you need to add and the additional configuration you need to set in local.conf

I only did build tests of rviz and Ignition/Gazebo.   I tried enabling optional functionality wherever I could, however some work would still be needed to add support for things like CUDA and Optik.  One also might want to choose one of the many desktops available in Yocto (eg. GNOME, GTK3, LxQt, XFCE, etc) and enable GPU accelerators if you want to test rviz and Ignition.

Feedback encouraged.  I expect that there is still some cleanup necessary to make the GUI apps work well for people running the desktop on the robot itself.

Regards,
Rob

whni, 2024-04-28:
@robwoolley Thank you so much for the support! I will test this asap and provide feedback if any.

For CUDA and Optik, I think this is also very important for any robot platform with nvidia gpu for hw acceleration, not only for desktop environment. Could you please also keep us updated on this?

We will keep tracking the humble and develop our own feature on this branch.

Thanks a lot!

whni, 2024-05-05:
Hi @robwoolley how is the status of this pull request?

I am bringing up turtlebot without GUI support. Everything looks good except that `robot_state_publisher` fails to start due to missing urdf_parser_plugin. However, urdf_parser_plugin looks a header ony recipe, not sure how to fix this. Do you have any idea? 
```
[robot_state_publisher-1] terminate called after throwing an instance of 'pluginlib::ClassLoaderException'
[robot_state_publisher-1]   what():  package 'urdf_parser_plugin' not found, searching: [/usr]
[ERROR] [robot_state_publisher-1]: process has died [pid 933, exit code -6, cmd '/usr/lib/robot_state_publisher/robot_state_publisher --ros-args --params-file /tmp/launch_params_vcvbvyxx --params-file /tmp/launch_params_kvj4v1d7'].
```

The dependency chain is as:
[robot-state-publisher_3.0.3-2.bb](https://github.com/ros/meta-ros/blob/kirkstone/meta-ros2-humble/generated-recipes/robot-state-publisher/robot-state-publisher_3.0.3-2.bb) <- [urdf_2.6.0-2.bb](https://github.com/ros/meta-ros/blob/kirkstone/meta-ros2-humble/generated-recipes/urdf/urdf_2.6.0-2.bb) <- [urdf-parser-plugin_2.6.0-2.bb](https://github.com/ros/meta-ros/blob/kirkstone/meta-ros2-humble/generated-recipes/urdf/urdf-parser-plugin_2.6.0-2.bb)

There is an open issue saying the same problem: https://github.com/ros/robot_state_publisher/issues/187

robwoolley, 2024-05-06:
Thanks for reporting the urdf_parser_plugin bug and fix.  It seems that the problem was fixed in later releases, but should definitely be added to ROS_EXEC_DEPENDS variable for the urdf recipe in humble.

whni, 2024-05-06:
@robwoolley Thanks for the update. Now I start to experiment slam function on turtlebot and want to see mapping data on rviz on the remote pc, for which I think slam-toolbox is necessary on the robot machine. I am not able to build slam-toolbox without this commit.

robwoolley, 2024-05-21:
I have posted a new set of patches for ignition/gazebo support here: https://github.com/ros/meta-ros/pull/1150

closed 2025-09-04 by robwoolley