ros/meta-ros #1141: Apriltag patch does not work

mboel, 2024-05-03:
I have an issue with the Apriltag package in the Kirkstone branch. The patch in apriltag_3.4.0-1.bbappend is not valid. 
Removing the patch seems to fix the problem.

robwoolley, 2024-05-04:
Thanks for reporting this.  I tested with Kirkstone + Humble and reproduced the problem.

It looks like apriltag now has this line:
    set(PY_DEST ${CMAKE_INSTALL_PREFIX}/lib/python${Python3_VERSION_MAJOR}.${Python3_VERSION_MINOR}/site-packages/)

Which causes the patch to fail to apply.  I may need to refresh the patch if the hardcoded path

robwoolley, 2024-05-21:
I have submitted a fix for apriltag as part of the patch series I am working on.  You can find the fix here: https://github.com/ros/meta-ros/pull/1150

robwoolley, 2025-09-03:
Thanks, I confirmed that apriltag now builds on kirkstone+humble as of this commit: 32aa79a7cd34f7120896fe6a49e13f57b92a58d6