ros/meta-ros #1151: rosidl linking error

andy9a9, 2024-05-23:
Dear community,
I have a problem with `meta-ros2-iron` compilation. There is a mixing target and native symbols during linking. I was trying to build the `bitbake lifecycle-msgs -c compile` but lot of components are failing due to the same error.
```
build/../meta-ros/meta-ros2-iron/generated-recipes/rcl-interfaces/lifecycle-msgs_1.6.0-2.bb:do_compile
build/../meta-ros/meta-ros2-iron/generated-recipes/rcl-interfaces/rcl-interfaces_1.6.0-2.bb:do_compile
build/../meta-ros/meta-ros2-iron/generated-recipes/common-interfaces/sensor-msgs_5.0.0-2.bb:do_compile
build/../meta-ros/meta-ros2-iron/generated-recipes/common-interfaces/std-srvs_5.0.0-2.bb:do_compile
build/../meta-ros/meta-ros2-iron/generated-recipes/common-interfaces/nav-msgs_5.0.0-2.bb:do_compile
build/../meta-ros/meta-ros2-iron/generated-recipes/common-interfaces/diagnostic-msgs_5.0.0-2.bb:do_compile
build/../meta-ros/meta-ros2-iron/generated-recipes/geometry2/tf2-msgs_0.31.5-1.bb:do_compile
build/../meta-ros/meta-ros2-iron/generated-recipes/rosbag2/rosbag2-interfaces_0.22.3-1.bb:do_compile
```

```
FAILED: liblifecycle_msgs__rosidl_generator_c.so 
: && /home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-gcc -fPIC -mcpu=cortex-a53 -march=armv8-a+crc+crypto -mbranch-protection=standard  --sysroot=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/git=/usr/src/debug/lifecycle-msgs/1.6.0-2-r0  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/git=/usr/src/debug/lifecycle-msgs/1.6.0-2-r0  -fmacro-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/build=/usr/src/debug/lifecycle-msgs/1.6.0-2-r0  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/build=/usr/src/debug/lifecycle-msgs/1.6.0-2-r0  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot=  -fmacro-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot=  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot-native=  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/git=/usr/src/debug/lifecycle-msgs/1.6.0-2-r0  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/git=/usr/src/debug/lifecycle-msgs/1.6.0-2-r0  -fmacro-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/build=/usr/src/debug/lifecycle-msgs/1.6.0-2-r0  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/build=/usr/src/debug/lifecycle-msgs/1.6.0-2-r0  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot=  -fmacro-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot=  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot-native= -shared -Wl,-soname,liblifecycle_msgs__rosidl_generator_c.so -o liblifecycle_msgs__rosidl_generator_c.so CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/state__description.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/state__functions.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/state__type_support.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/transition__description.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/transition__functions.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/transition__type_support.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/transition_description__description.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/transition_description__functions.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/transition_description__type_support.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/transition_event__description.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/transition_event__functions.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/msg/detail/transition_event__type_support.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/change_state__description.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/change_state__functions.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/change_state__type_support.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/get_available_states__description.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/get_available_states__functions.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/get_available_states__type_support.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/get_available_transitions__description.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/get_available_transitions__functions.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/get_available_transitions__type_support.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/get_state__description.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/get_state__functions.c.o CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/rosidl_generator_c/lifecycle_msgs/srv/detail/get_state__type_support.c.o  -Wl,-rpath,/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot-native/usr/lib:  /home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot-native/usr/lib/libservice_msgs__rosidl_generator_c.so  /home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot-native/usr/lib/libbuiltin_interfaces__rosidl_generator_c.so  /home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot/usr/lib/librosidl_runtime_c.so  /home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot/usr/lib/librcutils.so  -ldl && :
/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/../../libexec/aarch64-oe-linux/gcc/aarch64-oe-linux/12.3.0/ld: /home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot-native/usr/lib/libservice_msgs__rosidl_generator_c.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.
```
My good feeling is, that it's trying to link the `native` library with `target`. I was thinking about the problem with `service-msgs` or `rosidl`. So I updated the recipes under `rcl-interfaces` to versions `2.0.2.-2` and `rosidl` to `4.6.0-2`, but it didn't help.

The setup is use the `meta-ros2-iron` from `mickledore` branch (because all other layers have this revision) and build the ros components. But it it's not working... :(.

What's the real issue and how to fix it?
Thanks

andy9a9, 2024-05-23:
ok, I identify the issue. When I looked into `CMakeCache.txt` there was a generated line:
```
//Path to a library.
_lib:FILEPATH=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/lifecycle-msgs/1.6.0-2-r0/recipe-sysroot-native/usr/lib/libservice_msgs__rosidl_typesupport_fastrtps_cpp.so
```
If we look deeper into problem, the path is pointing to `recipe-sysroot-native/usr/lib/` because the `target` doesn't exist in `recipe-sysroot/usr/lib/`. Not sure what's the right way to fix the transition dependencies, while the recipes are generated. I see there a quick fix to add `ROS_BUILD_DEPENDS += "service-msgs"` into target recipes.

robwoolley, 2024-05-23:
Hi,

This was fixed in the other branches by adding service-msgs as a target recipe so the link would choose it over service-msgs-native: https://github.com/ros/meta-ros/commit/4fa5442ff905cf371d717f868634d23714e8c28d

This commit is not available in Mickledore + Iron as that is not a supported combination: https://github.com/ros/meta-ros?tab=readme-ov-file#supported-combinations  (Unfortunately, Mickledore went end-of-life on November 2023).

I am providing "best effort" support for Mickledore + Humble, if that appeals to you.  I am also open to merging fixes to meta-ros2-iron in Mickledore if someone volunteers to maintain it as "best-effort" or "contrib".

Regards,
Rob

andy9a9, 2024-05-23:
Ok, I see, thanks for the hints. Now I fixed the similar issues. But now I found an another one, but it's for `rosbag2-storage-mcap` component. Upgrading to `0.22.6-1` didn't help.

```
/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-g++ -DROSBAG2_STORAGE_MCAP_BUILDING_DLL -DROSBAG2_STORAGE_MCAP_BUILDING_LIBRARY -DROSBAG2_STORAGE_MCAP_HAS_SET_READ_ORDER -DROSBAG2_STORAGE_MCAP_HAS_STORAGE_FILTER_TOPIC_REGEX -DROSBAG2_STORAGE_MCAP_HAS_STORAGE_OPTIONS -DROSBAG2_STORAGE_MCAP_HAS_UPDATE_METADATA -DROSBAG2_STORAGE_MCAP_HAS_YAML_HPP -DROSBAG2_STORAGE_MCAP_OVERRIDE_SEEK_METHOD -DROSBAG2_STORAGE_MCAP_WRITER_CREATES_DIRECTORY -Drosbag2_storage_mcap_EXPORTS -I/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git/include -I/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot/usr/include/mcap_vendor -I/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot/usr/include/pluginlib -I/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot/usr/include/ament_index_cpp -I/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot/usr/include/class_loader -I/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot/usr/include/rcpputils -I/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot/usr/include/rcutils -I/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot/usr/include/rosbag2_storage -mcpu=cortex-a53 -march=armv8-a+crc+crypto -mbranch-protection=standard  --sysroot=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git=/usr/src/debug/rosbag2-storage-mcap/0.22.6-1-r0  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git=/usr/src/debug/rosbag2-storage-mcap/0.22.6-1-r0  -fmacro-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/build=/usr/src/debug/rosbag2-storage-mcap/0.22.6-1-r0  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/build=/usr/src/debug/rosbag2-storage-mcap/0.22.6-1-r0  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot=  -fmacro-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot=  -fdebug-prefix-map=/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden -DNDEBUG -fPIC -Wall -Wextra -Wpedantic -std=gnu++17 -MD -MT CMakeFiles/rosbag2_storage_mcap.dir/src/mcap_storage.cpp.o -MF CMakeFiles/rosbag2_storage_mcap.dir/src/mcap_storage.cpp.o.d -o CMakeFiles/rosbag2_storage_mcap.dir/src/mcap_storage.cpp.o -c /home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git/src/mcap_storage.cpp
/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git/src/mcap_storage.cpp:273:23: error: 'RecordOffset' is not a member of 'mcap'; did you mean 'EndOffset'?
  273 |   std::optional<mcap::RecordOffset> last_read_message_offset_;
      |                       ^~~~~~~~~~~~
      |                       EndOffset
/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git/src/mcap_storage.cpp:273:35: error: template argument 1 is invalid
  273 |   std::optional<mcap::RecordOffset> last_read_message_offset_;
      |                                   ^
/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git/src/mcap_storage.cpp:274:23: error: 'RecordOffset' is not a member of 'mcap'; did you mean 'EndOffset'?
  274 |   std::optional<mcap::RecordOffset> last_enqueued_message_offset_;
      |                       ^~~~~~~~~~~~
      |                       EndOffset
/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git/src/mcap_storage.cpp:274:35: error: template argument 1 is invalid
  274 |   std::optional<mcap::RecordOffset> last_enqueued_message_offset_;
      |                                   ^
/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git/src/mcap_storage.cpp: In member function 'bool rosbag2_storage_plugins::MCAPStorage::read_and_enqueue_message()':
/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git/src/mcap_storage.cpp:482:47: error: 'const struct mcap::MessageView' has no member named 'messageOffset'
  482 |   last_enqueued_message_offset_ = messageView.messageOffset;
      |                                               ^~~~~~~~~~~~~
/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git/src/mcap_storage.cpp: In member function 'bool rosbag2_storage_plugins::MCAPStorage::enqueued_message_is_already_read()':
/home/projects/build/tmp/work/cortexa53-crypto-oe-linux/rosbag2-storage-mcap/0.22.6-1-r0/git/src/mcap_storage.cpp:545:33: error: no match for 'operator==' (operand types are 'int' and 'const std::nullopt_t')
  545 |   if (last_read_message_offset_ == std::nullopt) {
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~ ^~ ~~~~~~~~~~~~
      |       |                                 |
      |       int                               const std::nullopt_t
```

closed 2024-05-24 by andy9a9