ros/meta-ros #1152: ros2 bag record with mcap does not work, meta-ros2-humble, kirkstone

mwest90, 2024-05-23:
**Describe the bug**
I'm trying to use ros2 bag record with mcap as storage backend.

**To Reproduce**
I'm on a meta-ros2-humble, kirkstone, this commit:
https://github.com/ros/meta-ros/tree/cefbb613078031a5c59f7716d52f91328c1ee0c2

On my target I'm running two terminals, one to publish a message and another to try to record it:
`ros2 topic pub /example_topic std_msgs/msg/String "{data: 'Hello, ROS 2'}"`
`ros2 bag record /example_topic -s mcap`

The record command throws the following error:
`Unable to load plugin 'mcap': Failed to load library /usr/lib/librosbag2_storage_mcap.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /usr/lib/libmcap.so: undefined symbol: ZSTD_CCtx_setParameter, at /usr/src/debug/rcutils/5.1.3-1-r0/git/src/shared_library.c:99`

**Expected behavior**
Expecting no errors and a actual mscap formatted rosbag.

robwoolley, 2024-05-24:
Thanks for filing this issue and providing details on the problem.  Based on what you described, I suspect that it is related to this: https://github.com/ros/meta-ros/commit/4e0f9ea43d304660b0020f640a0c5fbf7d37ba88

I had to remove the zstd-vendor meta-package because it was empty.  I suspect that it is necessary to also add:
ROS_EXEC_DEPENDS:append = " zstd"
or simple replace the ROS_EXEC_DEPENDS:remove line with ALLOW_EMPTY="1" to ensure that the meta-package is created with an automatic dependency on zstd.

mwest90, 2024-05-24:
@robwoolley thanks for the answer!

However I tried both of these approaches now and non of them fixes the error.

mwest90, 2024-06-05:
@robwoolley I had to update the patch to make this work, the following patch file and bbappend file solves the issues:
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 6de89d205..c849fbc8e 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -2,6 +2,7 @@ cmake_minimum_required(VERSION 3.5)
 project(mcap_vendor LANGUAGES C CXX ASM)
 
 ## Dependencies
+set(zstd_SOURCE_DIR ${CMAKE_CURRENT_SOURCE_DIR}/zstd)
 find_package(ament_cmake REQUIRED)
 find_package(zstd_vendor REQUIRED)
 find_package(zstd REQUIRED)
@@ -30,19 +31,8 @@ endif()
 
 ## Define vendor macro
 macro(build_mcap_vendor)
-  include(FetchContent)
-  fetchcontent_declare(mcap
-    URL https://github.com/foxglove/mcap/archive/refs/tags/releases/cpp/v0.8.0.tar.gz
-    URL_HASH SHA1=b44637791da2c9c1cec61a3ba6994f1ef63a228c # v0.8.0
-  )
-  fetchcontent_makeavailable(mcap)
-
-  fetchcontent_declare(lz4
-    GIT_REPOSITORY https://github.com/lz4/lz4.git
-    GIT_TAG d44371841a2f1728a3f36839fd4b7e872d0927d3 # v1.9.3
-  )
-  fetchcontent_makeavailable(lz4)
-
+  set(mcap_SOURCE_DIR ${CMAKE_CURRENT_SOURCE_DIR}/mcap)
+  set(lz4_SOURCE_DIR ${CMAKE_CURRENT_SOURCE_DIR}/lz4)
   file(GLOB _lz4_srcs
     ${lz4_SOURCE_DIR}/lib/*.c)
```

```

# Use Bitbake to fetch https://github.com/foxglove/mcap.git
FILESEXTRAPATHS:prepend := "${THISDIR}/${BPN}:"
#SRC_URI += "file://0001-CMakeLists.txt-fetch-uncrustify-with-bitbake-fetcher.patch"
SRC_URI += " \
    git://github.com/foxglove/mcap.git;protocol=https;name=mcap;destsuffix=git/mcap;branch=main;lfs=0 \
    git://github.com/lz4/lz4.git;protocol=https;name=lz4;destsuffix=git/lz4;branch=release \
    git://github.com/facebook/zstd.git;protocol=https;name=zstd;destsuffix=git/zstd;branch=release \
    file://0001-CMakeLists.txt-fetch-dependencies-with-bitbake-fetch.patch \
"

# releases/cpp/v0.8.0
SRCREV_mcap = "801c4ae3f34b23e9a27eb34b88ab7a0180d4b40f"
# v1.9.3
SRCREV_lz4 = "d44371841a2f1728a3f36839fd4b7e872d0927d3"
# v1.5.2
SRCREV_zstd = "e47e674cd09583ff0503f0f6defd6d23d8b718d3"
```

robwoolley, 2024-06-10:
Thanks!  I will try to get this merged in as soon as possible.

Bentis, 2025-04-09:
Encountered same issue and verified that mwest90's fix solves it.