ros/meta-ros #1161: Cannot login jetson-orin-nano-devkit ros-image-core-humble

jinchenglee, 2024-05-28:
**Describe the bug**
As title explains, cannot login the flashed ros-image-core-humble image on jetson-orin-nano-devkit. 

HDMI has no image after the NVIDIA boot screen expires (which picks up the burned SD image). In the meantime, on the connected debug serial port, I can see below:

```
...
         Starting User Login Management...                                     
[  OK  ] Finished IPv4 Packet Filtering Framework.
[  OK  ] Reached target Preparation for Network.
[  OK  ] Reached target Hardware activated USB gadget.
         Starting Network Configuration...
[  OK  ] Started D-Bus System Message Bus.
[  OK  ] Started Network Configuration.
         Starting Connection service...
[  OK  ] Started User Login Management.
[  OK  ] Started Connection service.
[  OK  ] Reached target Multi-User System.
[  OK  ] Reached target Network.
         Starting Record Runlevel Change in UTMP...
[  OK  ] Finished Record Runlevel Change in UTMP.
[   12.828633] audit: type=1334 audit(1651167746.968:6): prog-id=9 op=LOAD
[   12.828864] audit: type=1334 audit(1651167746.968:7): prog-id=10 op=LOAD

Robot Operating System 2 (ROS 2) with OpenEmbedded Linux 2.humble.2024-04-18.3-0 jetson-o0

jetson-orin-nano-devkit login: 
CTRL-A Z for help | 115200 8N1 | NOR | Minicom 2.7.1 | VT102 | Offline | ttyUSB0  
```
This means the image has successfully bootstraped. However, neither on host machine where serial debug port connected to /dev/ttyUSB0, nor on usb keyboard/mouse plugged into the device can type in anything. Just no response. 

**To Reproduce**

Host: 
```
:/work$ lsb_release -a
LSB Version:	core-11.1.0ubuntu2-noarch:printing-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal
```

Build repository setup:
```
:/work/yocto-meta-ros/meta-ros$ pwd
/work/yocto-meta-ros/meta-ros
:/work/yocto-meta-ros/meta-ros$ git log
commit 09ed8cab9e654aaec409126119c7fa4a0bdbe488 (HEAD -> build, origin/build-next, origin/build)
Author: mark-bchoi <afewgoodman.choi@gmail.com>
Date:   Wed May 15 10:47:08 2024 -0400

    kas: jetson-orin-nano-devkit: Change machine name
    
    Signed-off-by: Rob Woolley <rob.woolley@windriver.com>

```

Build command:
```
cd /work/yocto-meta-ros
kas build meta-ros/kas/oeros-kirkstone-humble-jetson-orin-nano-devkit.yml 
sudo ./deploy.sh ros-image-core-humble jetson-orin-nano-devkit
```

Deploy.sh script:
```
#!/bin/bash

image=$1
machine=$2

scriptdir="$( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null && pwd )"
deployfile=${image}-${machine}.tegraflash.tar.gz
tmpdir=`mktemp`

rm -rf $tmpdir
mkdir -p $tmpdir
echo "Using temp directory $tmpdir"
pushd $tmpdir
cp $scriptdir/build/tmp-glibc/deploy/images/${machine}/$deployfile .
tar -xvf $deployfile
set -e
sudo ./doflash.sh
popd
echo "Removing temp directory $tmpdir"
rm -rf $tmpdir
```

jinchenglee, 2024-05-28:
BTW, I've also tried jetson-orin-nano-devkit-nvme image with local added oeros-kirkstone-humble-jetson-orin-nano-devkit-nvme.yml file, the image could be generated successfully and I can flash the image into the installed nvme as well. However, it'll boot in a loop and never bootstrap to login prompt as above burned SD image. 

Anyways, this is another issue, probably worth another issue ticket. Let's focus on SD image first.

bchoineubility, 2024-05-29:
Hi @jinchenglee 

You can refer to tegra-demo distro. actually, ros-image-core image would missed such functionalities. for example I also gone  through ownership under /dev/* was root, it affected playing gstreamer before I apply something under tegra-demo. 

You can also add package-group for ros-image-core to tegra-demo in advance.

BR,
Mark

jinchenglee, 2024-05-30:
@bchoineubility , thanks for prompt response. But since there is no hdmi output nor can user type in via serial console, how do you usually login into the device? Alternatively, how do you use such system usually?

robwoolley, 2024-06-10:
Hi @jinchenglee.  Unfortunately, I don't have an Nvidia Jetson Orin Nano to test this on.  I suspect that as @bchoineubility suggested, there may be additional configuration options  in tegra-demo that need to be enabled in order for input to work.

My suggestion would be to check the value of console= on the kernel parameters.   If there are 2 options then only the second console value will accept input, the other will be read-only.

The other alternative would be to go talk to the OE4T / meta-tegra folks to see if they have an idea of what the problem could be.

closed 2024-06-10 by jinchenglee