ros/meta-ros #1167: Issue with nav2-mppi-controller

avanmalleghem, 2024-06-04:
**Describe the bug**

When using nav2-mppi-controller with the current version of xtl (0.7.7), xtensor (0.24.7) and xsimd (11.2.0) dependencies, it has a weird behavior. 
The behavior is described here : https://github.com/ros-navigation/navigation2/issues/4380 (you even have a video)

I tried to use xtl 0.7.2, xtensor 0.23.10 and xsimd 7.6.0 (versions used on ubuntu 22 when installed from binaries) and it solves the issue so it is clearly linked to one of these libraries. I didn't have time to go deeper in it for now but I guess next steps are : 
- Find recipes that have dependencies on these libraries.
- Find the recipe causing issue among these 3 and downgrade it to the right commit if it has no regression

Maybe I will have time to do so in next weeks but I'm not sure so if someone wants to help on this one...

**To Reproduce**
I'm on the latest kirkstone version (commit 9a857afebc18e0f426d87226bd085727ea9ce4ae) on a jetson nano (using meta-tegra) but I don't think it is machine dependant. I think you can reproduce it whatever the machine is.

- Version : kirkstone
- ROS distro : humble

robwoolley, 2024-06-26:
Thanks for reporting the issue.  I have been tracking the progress in the upstream bug: https://github.com/ros-navigation/navigation2/issues/4380

I haven't been able to investigate it yet as I have been working to close out the work on ros-image-world.  I hope to resolve that soon and get a chance to look into this.

robwoolley, 2025-01-21:
@avanmalleghem I saw your LinkedIn post that this was fixed upstream.  I notice that there is a comment that backporting it to Humble isn't feasible due to massive API changes: https://github.com/ros-navigation/navigation2/issues/4380#issuecomment-2594134802

What would be a good next step for this issue?  Would you like to change the targeted ROS distro to jazzy or rolling?  Should we just close it as Humble support isn't forthcoming?  Or would you prefer to leave it open as unresolved?

avanmalleghem, 2025-01-27:
@robwoolley it isn't available on jazzy so... Only for rolling I would say.
If you want to make it work for ROS Humble, you must use other versions for xtl, xtensor and xsimd (as suggested above)
As you prefer for the ticket status. Don't hesitate if I can help