ros/meta-ros #1173: Missing license information about code inherited from elsewhere

priv-kweihmann, 2024-06-16:
Current layer claims that everything is licensed under MIT and copyrighted by
Copyright (c) 2012, BMW Car IT GmbH
Copyright (c) 2018-2020 LG Electronics, Inc.

That is not the case as over the past recipes and files from other layers, with non MIT licenses were inherited
(or not to say blindly copied over with sometimes stripping an author and license information).

I don't see that being documented anywhere here

robwoolley, 2024-06-22:
Thanks for pointing this out.  I will take appropriation action.

Rob

priv-kweihmann, 2024-10-03:
@robwoolley any news here?
What's the outcome of the discussion (I remember commenting on the minutes but never saw some feedback)?

robwoolley, 2024-11-21:
A new recipe for cppcheck was created to support ament-cppcheck.   This seems to be the best fit for supporting ROS.

closed 2024-11-21 by robwoolley