ros/meta-ros #1174: ROS1 Noetic and Yocto Scarthgaph with Python 3.12

krisklau, 2024-06-19:
**Describe the bug**
Incompatibility with Python 3.12

```
roscore
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/roscore", line 37, in <module>
    from roslaunch.nodeprocess import DEFAULT_TIMEOUT_SIGINT, DEFAULT_TIMEOUT_SIGTERM
  File "/opt/ros/noetic/lib/python3.12/site-packages/roslaunch/__init__.py", line 57, in <module>
    from .launch import ROSLaunchRunner
  File "/opt/ros/noetic/lib/python3.12/site-packages/roslaunch/launch.py", line 58, in <module>
    from roslaunch.rlutil import update_terminal_name
  File "/opt/ros/noetic/lib/python3.12/site-packages/roslaunch/rlutil.py", line 47, in <module>
    import rosclean
  File "/opt/ros/noetic/lib/python3.12/site-packages/rosclean/__init__.py", line 40, in <module>
    from distutils.spawn import find_executable
ModuleNotFoundError: No module named 'distutils'

```

**To Reproduce**
Build an image with Ros1 noetic in Scarthgap. This will include python 3.12. This version of python deprecates distutils, which causes problems for some packages. `rosclean` in the above example. Simply boot the image and run `roscore`. 

Upon patching for `distutils`, we hit this error as well: https://github.com/ros/ros_comm/pull/2297

**Expected behavior**
We can patch these package by package, but I am puzzled as to why this has not been reported before. Has the combination of Scarthgap and Noetic been tested?

krisklau, 2024-06-20:
Note we will supply patches for the things we encounter, I just wanted to make sure nothing was missing from my configuration.

poett1, 2024-09-12:
@krisklau I just started working on ROS Noetic + Scarthgap and did not face this error. However, I had an infinite loop at [findCaller() from roslogging.py](https://github.com/ros/ros_comm/blob/noetic-devel/tools/rosgraph/src/rosgraph/roslogging.py#L53) on every ros command (roscore, rosrun, roslaunch), which could be solved by [this comment](https://github.com/ros/ros_comm/issues/2352#issuecomment-2287231681).
I'm wondering if this is tested at all, since this issue is due to Python 3.12 (which is the default for Scarthgap).