ros/meta-ros #1176: Installing error for clearpath-platform

bchoineubility, 2024-06-25:
Hi,

I have an issue to building clearpath-platform.

```
| install: cannot stat '/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/clearpath-platform/0.2.7-1-r0/build/liblighting_lib.so': No such file or directory
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/bchoi/nvidia-yocto-bsp/sources/meta-ros/meta-ros2-humble/generated-recipes/clearpath-common/clearpath-platform_0.2.7-1.bb:do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4118 tasks of which 4117 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/bchoi/nvidia-yocto-bsp/sources/meta-ros/meta-ros2-humble/generated-recipes/clearpath-common/clearpath-platform_0.2.7-1.bb:do_install
Summary: There was 1 ERROR message, returning a non-zero exit code.
```

it is because of bbappend file on the meta-ros2-humble/recipes-bbappends/clearpath-common/clearpath-platform_0.2.7-1.bbappend which try to install liblighting_lib.so.  
```
   do_install:append() {
       install -d ${D}${libdir}
       install -D -m 0644 ${B}/liblighting_lib.so ${D}/${libdir}
   }
```
By the way, I think lighting_node do not link it any more in the CMakeLists.txt
```
 34 # Lighting
 35 set(LIGHTING_EXECUTABLE lighting_node)
 36 set(LIGHTING_LIB clearpath_platform_lighting)
 37 set(LIGHTING_DEPENDENCIES
 38     clearpath_platform_msgs
 39     sensor_msgs
 40     rclcpp
 41 )
 42
 43 add_library(
 44   ${LIGHTING_LIB}
 45   SHARED
 46   src/lighting/lighting.cpp
 47   src/lighting/sequence.cpp
 48   src/lighting/color.cpp
 49 )
 50
 51 ament_target_dependencies(${LIGHTING_LIB} ${LIGHTING_DEPENDENCIES})
 52
 53 target_include_directories(
 54   ${LIGHTING_LIB}
 55   PRIVATE
 56   include
 57 )
 58
 59 add_executable(${LIGHTING_EXECUTABLE} src/lighting/main.cpp)
 60 target_link_libraries(${LIGHTING_EXECUTABLE} ${LIGHTING_LIB})
```

So, I think we do not need to bbappend file to install manually for lighting_node.

Can you please check it?

BR,
Mark

robwoolley, 2024-06-26:
Thanks for reporting this, Mark.  I see you are using Humble, which Yocto release are you using?

bchoineubility, 2024-06-26:
Yes I am using humble now.  the commit id is `9a857afebc18e0f426d87226bd085727ea9ce4ae` .