ros/meta-ros #1195: Rlease plan

andy9a9, 2024-07-15:
Dear community!

I would like to ask you a question about release plan. On the https://docs.ros.org/en/rolling/Releases.html I can see, that [Jazzy](https://docs.ros.org/en/rolling/Releases/Release-Jazzy-Jalisco.html) has EOL set to 2029, which seems to me the LTS version.

If I look at the [Release table](https://github.com/ros/meta-ros/blob/master/README.md#supported-combinations) I don't see such a combination. Yocto's `Scarthgap` branch is a LTS and I would like to use it with `Jazzy` I see here a `meta-ros2-jazzy` layer but not in a table. So is it possible to use such a combination? Would it be possible to update the table and mark a `Jazzy` as a LTS as well?

Thank you

robwoolley, 2024-07-17:
Hi @andy9a9,

Yes, it is possible to use meta-ros2-jazzy with Scarthgap, Kirkstone, and with Styhead (ie "master").  You can see the kas configurations here: https://github.com/ros/meta-ros/tree/build/kas

I will update the table to reflect that soon.   Most of my testing lately has been on Rolling while I have been working to set up CI/CD builds.  If you encounter any problems with Jazzy let me know and I can backport any fixes you need.

Regards,
Rob

andy9a9, 2024-07-23:
Hello @robwoolley ,

That's a great news, thanks!

- I tried to migrate my layer the `iron` branch to `jazzy`, here are my changes which I had to do (I'm not at the end yet):
  - missing commit https://github.com/ros/meta-ros/commit/ea5be5a69008ad12628b7a224afa958b410e0a03
  - extend some recipes for `native` variant
    -  `rqt-runtime-monitor` -> `python3-pyqt5-sip` -> `python3-pyqt5` -> `qtquickcontrols2`
      - this is little bit funny, while `python3-pyqt5-native` recipe doesn't exist since `kirkstone`
  - unable to build `liblz4-vendor`
    - stacked at the failing chain `ros-base`, `rosbag2`, `rosbag2-storage-default-plugins`, `rosbag2-storage-mcap`, `mcap-vendor`, `liblz4-vendor`, `ROS_UNRESOLVED_DEP-liblz4-dev`

More info will come once I move forward...

robwoolley, 2024-07-23:
Hi @andy9a9 ,

The sip tool jumped several major releases between kirkstone and scarthgap.

Some major fixes were required.  I need to upstream them to oe-core and meta-qt5.

Details can be found in the working group minutes: https://docs.google.com/document/d/1LqUjcu6vdlqVJO62SreCyjzddNDZhfO2n-7qYghY_cQ/edit?usp=drivesdk

Let me know if there's anything I can help with.

Regards,
Rob

andy9a9, 2024-07-23:
Good to know :).
But have can I proceed with the `liblz4-vendor` error? I see there an options:
- remove depends `liblz4-vendor` from `mcap-vendor`
- create something like `SKIP_RECIPE[liblz4-vendor] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'liblz4-vendor', 'Depends on unavailable ROS_UNRESOLVED_DEP-liblz4-dev', '', d)}"`
- downgrade `mcap-vendor` to get rid of `liblz4-vendor` dependency just for now
- or ?

robwoolley, 2024-07-25:
As a temporary measure, I set the variable in ros-distro.inc (in master-next meta-ros2-rolling) ROS_UNRESOLVED_DEP-liblz4 = "lz4"

The correct long-term fix is to add an entry for OpenEmbedded under liblz4 in the rosdep metadata files.

andy9a9, 2024-07-26:
Hi @robwoolley ,

thanks for `ROS_UNRESOLVED_DEP-liblz4 = "lz4"` hint. I made a little progress here:
- some packages are failing on `[buildpaths`] in final
  - there is a class `ros2_distro.bbclass` which is dealing with that, but not fully, while is not handling paths under `ros_prefix`
    - I suggest to extend the class a little bit like:
      - ```patch
        +ROS_PREFIX_PATH ?= "${datadir}"
          do_install:append:class-target() {
        -    if [ -d ${D}${datadir}/ament_index/resource_index/parent_prefix_path ]; then
        -        find ${D}${datadir}/ament_index/resource_index/parent_prefix_path -type f \
        +    if [ -d ${D}${ROS_PREFIX_PATH}/ament_index/resource_index/parent_prefix_path ]; then
        +        find ${D}${ROS_PREFIX_PATH}/ament_index/resource_index/parent_prefix_path -type f \
                         | xargs --no-run-if-empty sed -i -e 's@${STAGING_DIR_HOST}/@/@g' -e 's@${STAGING_DIR_NATIVE}/@/@g'
              else
                    true
        ```
    - default behavior is untouched, but recipes could override the path via `ROS_PREFIX_PATH = `"${ros_datadir}"`
      - required changes
        - `eigen3-cmake-module`
        - `shared-queues-vendor`
- `shared-queues-vendor_0.26.1-2`
  - this component will never be build from `dc17fed08353f30a7e20676df7c8fc5e842ed011`
    - previous branch `release/jazzy/shared_queues_vendor` doesn't exist anymore and `release/rolling/shared_queues_vendor/0.24.0-1` is not a tag
      - so `ROS_BRANCH = "nobranch=1"` has to be set
- `orocos-kdl-vendor_0.5.0-3`
  - same problem with building as with `shared-queues-vendor_0.26.1-2`
    - `nobranch`
  - `0001-CMakeLists.txt-use-system-orocos-kdl.patch`
    - never used
  - `0001-CMakeLists.txt-fetch-orocos-kdl-with-bitbake-fetcher.patch`
    - can't be applied
  - requires `python-orocos-kdl` as a build-dep, but this component requires `orocos-kdl-vendor`
    - circular loop
    - has to be fixed somehow!
- `yaml-cpp-vendor_9.0.0-2`
  - requires similar changes as here `https://github.com/ros/meta-ros/commit/ad82ae26c7d098ab8d9bccf59de971e089e862f2` but for different version
    - ```patch
       @@ -9,26 +9,18 @@ ament_add_default_options()
       # NO_CMAKE_PACKAGE_REGISTRY used to avoid finding the library downloaded in WORKSPACE B
       # when building workspace A.
       # This should only find a system installed yaml-cpp and thus the environment hook isn't needed.
      -find_package(yaml-cpp 0.8.0 QUIET NO_CMAKE_PACKAGE_REGISTRY)
      -
      +find_package(yaml-cpp QUIET NO_CMAKE_PACKAGE_REGISTRY)
      +if(yaml-cpp_FOUND)
      +  message(STATUS "Found yaml-cpp ${yaml-cpp_VERSION}")
      +else()
      +  message(FATAL_ERROR "yaml-cpp not found -- missing from DEPENDS?")
      +endif()
      + 
       set(YAML_CPP_CXX_FLAGS "${CMAKE_CXX_FLAGS}")
       if(MSVC)
         set(YAML_CPP_CXX_FLAGS "/wd4251 /wd4275 ${YAML_CPP_CXX_FLAGS}")
       endif()
       
      -ament_vendor(yaml_cpp_vendor
      -  SATISFIED ${yaml-cpp_FOUND}
      -  VCS_URL https://github.com/jbeder/yaml-cpp.git
      -  VCS_VERSION 0.8.0
      -  GLOBAL_HOOK
      -  CMAKE_ARGS
      -    "-DCMAKE_CXX_FLAGS=${YAML_CPP_CXX_FLAGS}"
      -    -DYAML_CPP_BUILD_CONTRIB:BOOL=OFF
      -    -DYAML_CPP_BUILD_TESTS:BOOL=OFF
      -    -DYAML_CPP_BUILD_TOOLS:BOOL=OFF
      -    -DYAML_BUILD_SHARED_LIBS:BOOL=${BUILD_SHARED_LIBS}
      -)
      -
       ament_package(
         CONFIG_EXTRAS "yaml_cpp_vendor-extras.cmake.in"
       )
      ```
I'm still not at the end :)

andy9a9, 2024-07-29:
Hello again,

so I went into end of compilation. Here are the latest changes which I had to made:
- `python-qt-binding`
  - ```bitbake
    # python3-pyqt5-native dependency which wasn't needed
    DEPENDS:remove = "sip3-native python3-pyqt5-native"
    DEPENDS:append = "sip-native "
    ```
- `ros2_distro.bbclass`
  - this https://github.com/ros/meta-ros/commit/90d09961ad890fb9c2cb315c9a0b8d792e10e535 change the prefix, but didn't change everything
  - ```patch
     do_install:append:class-target() {
    -    if [ -d ${D}${ros_datadir}/ament_index/resource_index/parent_prefix_path ]; then
    -        find ${D}${ros_datadir}/ament_index/resource_index/parent_prefix_path -type f \
    +    if [ -d ${D}${ROS_PREFIX_PATH}/ament_index/resource_index/parent_prefix_path ]; then
    +        find ${D}${ROS_PREFIX_PATH}/ament_index/resource_index/parent_prefix_path -type f \
                 | xargs --no-run-if-empty sed -i -e 's@${STAGING_DIR_HOST}/@/@g' -e 's@${STAGING_DIR_NATIVE}/@/@g'
     else
         true
    ```
- `ament-cmake-gen-version-h`
  - reflect changes from https://github.com/ament/ament_cmake/pull/526
    - ``` patch
      diff --git a/cmake/ament_generate_version_header.cmake b/cmake/ament_generate_version_header.cmake
      index d9470b40..5d2a2795 100644
      --- a/cmake/ament_generate_version_header.cmake
      +++ b/cmake/ament_generate_version_header.cmake
      @@ -147,7 +147,7 @@ function(ament_generate_version_header target)
             "${SCRIPT_TEMPLATE_FILE}"
           COMMENT "Generating ${ARG_HEADER_PATH}")

      -  add_custom_target("ament_generate_version_header__${target}"
      +  add_custom_target("ament_generate_version_header__${target}" ALL
           DEPENDS "${GENERATED_HEADER_FILE}")
         add_dependencies("${target}" "ament_generate_version_header__${target}")
      ```
- `control-toolbox`
  - `ROS_BUILDTOOL_DEPENDS += "generate-parameter-library-py-native"`
- `orocos-kdl-vendor`
  - `f1bbaed1b5190dbfefd08ca445e53d20d1a0066e` has a problem with compilation, switching back to original version
    - `SRCREV_release = "92eb6e580958a6834df2c1c1ecee92cdbac2f363"`
  - refresh `0001-CMakeLists.txt-fetch-orocos-kdl-with-bitbake-fetcher`
  - ``` bitbake
    # remove ros_prefix from pkconfig files
    do_install:append:class-target() {
        for f in ${D}${ros_libdir}/pkgconfig/* ; do
            sed -i $f -e 's@${STAGING_DIR_HOST}${includedir}@\$\{includedir}@g' -e 's@${STAGING_DIR_NATIVE}${includedir}@\$\{includedir}@g'
        done
    }
    ```
- create/apply `disable-compiler-warnings` patch
  - `controller-interface`
  - `controller-manager`
  - `hardware-interface`
  - `forward-command-controller`
  - `imu-sensor-broadcaster`
  - `joint-state-broadcaster`
  - `joint-trajectory-controller`
  - `pid-controller`
    - `ROS_BUILDTOOL_DEPENDS += "generate-parameter-library-py-native"`
  - `position-controllers`
  - `range-sensor-broadcaster`
  - `velocity-controllers`
- `rosbag2-compression-zstd`
  - refresh `0001-CMakeLists.txt-drop-dependency-on-zstd_vendor`
- `rsl`
  - `ROS_BUILDTOOL_DEPENDS += "rosidl-default-runtime-native"`

Basically that it! :)

andy9a9, 2024-09-08:
@robwoolley Are there any news? :)

robwoolley, 2026-11-02:
Sorry, I think I lost track of this one because of the title. Closing it for now as the latest builds should resolve any issues mentioned above and the release dates for all combinations are up-to-date in the README.md on each branch.  Please feel free to open a new issue if there is anything outstanding.

closed 2026-11-02 by robwoolley