ros/meta-ros #1199: Missing or unbuildable dependency chain was: ['eigenpy', 'python3-scipy']

Marshall-Wen-Amba, 2024-07-30:
We've been using meta-ros(humble) along with Poky Kirkstone for years. Recently, we upgraded them both to the latest. But met below issues when trying to build the firmware.

`````
Parsing of 5042 .bb files complete (4359 cached, 683 parsed). 6823 targets, 249 skipped, 1 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'python3-scipy' (but /cygdrive/y/mwen/server_code/cooper_sdk/ambarella/oss_yocto/meta-ros/meta-ros2-humble/generated-recipes/eigenpy/eigenpy_3.7.0-1.bb DEPENDS on or otherwise requires it). Close matches:
  python3-cppy
  python3-ipy
  python3-scapy
NOTE: Runtime target 'eigenpy' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['eigenpy', 'python3-scipy']
ERROR: Required build target 'ipcam-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['ipcam-image', 'eigenpy', 'python3-scipy']
`````

Through narrowing down, we found that it was introduced by below BIG update. We tried to check whether **python3-scipy** is supported by meta-openembedded or poky. Neither of them supports this package yet. :(
Then we found below one in github. But it seems to be a very old version. And can't go smoothly with latest meta-ros(humble).
https://github.com/gpanders/meta-scipy

I'm wondering how to resolve this issue properly. Where to get a well supported **python3-scipy**?

>
commit bbbe838364749a61c9dbfa8202971888349d96ca
Author: Rob Woolley <rob.woolley@windriver.com>
Date:   Wed Jun 26 02:37:40 2024 +0000
>

`````
commit bbbe838364749a61c9dbfa8202971888349d96ca
Author: Rob Woolley <rob.woolley@windriver.com>
Date:   Wed Jun 26 02:37:40 2024 +0000

    {humble} Sync to files/humble/generated/cache.yaml as of 20240620162234

    Recipes generated by superflore for all packages in ROS distribution humble.

    This pull request was generated by running the following command:

    superflore-gen-oe-recipes --dry-run --no-branch --ros-distro humble --output-repository-path . --upstream-branch HEAD

    Humble Changes:
    ===============
    * ackermann-steering-controller 2.34.0-1 --> 2.35.0-1
    * actionlib-msgs 4.2.3-1 --> 4.2.4-1
    * admittance-controller 2.34.0-1 --> 2.35.0-1
    * apriltag-ros 3.1.2-1 --> 3.2.2-3
`````

Marshall-Wen-Amba, 2024-09-13:
@robwoolley  Hi Rob, I'd like to ping you here to see if you have any comments on this issue. Thanks~~

robwoolley, 2025-09-04:
Sorry for the delayed response.  This was a complicated problem to solve as there was no supported python3-scipy at the time.  I pushed some changes over the summer to migrate from meta-scipy to meta-python-ai.  This is a supported version of python3-scipy.

It should be available on all supported combinations.

closed 2025-09-04 by robwoolley

Marshall-Wen-Amba, 2025-09-04:
Never mind. Thanks for all your GREAT contributions to meta-ros continuously. Really huge work!

Yeah, the issue is confirmed to be resolved with the latest meta-ros. This issue can be closed as I think.