ros/meta-ros #1201: Failed to compile (problem patching)

GustavoPimentel83, 2024-08-05:
**Describe the bug**
Fail to apply patch (0001-Add-in-a-missing-cstdint-include.-1321-1322.patch)

**To Reproduce**
I'm compiling Yocto with Poky, Open Embedded, RAUC, and ROS for the Raspberry PI 4 platform, using an Ubuntu 24.04 OS host.
All the submodule repositories are fixed to the scarthgap tag/branch. I've already confirmed that are updated with the latest changes, In the local.conf the ROS distro is fixed to humble

Build Configuration:
BB_VERSION           = "2.8.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi4-64"
DISTRO               = "poky"
DISTRO_VERSION       = "5.0.2"
TUNE_FEATURES        = "aarch64 crc cortexa72"
TARGET_FPU           = ""
DISTRO_NAME          = "Poky (Yocto Project Reference Distro)"
ROS_DISTRO           = "humble"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"
meta
meta-poky
meta-yocto-bsp       = "scarthgap:f43f393ef0246b7bee6eed8bcf8271cf2b8cdf40"
meta-oe
meta-python
meta-networking
meta-multimedia      = "scarthgap:80e01188fa822d87d301ee71973c462d7a865493"
meta-raspberrypi     = "scarthgap:1918a27419dcd5e79954c0dc0edddcde91057a7e"
meta-rauc            = "scarthgap:f4d80f95952e15b1f98b5adacce7e21508d89b74"
meta-rauc-raspberrypi = "scarthgap:61612093d3e042437429f2875497e0dced07ecf3"
meta-ros2-humble
meta-ros2
meta-ros-common      = "scarthgap:b96ba3d66488aca5868fa1a94643a7f52b2cfc07"

**ERROR**
ERROR: rosbag2-compression-0.15.11-1-r0 do_patch: Applying patch '0001-Add-in-a-missing-cstdint-include.-1321-1322.patch' on target directory '/media/storage/Sandbox/Robotics/Yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-compression/0.15.11-1/git'
CmdError('quilt --quiltrc /media/storage/Sandbox/Robotics/Yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-compression/0.15.11-1/recipe-sysroot-native/etc/quiltrc push', 0, 'stdout: Applying patch 0001-Add-in-a-missing-cstdint-include.-1321-1322.patch
patching file include/rosbag2_compression/compression_options.hpp
Hunk #1 FAILED at 15.
1 out of 1 hunk FAILED -- rejects in file include/rosbag2_compression/compression_options.hpp
Patch 0001-Add-in-a-missing-cstdint-include.-1321-1322.patch can be reverse-applied stderr: ')
ERROR: Logfile of failure stored in: /media/storage/Sandbox/Robotics/Yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-compression/0.15.11-1/temp/log.do_patch.86608
ERROR: Task (/media/storage/Sandbox/Robotics/Yocto/poky/build/../../meta-ros/meta-ros2-humble/generated-recipes/rosbag2/rosbag2-compression_0.15.11-1.bb:do_patch) failed with exit code '1'

**My perspective**
From what understood it seems the meta-ros/meta-ros2-humble/recipes-bbappends/rosbag2/rosbag2-compression_0.15.11-1.bbappend is including a patch (line 4) ([0001-Add-in-a-missing-cstdint-include.-1321-1322.patch](https://github.com/ros/meta-ros/blob/scarthgap/meta-ros2-humble/recipes-bbappends/rosbag2/rosbag2-compression/0001-Add-in-a-missing-cstdint-include.-1321-1322.patch)) that is already present in the [humble source code](https://github.com/ros2/rosbag2/blob/humble/rosbag2_compression/include/rosbag2_compression/compression_options.hpp).

**Possible fix**
Path and entry removal from the recipe

**Expected behavior**
Compilation without failing

jiaxshi, 2024-08-07:
Hi 
You can refer to #1193

robwoolley, 2024-11-21:
@GustavoPimentel83 gustavopimentel83: "@robwoolley, the patch https://github.com/ros/meta-ros/commit/b4db6314bae1c78498158e267200a9c15a739cad needs to be applied as well to the scarthgap branch to avoid compilation issues."

Thanks @GustavoPimentel83 !