ros/meta-ros #1205: Error building rmw-dds-common for raspberrypi0-wifi

jdiez17, 2024-08-19:
**Describe the bug**
`rmw-dds-common` (and possibly other packages) fail to compile when building for a 32 bit target on a 64 bit machine.

**To Reproduce**

```
header:
  version: 13

repos:
  src/poky:
    url: http://git.yoctoproject.org/poky
    branch: scarthgap
    layers:
      meta:
      meta-poky:

  src/openembedded:
    url: https://github.com/openembedded/meta-openembedded
    branch: scarthgap
    layers:
      meta-oe:
      meta-python:
      meta-networking:
      meta-filesystems:
      meta-multimedia:

  src/ros:
    url: https://github.com/ros/meta-ros
    branch: scarthgap
    layers:
      meta-ros-common:
      meta-ros2:
      meta-ros2-humble:

  src/meta-raspberry:
    url: https://github.com/agherzan/meta-raspberrypi
    branch: scarthgap

machine: raspberrypi0-wifi
target: rmw-dds-common
```

Error message:

```
| FAILED: CMakeFiles/rmw_dds_common__rosidl_typesupport_fastrtps_c__pyext.dir/rosidl_generator_py/rmw_dds_common/_rmw_dds_common_s.ep.rosidl_typesupport_fastrtps_c.c.o
| /home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc --sysroot=
/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot -DROS_PACKAGE_NAME=\"rmw_dds_common\" -Drmw_dds_common__rosidl_typesupport_fa
strtps_c__pyext_EXPORTS -I/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/build/rosidl_generator_c -I/home/builder/raccoon-os/build/tmp/work
/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/build/rosidl_generator_py -I/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/rec
ipe-sysroot/opt/ros/humble/include/rosidl_runtime_c -I/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot/opt/ros/humble/include/
rosidl_typesupport_c -I/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot/opt/ros/humble/include/rosidl_typesupport_interface -I
/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot/opt/ros/humble/include/rmw -I/home/builder/raccoon-os/build/tmp/work/arm1176j
zfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/build/rosidl_typesupport_fastrtps_c -I/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/re
cipe-sysroot/opt/ros/humble/include -I/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot/opt/ros/humble/include/rcutils -I/home/
builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot/opt/ros/humble/include/rosidl_runtime_cpp -I/home/builder/raccoon-os/build/tmp/work
/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot/opt/ros/humble/include/rosidl_typesupport_fastrtps_cpp -I/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-p
oky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot/opt/ros/humble/include/rosidl_typesupport_fastrtps_c -I/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-
dds-common/1.6.0-2/recipe-sysroot-native/usr/include/python3.12 -marm -mfpu=vfp -mfloat-abi=hard -mcpu=arm1176jzf-s -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security
 -Werror=format-security -D_TIME_BITS=64 -D_FILE_OFFSET_BITS=64  --sysroot=/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot  -
O2 -pipe -g -feliminate-unused-debug-types -fcanon-prefix-map  -fmacro-prefix-map=/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/git=/usr/s
rc/debug/rmw-dds-common/1.6.0-2  -fdebug-prefix-map=/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/git=/usr/src/debug/rmw-dds-common/1.6.0-
2  -fmacro-prefix-map=/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/build=/usr/src/debug/rmw-dds-common/1.6.0-2  -fdebug-prefix-map=/home/
builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/build=/usr/src/debug/rmw-dds-common/1.6.0-2  -fdebug-prefix-map=/home/builder/raccoon-os/build/tmp
/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot=  -fmacro-prefix-map=/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-commo
n/1.6.0-2/recipe-sysroot=  -fdebug-prefix-map=/home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot-native=  -fPIC -Wall -Wextra -M
D -MT CMakeFiles/rmw_dds_common__rosidl_typesupport_fastrtps_c__pyext.dir/rosidl_generator_py/rmw_dds_common/_rmw_dds_common_s.ep.rosidl_typesupport_fastrtps_c.c.o -MF CMakeFiles/rmw_dds_commo
n__rosidl_typesupport_fastrtps_c__pyext.dir/rosidl_generator_py/rmw_dds_common/_rmw_dds_common_s.ep.rosidl_typesupport_fastrtps_c.c.o.d -o CMakeFiles/rmw_dds_common__rosidl_typesupport_fastrtp
s_c__pyext.dir/rosidl_generator_py/rmw_dds_common/_rmw_dds_common_s.ep.rosidl_typesupport_fastrtps_c.c.o -c /home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds
-common/1.6.0-2/build/rosidl_generator_py/rmw_dds_common/_rmw_dds_common_s.ep.rosidl_typesupport_fastrtps_c.c
| In file included from /home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot-native/usr/include/python3.12/Python.h:38,
|                  from /home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/build/rosidl_generator_py/rmw_dds_common/_rmw_dds_common_s.ep.rosidl
_typesupport_fastrtps_c.c:3:
| /home/builder/raccoon-os/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/rmw-dds-common/1.6.0-2/recipe-sysroot-native/usr/include/python3.12/pyport.h:586:2: error: #error "LONG_BIT defin
ition appears wrong for platform (bad gcc/glibc config?)."
|   586 | #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."
|       |  ^~~~~
```

This appears to happen because the code generated by rosidl includes <Python.h>, which is installed in the sysroot as python-native is a dependency of some of the build tools (see [recipe](https://github.com/ros/meta-ros/blob/master/meta-ros2-humble/generated-recipes/rmw-dds-common/rmw-dds-common_1.6.0-2.bb) and [bbappend](https://github.com/ros/meta-ros/blob/650e92f8e77d31f2db6e4f8de16527c053211675/meta-ros2-humble/recipes-bbappends/rmw-dds-common/rmw-dds-common_1.6.0-2.bbappend)). 

Instead of using the python.h configured for the target, we end up using the host's config. I tried adding `python3` to 
`ROS_BUILD_DEPENDS` with the same result. I suspect this issue may affect many other packages.

jdiez17, 2024-08-20:
I found that adding `inherit python3targetconfig` (as mentioned in openembedded/meta-openembedded#718) to `rmw-dds-common` fixes the compilation step for that package, but as expected many other packages failed in the same way. 

In the end I replaced `python3native` for `python3targetconfig` in ros_ament_cmake.bbclass:

```diff
diff --git a/meta-ros2/classes/ros_ament_cmake.bbclass b/meta-ros2/classes/ros_ament_cmake.bbclass
index d1705a8b4..a559fab5e 100644
--- a/meta-ros2/classes/ros_ament_cmake.bbclass
+++ b/meta-ros2/classes/ros_ament_cmake.bbclass
@@ -23,7 +23,7 @@ EXTRA_OECMAKE:append:class-target = " -DPYTHON_SOABI=${PYTHON_SOABI}"
 #
 export AMENT_PREFIX_PATH="${STAGING_DIR_HOST}${prefix}:${STAGING_DIR_NATIVE}${prefix}:${STAGING_DIR_HOST}${ros_prefix}:${STAGING_DIR_NATIVE}${ros_prefix}"

-inherit cmake python3native
+inherit cmake python3targetconfig

 FILES:${PN}:prepend = " \
     ${datadir}/ament_index \
```

This let me successfully build `ros-core` for the rpi0-wifi. I haven't built it into an image and tried it on the target yet.

However, this leads to the question "how did building for 32-bit targets on 64-bit hosts ever work?"
Looking at the history of `ros_ament_cmake.bbclass`, this `python3native` inherited class was there from the beginning, i.e. it didn't change recently.
I know at least one colleague that has built ROS2 using meta-ros for an ATSAMA5D27 (32-bit).
So the mystery remains. I'm not sure if the proposed patch is appropriate or a hack.

Arrooy, 2025-03-25:
I'm experiencing the same issue.  The patch works but i would expect raspberrypi0-wifi to work out of the box in meta-ros.

carTloyal123, 2025-05-02:
I am building for a different platform and still had to use the patch from @jdiez17 before my builds would succeed. Several packages would not build without it so +1 for keeping it around if possible.