ros/meta-ros #1208: scarthgap: license issue in build time

jiaxshi, 2024-08-23:
**Describe the bug**
Some recipes can not be built due to "do_create_spdx: Cannot find any text for license BSD"

There are multi-varaint of BSD license. Such as:
```
./BSD-2-Clause
./BSD-3-Clause
./BSD-4-Clause
./BSD-1-Clause
./BSD-3-Clause-Open-MPI
```
"BSD" is used for short in package.xml of some ROS projects. Then "LICENSE = "BSD" is used in generated recipes by superflore. And there 's not BSD license file which causes "Cannot find any text for license BSD" failure.

**Known recipes:**
sensor-msgs-py
image-transport
image-transport-plugins
cv-bridge

** Temporary solution**
We can add bbappends to reset LICENSE one by one...  or
cp BSD-3-Clause BSD

robwoolley, 2024-11-21:
Closing issue as resolved.

jiaxshi, 2024-11-22:
Hi @robwoolley 
Maybe I missed something, what's the official solution for the license issue?

robwoolley, 2024-11-22:
Sorry, it referenced a closed issue,.so I thought it had been resolved.   I see that you created the issue to track the larger problem.

I will reopen it.

shr-project, 2024-11-22:
https://github.com/ros/meta-ros/issues/1216#issuecomment-2342686881

eigendude, 2026-02-04:
I tackled a few more ROS components: https://github.com/ros-perception/pcl_msgs/pull/22 and https://github.com/ros-perception/perception_pcl/pull/516. Do they automatically get pulled in by superflore? Will these packages get fixed in the scarthgap branch?

robwoolley, 2026-11-03:
Closing as resolved.  Please create a new issue if there are any remaining problems.

closed 2026-11-03 by robwoolley