ros/meta-ros #1218: The all packages show <ModuleNotFoundError: No module named 'catkin'> this bug. if me not add some ENV?

Eric3495, 2024-08-30:
NOTE: Executing Tasks
ERROR: cmake-modules-0.5.0-1-r0 do_configure: ExecutionError('/home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/temp/run.do_configure.364349', 1, None, None)
ERROR: Logfile of failure stored in: /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/temp/log.do_configure.364349
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/home/fjyx/loongson/2k1000la-yocto/yocto/meta-ros/meta-ros1-noetic/generated-recipes/catkin/catkin_0.8.10-1.bb:do_populate_sysroot', '/home/fjyx/loongson/2k1000la-yocto/yocto/meta-loongson/recipes-devtools/cmake/cmake-native_3.25.2.bb:do_populate_sysroot', '/home/fjyx/loongson/2k1000la-yocto/yocto/meta-external-toolchain/recipes-external/gcc/gcc-runtime-external.bb:do_populate_sysroot', '/home/fjyx/loongson/2k1000la-yocto/yocto/meta-loongson/recipes-devtools/python/python3_3.7.8.bb:do_populate_sysroot', '/home/fjyx/loongson/2k1000la-yocto/yocto/meta-external-toolchain/recipes-external/glibc/glibc-external.bb:do_populate_sysroot', 'virtual:native:/home/fjyx/loongson/2k1000la-yocto/yocto/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/home/fjyx/loongson/2k1000la-yocto/yocto/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/fjyx/loongson/2k1000la-yocto/yocto/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/home/fjyx/loongson/2k1000la-yocto/yocto/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', '/home/fjyx/loongson/2k1000la-yocto/yocto/meta-external-toolchain/recipes-external/gcc/gcc-external-cross.bb:do_populate_sysroot', 'virtual:native:/home/fjyx/loongson/2k1000la-yocto/yocto/meta-loongson/recipes-devtools/python/python3_3.7.8.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['catkin-native', 'cmake-native', 'gcc-runtime-external', 'python3', 'glibc-external', 'ninja-native', 'patch-native', 'pseudo-native', 'quilt-native', 'gcc-external-cross-loongarch64', 'python3-native', 'zlib-native', 'openssl-native', 'bzip2-native', 'ncurses-native', 'xz-native', 'zstd-native', 'libgcc-external', 'libffi', 'util-linux', 'libxcrypt', 'bzip2', 'gdbm', 'xz', 'libtirpc', 'zlib', 'sqlite3', 'openssl', 'libnsl2', 'opkg-utils', 'readline', 're2c-native', 'autoconf-native', 'attr-native', 'automake-native', 'libtool-native', 'python3-catkin-pkg-native', 'googletest-native', 'python3-empy-native', 'python3-setuptools-native', 'python3-nose-native', 'binutils-external-cross-loongarch64', 'libtirpc-native', 'libnsl2-native', 'sqlite3-native', 'readline-native', 'pkgconfig-native', 'libffi-native', 'util-linux-native', 'gdbm-native', 'gettext-minimal-native', 'libcap-ng', 'util-linux-libuuid', 'libpam', 'ncurses', 'bash-completion', 'gnu-config-native', 'm4-native', 'texinfo-dummy-native', 'python3-pyparsing-native', 'util-linux-libuuid-native', 'libpcre2-native', 'libcap-ng-native', 'flex', 'cracklib']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: oe.external.run(['/opt/loongson-gnu-toolchain-8.3/bin/loongarch64-linux-gnu-gcc', '-dumpversion'])
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 8.3.0
| -- The CXX compiler identification is GNU 8.3.0
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Check for working C compiler: /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot-native/usr/bin/loongarch64-Loongson-linux/loongarch64-Loongson-linux-gcc - skipped
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot-native/usr/bin/loongarch64-Loongson-linux/loongarch64-Loongson-linux-g++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Using CATKIN_DEVEL_PREFIX: /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/devel
| -- Using CMAKE_PREFIX_PATH: /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot/opt/ros/noetic;/home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot/usr
| -- Found PythonInterp: /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found suitable version "3.7.8", minimum required is "3")
| -- Using PYTHON_EXECUTABLE: /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3
| -- Using default Python package layout
| -- Found PY_em: /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot-native/usr/lib/python3.7/site-packages/em.py
| -- Using empy: /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot-native/usr/lib/python3.7/site-packages/em.py
| -- Using CATKIN_ENABLE_TESTING: 0
| -- catkin 0.8.10
| -- BUILD_SHARED_LIBS is on
| Traceback (most recent call last):
|   File "/home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/build/catkin_generated/generate_cached_setup.py", line 12, in <module>
|     from catkin.environment_cache import generate_environment_script
| ModuleNotFoundError: No module named 'catkin'
| 
| During handling of the above exception, another exception occurred:
| 
| Traceback (most recent call last):
|   File "/home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/build/catkin_generated/generate_cached_setup.py", line 20, in <module>
|     from catkin.environment_cache import generate_environment_script
| ModuleNotFoundError: No module named 'catkin'
| CMake Error at /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot-native/opt/ros/noetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
| 
|   execute_process(/home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3
|   "/home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/build/catkin_generated/generate_cached_setup.py")
|   returned error code 1
| Call Stack (most recent call first):
|   /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot-native/opt/ros/noetic/share/catkin/cmake/all.cmake:209 (safe_execute_process)
|   /home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/recipe-sysroot-native/opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
|   CMakeLists.txt:4 (find_package)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/build/CMakeFiles/CMakeOutput.log".
| WARNING: exit code 1 from a shell command.
| ERROR: ExecutionError('/home/fjyx/loongson/2k1000la-yocto/yocto/build/tmp/work/loongarch64-Loongson-linux/cmake-modules/0.5.0-1-r0/temp/run.do_configure.364349', 1, None, None)
ERROR: Task (/home/fjyx/loongson/2k1000la-yocto/yocto/meta-ros/meta-ros1-noetic/generated-recipes/cmake-modules/cmake-modules_0.5.0-1.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2096 tasks of which 2095 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/fjyx/loongson/2k1000la-yocto/yocto/meta-ros/meta-ros1-noetic/generated-recipes/cmake-modules/cmake-modules_0.5.0-1.bb:do_configure
Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

Eric3495, 2024-12-05:
yocto version 4.3

robwoolley, 2024-12-07:
Unfortunately, the Yocto Project 4.3 or "Nanbield" release went end-of-life last April.  The last commits for the Nanbield+Noetic combination are from November 2023.

I did want to try to help you out, so I ran these old commands to set up a nanbield+noetic build:

```
mkdir nanbield
cd nanbield
git clone -b build-next --single-branch https://github.com/ros/meta-ros.git build
mkdir conf
ln -snf ../conf build/.
distro=ros1
ros_distro=noetic
oe_release_series=nanbield
cfg=$distro-$ros_distro-$oe_release_series.mcf
cp build/files/$cfg conf/.
build/scripts/mcf -f conf/$cfg
source openembedded-core/oe-init-build-env
bitbake cmake-modules
```

This was able to complete successfully.

When I check the recipe-sysroot-native for the cmake-modules recipe, I see catkin present:

```
../tmp-glibc/work/core2-64-oe-linux/cmake-modules/0.5.0-1/recipe-sysroot-native/opt/ros/noetic/lib/python3.11/site-packages/catkin-0.8.10-py3.11.egg-info
../tmp-glibc/work/core2-64-oe-linux/cmake-modules/0.5.0-1/recipe-sysroot-native/opt/ros/noetic/lib/python3.11/site-packages/catkin
```

I also see that PYTHONPATH is set correctly in the script used for the do_configure task:
```
$ grep PYTHONPATH ../tmp-glibc/work/core2-64-oe-linux/cmake-modules/0.5.0-1/temp/run.do_configure
export PYTHONPATH="/ala-lpggp31/rwoolley/nanbield/tmp-glibc/work/core2-64-oe-linux/cmake-modules/0.5.0-1/recipe-sysroot-native/opt/ros/noetic/lib/python3.11/site-packages"
        export PYTHONPATH=/ala-lpggp31/rwoolley/nanbield/tmp-glibc/work/core2-64-oe-linux/cmake-modules/0.5.0-1/recipe-sysroot/usr/lib/python-sysconfigdata:$PYTHONPATH
```

I suggest checking your build to see what you see.  It seems like something may be interfering with Python's ability to find the catkin module.

I see that there is work being done to support "scarthgap": https://github.com/loongarch64/meta-loongarch/blob/master/conf/layer.conf

If you can, you may consider moving to a supported Yocto release.  I haven't tried meta-loongarch myself, but it may prove to be less difficult in the long run.

robwoolley, 2026-11-03:
Closing as resolved.  Please open a new issue if you are still having issues.

closed 2026-11-03 by robwoolley