ros/meta-ros #1219: meta-ros2-jazzy hardware-interfaces fails to build with Scarthgap

grsandeep85, 2024-09-18:
**Issue Description**
meta-ros2-jazzy hardware-interfaces fails to build with Scarthgap, also other ros2-control recipes has dependencies on hardware-interfaces.

**Steps To Reproduce**

Add meta-ros-common meta-ros2 meta-ros2-jazzy to scrathgap poky build and run below command.

```
$ bitbake hardware-interfaces
```

**Error Logs**
```
| In file included from /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions.hpp:18,
|                  from /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/client.hpp:40,
|                  from /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/callback_group.hpp:24,
|                  from /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/node.hpp:40,
|                  from /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/git/include/hardware_interface/resource_manager.hpp:34,
|                  from /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/git/src/resource_manager.cpp:15:
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp: In constructor 'rclcpp::exceptions::InvalidNodeNameError::InvalidNodeNameError(const char*, const char*, size_t)':
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:71:79: error: declaration of 'invalid_index' shadows a member of 'rclcpp::exceptions::InvalidNodeNameError' [-Werror=shadow]
|    71 |   InvalidNodeNameError(const char * node_name, const char * error_msg, size_t invalid_index)
|       |                                                                        ~~~~~~~^~~~~~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:64:16: note: shadowed declaration is here
|    64 |   const size_t invalid_index;
|       |                ^~~~~~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:71:61: error: declaration of 'error_msg' shadows a member of 'rclcpp::exceptions::InvalidNodeNameError' [-Werror=shadow]
|    71 |   InvalidNodeNameError(const char * node_name, const char * error_msg, size_t invalid_index)
|       |                                                ~~~~~~~~~~~~~^~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:63:21: note: shadowed declaration is here
|    63 |   const std::string error_msg;
|       |                     ^~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp: In constructor 'rclcpp::exceptions::InvalidNodeNameError::InvalidNodeNameError(const char*, const char*, size_t)':
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:71:79: error: declaration of 'invalid_index' shadows a member of 'rclcpp::exceptions::InvalidNodeNameError' [-Werror=shadow]
|    71 |   InvalidNodeNameError(const char * node_name, const char * error_msg, size_t invalid_index)
|       |                                                                        ~~~~~~~^~~~~~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:64:16: note: shadowed declaration is here
|    64 |   const size_t invalid_index;
|       |                ^~~~~~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:71:61: error: declaration of 'error_msg' shadows a member of 'rclcpp::exceptions::InvalidNodeNameError' [-Werror=shadow]
|    71 |   InvalidNodeNameError(const char * node_name, const char * error_msg, size_t invalid_index)
|       |                                                ~~~~~~~~~~~~~^~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:63:21: note: shadowed declaration is here
|    63 |   const std::string error_msg;
|       |                     ^~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp: In constructor 'rclcpp::exceptions::InvalidNodeNameError::InvalidNodeNameError(const char*, const char*, size_t)':
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:71:79: error: declaration of 'invalid_index' shadows a member of 'rclcpp::exceptions::InvalidNodeNameError' [-Werror=shadow]
|    71 |   InvalidNodeNameError(const char * node_name, const char * error_msg, size_t invalid_index)
|       |                                                                        ~~~~~~~^~~~~~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:64:16: note: shadowed declaration is here
|    64 |   const size_t invalid_index;
|       |                ^~~~~~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:71:61: error: declaration of 'error_msg' shadows a member of 'rclcpp::exceptions::InvalidNodeNameError' [-Werror=shadow]
|    71 |   InvalidNodeNameError(const char * node_name, const char * error_msg, size_t invalid_index)
|       |                                                ~~~~~~~~~~~~~^~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:63:21: note: shadowed declaration is here
|    63 |   const std::string error_msg;
|       |                     ^~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp: In constructor 'rclcpp::exceptions::InvalidNamespaceError::InvalidNamespaceError(const char*, const char*, size_t)':
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:80:81: error: declaration of 'invalid_index' shadows a member of 'rclcpp::exceptions::InvalidNamespaceError' [-Werror=shadow]
|    80 |   InvalidNamespaceError(const char * namespace_, const char * error_msg, size_t invalid_index)
|       |                                                                          ~~~~~~~^~~~~~~~~~~~~
| /scratch/sandeep/yocto/2024.2/yp-dev/build-ros/tmp/work/cortexa72-cortexa53-xilinx-linux/hardware-interface/4.8.0-2/recipe-sysroot/opt/ros/jazzy/include/rclcpp/rclcpp/exceptions/exceptions.hpp:64:16: note:  [... cut ...]
+# exceptions.hpp:89:63: error: declaration of 'error_msg' shadows a member of 'rclcpp::exceptions::InvalidTopicNameError' [-Werror=shadow]
+# exceptions.hpp:98:83: error: declaration of 'invalid_index' shadows a member of 'rclcpp::exceptions::InvalidServiceNameError' [-Werror=shadow]
+# exceptions.hpp:98:65: error: declaration of 'error_msg' shadows a member of 'rclcpp::exceptions::InvalidServiceNameError' [-Werror=shadow]
+CXXFLAGS += "-Wno-error=shadow"

+
+ROS_BUILDTOOL_DEPENDS += " \
     ament-cmake-ros-native \
     python3-numpy-native \
 "
```

graceagrace, 2025-04-30:
Hi @grsandeep85, it seems the root cause of the build error is the addition of `-Werror-shadow` to the CMakeLists.txt in the package sources. I confirmed adding `CXXFLAGS += "-Wno-error=shadow"` did not prevent this warning-to-error promotion. My solution was to patch the sources as follows:
```
From c58d630e29cf3ea754b7c051f4be842b80a00edf Mon Sep 17 00:00:00 2001
From: Amelia Grace <agrace@bastiansolutions.com>
Date: Wed, 30 Apr 2025 10:36:59 -0600
Subject: [PATCH] remove-werror-shadow-option

---
 CMakeLists.txt | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/CMakeLists.txt b/CMakeLists.txt
index 54474889f..a494a83e9 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -2,7 +2,7 @@ cmake_minimum_required(VERSION 3.5)
 project(hardware_interface)
 
 if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
-  add_compile_options(-Wall -Wextra -Werror=conversion -Werror=unused-but-set-variable -Werror=return-type -Werror=shadow)
+  add_compile_options(-Wall -Wextra -Werror=conversion -Werror=unused-but-set-variable -Werror=return-type)
 endif()
 
 set(THIS_PACKAGE_INCLUDE_DEPENDS
```
I don't like how it seems like we can't override warning-to-error promotions in package sources with `CXXFLAGS`. I've experienced this with other packages, recently, too (`control-toolbox` is one example).

grsandeep85, 2025-05-21:
Hi @graceagrace ,

Back in March during my spare time I investigated on this issue. I do have a working patch for this but I need to find some time and rebase with latest code base and summit the patch.

robwoolley, 2026-11-03:
This should be resolved now.  A quick check on Scarthgap builds hardware-interface successfully.  Please reopen the a new issue if the problem persists.

closed 2026-11-03 by robwoolley