ros/meta-ros #1246: yocto 3.3.6 add ros1 noetic bug

DevilMayFree, 2024-11-05:
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/home/yx/yocto/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/home/yx/yocto/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', '/home/yx/yocto/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', '/home/yx/yocto/meta-ros/meta-ros1-noetic/generated-recipes/roscpp-core/cpp-common_0.7.2-1.bb:do_populate_sysroot', '/home/yx/yocto/meta-external-toolchain/recipes-external/gcc/gcc-external-cross.bb:do_populate_sysroot', '/home/yx/yocto/meta-loongson/recipes-devtools/python/python3_3.7.8.bb:do_populate_sysroot', '/home/yx/yocto/meta-external-toolchain/recipes-external/glibc/glibc-external.bb:do_populate_sysroot', '/home/yx/yocto/meta-external-toolchain/recipes-external/gcc/gcc-runtime-external.bb:do_populate_sysroot', '/home/yx/yocto/meta-loongson/recipes-devtools/cmake/cmake-native_3.25.2.bb:do_populate_sysroot', 'virtual:native:/home/yx/yocto/meta-loongson/recipes-devtools/python/python3_3.7.8.bb:do_populate_sysroot', 'virtual:native:/home/yx/yocto/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/home/yx/yocto/meta/recipes-support/boost/boost_1.76.0.bb:do_populate_sysroot', 'virtual:native:/home/yx/yocto/meta-ros/meta-ros1-noetic/generated-recipes/catkin/catkin_0.8.10-1.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['ninja-native', 'patch-native', 'quilt-native', 'cpp-common', 'gcc-external-cross-loongarch64', 'python3', 'glibc-external', 'gcc-runtime-external', 'cmake-native', 'python3-native', 'pseudo-native', 'boost', 'catkin-native', 'catkin', 'console-bridge', 'util-linux', 'readline', 'sqlite3', 'libnsl2', 'libxcrypt', 'opkg-utils', 'zlib', 'libffi', 'openssl', 'xz', 'gdbm', 'bzip2', 'libtirpc', 're2c-native', 'libtool-native', 'autoconf-native', 'attr-native', 'automake-native', 'binutils-external-cross-loongarch64', 'libgcc-external', 'xz-native', 'openssl-native', 'bzip2-native', 'zlib-native', 'zstd-native', 'ncurses-native', 'gdbm-native', 'libtirpc-native', 'pkgconfig-native', 'libnsl2-native', 'readline-native', 'libffi-native', 'sqlite3-native', 'util-linux-native', 'icu', 'python3-empy-native', 'googletest-native', 'python3-setuptools-native', 'python3-catkin-pkg-native', 'python3-nose-native', 'python3-catkin-pkg', 'python3-empy', 'googletest', 'python3-nose', 'util-linux-libuuid', 'ncurses', 'bash-completion', 'libpam', 'libcap-ng', 'texinfo-dummy-native', 'm4-native', 'gnu-config-native', 'gettext-minimal-native', 'libpcre2-native', 'util-linux-libuuid-native', 'libcap-ng-native', 'python3-pyparsing-native', 'python3-pyparsing', 'flex', 'cracklib', 'unzip-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: oe.external.run(['/opt/loongson-gnu-toolchain-8.3/bin/loongarch64-linux-gnu-gcc', '-dumpversion'])
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 8.3.0
| -- The CXX compiler identification is GNU 8.3.0
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Check for working C compiler: /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/recipe-sysroot-native/usr/bin/loongarch64-Loongson-linux/loongarch64-Loongson-linux-gcc - skipped
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/recipe-sysroot-native/usr/bin/loongarch64-Loongson-linux/loongarch64-Loongson-linux-g++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Using CATKIN_DEVEL_PREFIX: /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/devel
| -- Using CMAKE_PREFIX_PATH: /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/recipe-sysroot/opt/ros/noetic;/home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/recipe-sysroot/usr
| -- This workspace overlays: /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/recipe-sysroot/opt/ros/noetic
| -- Found PythonInterp: /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found suitable version "3.7.8", minimum required is "3")
| -- Using PYTHON_EXECUTABLE: /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3
| -- Using default Python package layout
| -- Found PY_em: /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/recipe-sysroot-native/usr/lib/python3.7/site-packages/em.py
| -- Using empy: /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/recipe-sysroot-native/usr/lib/python3.7/site-packages/em.py
| -- Using CATKIN_ENABLE_TESTING: 0
| -- catkin 0.8.10
| -- BUILD_SHARED_LIBS is on
| CMake Error at /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/recipe-sysroot/opt/ros/noetic/share/cpp_common/cmake/cpp_commonConfig.cmake:207 (message):
|   Project 'rostime' tried to find library 'cpp_common'.  The library is
|   neither a target nor built/installed properly.  Did you compile project
|   'cpp_common'? Did you find_package() it before the subdirectory containing
|   its code is included?
| Call Stack (most recent call first):
|   /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/recipe-sysroot/opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
|   CMakeLists.txt:4 (find_package)
|
|
| -- Configuring incomplete, errors occurred!
| See also "/home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/build/CMakeFiles/CMakeOutput.log".
| WARNING: exit code 1 from a shell command.
| ERROR: ExecutionError('/home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/rostime/0.7.2-1-r0/temp/run.do_configure.3798469', 1, None, None)
ERROR: Task (/home/yx/yocto/meta-ros/meta-ros1-noetic/generated-recipes/roscpp-core/rostime_0.7.2-1.bb:do_configure) failed with exit code '1'
ERROR: roslib-1.15.8-1-r0 do_configure: ExecutionError('/home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/roslib/1.15.8-1-r0/temp/run.do_configure.3799382', 1, None, None)
ERROR: Logfile of failure stored in: /home/yx/yocto/build/tmp/work/loongarch64-Loongson-linux/roslib/1.15.8-1-r0/temp/log.do_configure.3799382

DevilMayFree, 2024-11-05:
yocto version: hardknott , use branch : hardknott

robwoolley, 2024-11-09:
Thanks for reporting this.  I just pushed updates for noetic to successfully build ros-image-core across all the supported Yocto releases: kirkstone, scarthgap, styhead, and walnascar (ie master).   Unfortunately, hardknott went end-of-life in April 2022: https://wiki.yoctoproject.org/wiki/Releases  and is no longer receiving updates.

I am interested in hearing feedback from people using old releases.   Is there a reason why you can't move forward to a newer Yocto release?   Also, Noetic goes end-of-life next year.   I am interested in feedback on whether developers are planning to migrate to ROS 2.   If there are any ways that we can assist in the migration or provide extended support for Noetic for a subset of ROS 1 packages, that would be interesting to know.

robwoolley, 2026-11-03:
Closing as assumed resolved.

closed 2026-11-03 by robwoolley