ros/meta-ros #1251: Issue with shared-queues-vendor Fetching on Ubuntu 22.04 with Scarthgap Yocto

Tiqo01, 2024-11-08:
Hello,

I'm encountering an issue while using Ubuntu 22.04 with the Scarthgap Yocto setup when trying to fetch the shared-queues-vendor package.

Here is the error log:

```
WARNING: shared-queues-vendor-0.29.0-1-r0 do_fetch: Failed to fetch URL git://github.com/ros2-gbp/rosbag2-release;name=release;branch=release/rolling/shared_queues_vendor;protocol=https, attempting MIRRORS if available
ERROR: shared-queues-vendor-0.29.0-1-r0 do_fetch: Fetcher failure: Unable to find revision dc17fed08353f30a7e20676df7c8fc5e842ed011 in branch release/rolling/shared_queues_vendor even from upstream
ERROR: shared-queues-vendor-0.29.0-1-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/ros2-gbp/rosbag2-release;name=release;branch=release/rolling/shared_queues_vendor;protocol=https')
ERROR: Logfile of failure stored in: /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/shared-queues-vendor/0.29.0-1/temp/log.do_fetch.1461030
ERROR: Task (/home/yocto/poky/build/meta-ros/meta-ros2-rolling/generated-recipes/rosbag2/shared-queues-vendor_0.29.0-1.bb:do_fetch) failed with exit code '1'
```

It seems the fetcher cannot find the required revision dc17fed08353f30a7e20676df7c8fc5e842ed011 for the release/rolling/shared_queues_vendor branch, even when attempting upstream sources.

Could you please provide guidance on how to resolve this issue or any potential workarounds?

robwoolley, 2024-11-09:
Thanks for reporting this.   This commit to master should fix the issue for you: https://github.com/ros/meta-ros/commit/976b0603a1ce83cb27d542bd87f1a97380a4ada7

I will merge it to scarthgap shortly

Tiqo01, 2024-11-09:
Thank you for the quick response ! 

I encountered an error when building the rosbag2-interfaces recipe. The build stops due to an ld linking error.

```
| collect2: error: ld returned 1 exit status
| ninja: build stopped: subcommand failed.
|
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/yocto/poky/build/meta-ros/meta-ros2-rolling/generated-recipes/rosbag2/rosbag2-interfaces_0.27.0-1.bb:do_compile) failed with exit code '1'
```
Any guidance on resolving this linking issue would be appreciated. Thank you!

The error log : 

```
DEBUG: Executing shell function do_compile
NOTE: VERBOSE=1 cmake --build /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/build --target all --
Change Dir: '/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/build'

Run Build Command(s): ninja -v -j 8 all
[1/174] cd /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/build && /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot-native/usr/bin/python3-native/python3 /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot/opt/ros/rolling/lib/rosidl_generator_type_description/rosidl_generator_type_description --generator-arguments-file /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/build/rosidl_generator_type_description__arguments.json
[2/174] cd /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/build && /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot-native/usr/bin/python3-native/python3 /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot/opt/ros/rolling/share/rosidl_generator_cpp/cmake/../../../lib/rosidl_generator_cpp/rosidl_generator_cpp --generator-arguments-file /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/build/rosidl_generator_cpp__arguments.json
[3/174] cd /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/build && /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot-native/usr/bin/python3-native/python3 /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot/opt/ros/rolling/share/rosidl_generator_c/cmake/../../../lib/rosidl_generator_c/rosidl_generator_c --generator-arguments-file /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/build/rosidl_generator_c__arguments.json
[4/174] /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc --sysroot=/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot -DROSIDL_GENERATOR_C_BUILDING_DLL_rosbag2_interfaces -DROS_PACKAGE_NAME=\"rosbag2_interfaces\" -I/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/build/rosidl_generator_c -I/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot/opt/ros/rolling/include/builtin_interfaces -I/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot/opt/ros/rolling/include/rosidl_runtime_c -I/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot/opt/ros/rolling/include/rcutils -I/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot/opt/ros/rolling/include/rosidl_typesupport_interface -I/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot-native/opt/ros/rolling/include/service_msgs -mcpu=cortex-a72+crc -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fcanon-prefix-map  -fmacro-prefix-map=/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/git=/usr/src/debug/rosbag2-interfaces/0.27.0-1  -fdebug-prefix-map=/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/git=/usr/src/debug/rosbag2-interfaces/0.27.0-1  -fmacro-prefix-map=/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/build=/usr/src/debug/rosbag2-interfaces/0.27.0-1  -fdebug-prefix-map=/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/build=/usr/src/debug/rosbag2-interfaces/0.27.0-1  -fdebug-prefix-map=/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot=  -fmacro-prefix-map=/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot=  -fdebug-prefix-map=/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot-native=  -std=gnu11 -fPIC -Wall -MD -MT CMakeFiles/rosbag2_interfaces__rosidl_generator_c.dir/rosidl_generator_c/rosbag2_interfaces/msg/detail/read_split_event__type_support.c.o -MF CMakeFiles/rosbag2_interfaces__rosidl_generator_c.dir/rosidl_generator_c/rosbag2_interfaces/msg/detail/read_split_event__type_support.c.o.d -o CMakeFiles/rosbag2_interfaces__rosidl_generator_c.dir/rosidl_generator_c/rosbag2_interfaces/msg/detail/read_split_eve [... cut ...]
/home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/13.3.0/ld: /home/yocto/poky/build/tmp/work/cortexa72-poky-linux/rosbag2-interfaces/0.27.0-1/recipe-sysroot-native/opt/ros/rolling/lib/libservice_msgs__rosidl_generator_c.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

WARNING: exit code 1 from a shell command.
```

Tiqo01, 2024-11-10:
I successfully compiled everything by cloning the 976b060 branch and modifying various `layer.conf `files to add `LAYERSERIES_COMPAT_ros2-rolling-layer = "walnascar styhead scarthgap"`. I also encountered the issue `ERROR: liblz4-vendor-0.27.0-1-r0 do_create_spdx: Cannot find any text for license BSD,` which I resolved by running `cp poky/meta/files/common-licenses/BSD-3-Clause poky/meta/files/common-licenses/BSD`. Thank you!

closed 2024-11-10 by Tiqo01