ros/meta-ros #1266: PYTHONPATH with trailing ":" leads to build issues for python packages

pbert519, 2024-11-28:
**Describe the bug**

Since commit [abf90dc52c710cf1584bccdd7f2ea0a5a564de0a](https://github.com/ros/meta-ros/commit/abf90dc52c710cf1584bccdd7f2ea0a5a564de0a#diff-137f704b9314158d6fd6b860c4f14032f9c37b1e275f0369755fad1a699b4228R33) on branch `kirkstone` `ament-cmake-core` does no longer build with the following error:

```
ERROR: ament-cmake-core-2.5.2-1-r0 do_configure: ExecutionError('/opt/yocto/TAT_Yocto_AUM/build/tmp/work/corei7-64-poky-linux/ament-cmake-core/2.5.2-1-r0/temp/run.do_configure.2326697', 1, None, None)
ERROR: Logfile of failure stored in: /opt/yocto/TAT_Yocto_AUM/build/tmp/work/corei7-64-poky-linux/ament-cmake-core/2.5.2-1-r0/temp/log.do_configure.2326697
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/poky/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/poky/meta/recipes-devtools/cmake/cmake-native_3.22.3.bb:do_populate_sysroot', '/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/poky/meta/recipes-devtools/gcc/gcc-cross_11.3.bb:do_populate_sysroot', '/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/poky/meta/recipes-devtools/gcc/gcc-runtime_11.3.bb:do_populate_sysroot', '/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/meta-ros/meta-ros-common/recipes-infrastructure/python/python3-catkin-pkg_0.4.24.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/meta-ros/meta-ros2-jazzy/generated-recipes/ament-package/ament-package_0.16.3-3.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/poky/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/poky/meta/recipes-devtools/python/python3_3.10.9.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['glibc', 'cmake-native', 'gcc-cross-x86_64', 'gcc-runtime', 'quilt-native', 'python3-catkin-pkg-native', 'ament-package-native', 'ninja-native', 'patch-native', 'pseudo-native', 'python3-native', 'python3-setuptools-native', 'python3-importlib-metadata-native', 'python3-pyparsing-native', 'python3-wheel-native', 'python3-installer-native', 'libtirpc-native', 'libtool-native', 'zlib-native', 'readline-native', 'libffi-native', 'xz-native', 'gdbm-native', 'sqlite3-native', 'util-linux-libuuid-native', 'libnsl2-native', 'ncurses-native', 'openssl-native', 'bzip2-native', 'linux-libc-headers', 're2c-native', 'binutils-cross-x86_64', 'gnu-config-native', 'mpfr-native', 'zstd-native', 'gmp-native', 'texinfo-dummy-native', 'flex-native', 'libmpc-native', 'libgcc', 'attr-native', 'curl-native', 'python3-toml-native', 'python3-zipp-native', 'python3-setuptools-scm-native', 'python3-pathlib2-native', 'python3-flit-core-native', 'unzip-native', 'gettext-minimal-native', 'perl-native', 'm4-native', 'python3-more-itertools-native', 'python3-packaging-native', 'python3-tomli-native', 'python3-six-native', 'make-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| -- Found Python3: /opt/yocto/TAT_Yocto_AUM/build/tmp/work/corei7-64-poky-linux/ament-cmake-core/2.5.2-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found version "3.10.9") found components: Interpreter
| WARNING: Package "%s" has newlines in the description
| -- ament_cmake_core 2.5.2
| Traceback (most recent call last):
|   File "/opt/yocto/TAT_Yocto_AUM/build/tmp/work/corei7-64-poky-linux/ament-cmake-core/2.5.2-1-r0/git/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
|     from ament_package.templates import get_environment_hook_template_path
| ModuleNotFoundError: No module named 'ament_package'
| -- Configuring incomplete, errors occurred!
| See also "/opt/yocto/TAT_Yocto_AUM/build/tmp/work/corei7-64-poky-linux/ament-cmake-core/2.5.2-1-r0/build/CMakeFiles/CMakeOutput.log".
| CMake Error at ament_cmake_package_templates-extras.cmake:41 (message):
| 
|   execute_process(/opt/yocto/TAT_Yocto_AUM/build/tmp/work/corei7-64-poky-linux/ament-cmake-core/2.5.2-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3
|   /opt/yocto/TAT_Yocto_AUM/build/tmp/work/corei7-64-poky-linux/ament-cmake-core/2.5.2-1-r0/git/cmake/package_templates/templates_2_cmake.py
|   /opt/yocto/TAT_Yocto_AUM/build/tmp/work/corei7-64-poky-linux/ament-cmake-core/2.5.2-1-r0/build/ament_cmake_package_templates/templates.cmake)
|   returned error code 1
| Call Stack (most recent call first):
|   CMakeLists.txt:19 (include)
| 
| 
| WARNING: exit code 1 from a shell command.
ERROR: Task (/opt/yocto/TAT_Yocto_AUM/sources/poky/../../sources/meta-ros/meta-ros2-jazzy/generated-recipes/ament-cmake/ament-cmake-core_2.5.2-1.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 848 tasks of which 847 didn't need to be rerun and 1 failed.

```

Maybe the root cause is that PYTHONPATH variable have a trailing `:`  if PYTHONPATH was not set.
`export PYTHONPATH:prepend = "${STAGING_DIR_NATIVE}${PYTHON_SITEPACKAGES_DIR}:"`

With `bitbake ament-cmake-core -c devshell` and printing python3 `sys.path`, `RECIPE_SYSROOT_NATIVE/opt/...` path is missing.

**To Reproduce**
I build ros2-jazzy for kirkstone on a x86_64 machine.
`bitbake ament-cmake-core`

```
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:a3b9a12fc160b213edc9062493933354171c30a3"
meta-ros-common      
meta-ros2            
meta-ros2-jazzy      = "HEAD:abf90dc52c710cf1584bccdd7f2ea0a5a564de0a"
```

**Expected behavior**
The python packages installed to `RECIPE_SYSROOT_NATIVE/opt/...` are found an the build succeds.

robwoolley, 2024-11-28:
Thanks, this fix from master should fix it: https://github.com/ros/meta-ros/commit/928b66f1c897abbd10c19747556ee94ef8f41d5e

It has been tested on the other combinations and I am in the process of porting it.   I actually wanted to create an open issue to track this so thank you for doing that for me. :D

jiaxshi, 2024-12-03:
Hi 
You can try with #1269 .