ros/meta-ros #1268: The build process of ament-cmake-core-native fails for ROS2 humble with yocto kirkstone

grischi, 2024-12-02:
**Describe the bug**
The build process of ament-cmake-core-native fails for ROS2 humble with yocto kirkstone (clean build).

**To Reproduce**
_List all the Yocto layers you have included in the build and the revisions you were using:_
2024-11-30 02:10:31 - INFO     - Repository poky checked out to c0dcc563322cbe8b4c323172d66d04d16d2067e8
2024-11-30 02:10:31 - INFO     - Repository freescale checked out to c525e0c19bdc46d45f71873b5f286f49abb69418
2024-11-30 02:10:31 - INFO     - Repository virtualization checked out to c996df33763f292da5e7513c574272d7de23eafc
2024-11-30 02:10:31 - INFO     - Repository freescale-distro checked out to d5bbb487b2816dfc74984a78b67f7361ce404253
2024-11-30 02:10:31 - INFO     - Repository clang checked out to b33aa9c7f54807c113c44b0f2c15a5c5d88ae5ee
2024-11-30 02:10:31 - INFO     - Repository timesys checked out to c2f65e99c8ac3001452a2c436a4a22413d2f95ef
2024-11-30 02:10:31 - INFO     - Repository openembedded checked out to 4ad41baed6236d499804cbfc4f174042d84fce97
2024-11-30 02:10:31 - INFO     - Repository nxp-qoriq checked out to f1b87b247b1dbd9392e421b82ece1d91408fef8b
2024-11-30 02:10:32 - INFO     - Repository ros checked out to 72f70f7c1147197bd1e447c03187628b4b62a738
2024-11-30 02:10:32 - INFO     - Repository bitbake checked out to 40fd5f4eef7460ca67f32cfce8e229e67e1ff607

_What MACHINE you're building for and what host OS you're using. The "Build Configuration:" section shown by bitbake at the beginning of the build is very good start:_
2024-11-30 02:12:26 - INFO     - Build Configuration:
2024-11-30 02:12:26 - INFO     - BB_VERSION           = "2.0.0"
2024-11-30 02:12:26 - INFO     - BUILD_SYS            = "x86_64-linux"
2024-11-30 02:12:26 - INFO     - NATIVELSBSTRING      = "ubuntu-22.04"
2024-11-30 02:12:26 - INFO     - TARGET_SYS           = "aarch64-ak-linux"
2024-11-30 02:12:26 - INFO     - TUNE_FEATURES        = "aarch64"
2024-11-30 02:12:26 - INFO     - TARGET_FPU           = ""
2024-11-30 02:12:26 - INFO     - ROS_DISTRO           = "humble"
2024-11-30 02:12:26 - INFO     - ROS_VERSION          = "2"
2024-11-30 02:12:26 - INFO     - ROS_PYTHON_VERSION   = "3"
2024-11-30 02:12:26 - INFO     - meta-clang           = "kirkstone:b33aa9c7f54807c113c44b0f2c15a5c5d88ae5ee"
2024-11-30 02:12:26 - INFO     - meta-freescale       = "kirkstone:c525e0c19bdc46d45f71873b5f286f49abb69418"
2024-11-30 02:12:26 - INFO     - meta-freescale-distro = "kirkstone:d5bbb487b2816dfc74984a78b67f7361ce404253"
2024-11-30 02:12:26 - INFO     - meta-microsys-layerscape = "kirkstone:cda955d5029a82b875adbb61ddad2ce836f18a30"
2024-11-30 02:12:26 - INFO     - meta-filesystems
2024-11-30 02:12:26 - INFO     - meta-gnome
2024-11-30 02:12:26 - INFO     - meta-multimedia
2024-11-30 02:12:26 - INFO     - meta-networking
2024-11-30 02:12:26 - INFO     - meta-oe
2024-11-30 02:12:26 - INFO     - meta-perl
2024-11-30 02:12:26 - INFO     - meta-python
2024-11-30 02:12:26 - INFO     - meta-webserver       = "kirkstone:4ad41baed6236d499804cbfc4f174042d84fce97"
2024-11-30 02:12:26 - INFO     - meta-qoriq           = "kirkstone:f1b87b247b1dbd9392e421b82ece1d91408fef8b"
2024-11-30 02:12:26 - INFO     - meta-ros-common
2024-11-30 02:12:26 - INFO     - meta-ros2
2024-11-30 02:12:26 - INFO     - meta-ros2-humble     = "kirkstone:72f70f7c1147197bd1e447c03187628b4b62a738"
2024-11-30 02:12:26 - INFO     - meta-timesy          = "kirkstone:c2f65e99c8ac3001452a2c436a4a22413d2f95ef"
2024-11-30 02:12:26 - INFO     - meta-virtualization  = "kirkstone:c996df33763f292da5e7513c574272d7de23eafc"
2024-11-30 02:12:26 - INFO     - meta
2024-11-30 02:12:26 - INFO     - meta-poky
2024-11-30 02:12:26 - INFO     - meta-yocto-bsp       = "kirkstone:c0dcc563322cbe8b4c323172d66d04d16d2067e8"

_Then show the exact error you're seeing, often it's useful to attach whole log.do_<task> from ${WORKDIR} and if it doesn't show much details, then additional files as well (like config.log from ${B} directory for do_configure errors with autotools or CMakeOutput.log/CMakeError.log files with cmake)._

2024-11-30 06:57:03 - ERROR    - ERROR: ament-cmake-core-native-1.3.10-1-r0 do_configure: ExecutionError('/yocto/image/build/tmp/work/x86_64-linux/ament-cmake-core-native/1.3.10-1-r0/temp/run.do_configure.435027', 1, None, None)
2024-11-30 06:57:03 - ERROR    - ERROR: Logfile of failure stored in: /yocto/image/build/tmp/work/x86_64-linux/ament-cmake-core-native/1.3.10-1-r0/temp/log.do_configure.435027
2024-11-30 06:57:03 - INFO     - Log data follows:
2024-11-30 06:57:03 - INFO     - | DEBUG: Executing python function extend_recipe_sysroot
2024-11-30 06:57:03 - INFO     - | NOTE: Direct dependencies are ['/yocto/image/build/../layers/poky/meta/recipes-devtools/cmake/cmake-native_3.22.3.bb:do_populate_sysroot', '/yocto/image/build/../layers/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/yocto/image/build/../layers/meta-ros/meta-ros-common/recipes-infrastructure/python/python3-catkin-pkg_0.4.24.bb:do_populate_sysroot', 'virtual:native:/yocto/image/build/../layers/meta-ros/meta-ros2-humble/generated-recipes/ament-package/ament-package_0.14.0-4.bb:do_populate_sysroot', 'virtual:native:/yocto/image/build/../layers/poky/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/yocto/image/build/../layers/poky/meta/recipes-devtools/python/python3_3.10.15.bb:do_populate_sysroot']
2024-11-30 06:57:03 - INFO     - | NOTE: Installed into sysroot: []
2024-11-30 06:57:03 - INFO     - | NOTE: Skipping as already exists in sysroot: ['cmake-native', 'quilt-native', 'python3-catkin-pkg-native', 'ament-package-native', 'ninja-native', 'python3-native', 'python3-setuptools-native', 'python3-pyparsing-native', 'python3-installer-native', 'python3-wheel-native', 'python3-importlib-metadata-native', 'ncurses-native', 'sqlite3-native', 'libtirpc-native', 'gdbm-native', 'openssl-native', 'util-linux-libuuid-native', 'libtool-native', 'bzip2-native', 'libffi-native', 'libnsl2-native', 'readline-native', 'zlib-native', 'xz-native', 'curl-native', 'zstd-native', 're2c-native', 'unzip-native', 'python3-flit-core-native', 'python3-pathlib2-native', 'python3-zipp-native', 'python3-setuptools-scm-native', 'python3-toml-native', 'gettext-minimal-native', 'texinfo-dummy-native', 'perl-native', 'python3-six-native', 'python3-more-itertools-native', 'python3-packaging-native', 'python3-tomli-native', 'make-native']
2024-11-30 06:57:03 - INFO     - | DEBUG: Python function extend_recipe_sysroot finished
2024-11-30 06:57:03 - INFO     - | DEBUG: Executing shell function do_configure
2024-11-30 06:57:03 - INFO     - | -- Found Python3: /yocto/image/build/tmp/work/x86_64-linux/ament-cmake-core-native/1.3.10-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found version "3.10.15") found components: Interpreter
2024-11-30 06:57:03 - INFO     - | WARNING: Package "%s" has newlines in the description
2024-11-30 06:57:03 - INFO     - | -- ament_cmake_core 1.3.10
2024-11-30 06:57:03 - INFO     - | Traceback (most recent call last):
2024-11-30 06:57:03 - INFO     - |   File "/yocto/image/build/tmp/work/x86_64-linux/ament-cmake-core-native/1.3.10-1-r0/git/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
2024-11-30 06:57:03 - INFO     - |     from ament_package.templates import get_environment_hook_template_path
2024-11-30 06:57:03 - INFO     - | ModuleNotFoundError: No module named 'ament_package'
2024-11-30 06:57:03 - INFO     - | CMake Error at ament_cmake_package_templates-extras.cmake:41 (message):
2024-11-30 06:57:03 - INFO     - |
2024-11-30 06:57:03 - INFO     - |   execute_process(/yocto/image/build/tmp/work/x86_64-linux/ament-cmake-core-native/1.3.10-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3
2024-11-30 06:57:03 - INFO     - |   /yocto/image/build/tmp/work/x86_64-linux/ament-cmake-core-native/1.3.10-1-r0/git/cmake/package_templates/templates_2_cmake.py
2024-11-30 06:57:03 - INFO     - |   /yocto/image/build/tmp/work/x86_64-linux/ament-cmake-core-native/1.3.10-1-r0/build/ament_cmake_package_templates/templates.cmake)
2024-11-30 06:57:03 - INFO     - |   returned error code 1
2024-11-30 06:57:03 - INFO     - | Call Stack (most recent call first):
2024-11-30 06:57:03 - INFO     - |   CMakeLists.txt:19 (include)
2024-11-30 06:57:03 - INFO     - |
2024-11-30 06:57:03 - INFO     - |
2024-11-30 06:57:03 - INFO     - | -- Configuring incomplete, errors occurred!
[... cut ...]
2024-11-30 06:57:03 - INFO     - | WARNING: exit code 1 from a shell command.
2024-11-30 06:57:03 - INFO     - NOTE: recipe ament-cmake-core-native-1.3.10-1-r0: task do_configure: Failed
2024-11-30 06:57:03 - ERROR    - ERROR: Task (virtual:native:/yocto/image/build/../layers/meta-ros/meta-ros2-humble/generated-recipes/ament-cmake/ament-cmake-core_1.3.10-1.bb:do_configure) failed with exit code '1'

**Expected behavior**
Last clean build that succeeded was:
2024-11-23 02:10:40 - INFO     - Repository ros checked out to 201ad61fabaa09517f64a174e7fa5cc0ec9c162e

jiaxshi, 2024-12-03:
Hi 
I encoutered the same issue, you can refer to my fix #1269 .

grischi, 2024-12-03:
seems to be related to #1266

closed 2025-09-04 by robwoolley