ros/meta-ros #1271: scarthgap-next: humble: ompl 1.6.0 not installing

graceagrace, 2024-12-09:
Hi there,

The latest version of OMPL for ROS2 Humble is failing at the `do_install()` step on `scarthgap-next`. Logfile is attached as a separate file, as it's quite verbose.
[scarthgap-next-humble-ompl-1.6.0-do-install.txt](https://github.com/user-attachments/files/18064676/scarthgap-next-humble-ompl-1.6.0-do-install.txt)

graceagrace, 2024-12-10:
I was initially not sure if the problem mentioned in #1237, but it turns out it is. The fix recommended by the submitter of that issue (commenting out https://github.com/ros/meta-ros/blob/1f17f925d06cbc0a1b6cd22f9fb00b31d9001e18/meta-ros2-humble/recipes-bbappends/ompl/ompl_%25.bbappend#L26C1-L28C2) is the solution, as well.

@mark-bchoi, you mentioned in the linked issue that these lines should not be present in the .bbappend file, but they are there on the `scarthgap` and `scarthgap-next` branches. Is a PR necessary to fix this?

robwoolley, 2025-09-04:
Confirming that the merged patch fixes the issue.  ompl builds successfully with scarthgap and humble.

closed 2025-09-04 by robwoolley