ros/meta-ros #1279: pybind11-vendor wants to fetch code during the compile stage

vermaete, 2025-01-14:
On scarthgap and ros2-jazzy the recipe pybind11-vendor wants to fetch code during the compile stage.
What should be during configuration.

I see some examples how to solve this in the CMakefile and bbappend.
But for now my solution is not yet working.
But is there already a solution available?

jiaxshi, 2025-01-17:
Hi
If you encounter an ethernet issue, you can try with
`do_configure[network] = "1"`

vermaete, 2025-01-19:
I assume the issue is here: https://github.com/ros2/pybind11_vendor/blob/rolling/CMakeLists.txt#L9
@jiaxshi : Yocto should build without having network access (except fetching step).  For those poor people with air-gapped networks at work :-(

vermaete, 2025-01-22:
@mark-bchoi I hope I have collected all info you need.

I have changed the CMakeLists.txt file after doing a `devtool modify pybind11-vendor`
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 69666b1..e724fbb 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -4,7 +4,7 @@ project(pybind11_vendor CXX)
 find_package(ament_cmake REQUIRED)
 find_package(ament_cmake_vendor_package REQUIRED)

-find_package(pybind11 QUIET)
+find_package(pybind11 REQUIRED)

 ament_vendor(pybind11_vendor
   SATISFIED ${pybind11_FOUND}
```

The log of the 'bitbake pybdind11-vendor`:
```
ERROR: pybind11-vendor-3.1.2-2-r0 do_configure: ExecutionError('/var/tmp/user/x/yocto/build/tmp/work/cortexa72-cortexa53-xilinx-linux/pybind11-vendor/3.1.2-2/temp/run.do_configure.3950729', 1, None, None)
ERROR: Logfile of failure stored in: /var/tmp/user/x/yocto/build/tmp/work/cortexa72-cortexa53-xilinx-linux/pybind11-vendor/3.1.2-2/temp/log.do_configure.3950729
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/var/tmp/user/x/yocto/sources/poky/../meta-xilinx/meta-xilinx-core/recipes-devtools/python/python3-pybind11_2.13.1.bb:do_populate_sysroot', '/var/tmp/user/x/yocto/sources/poky/meta/recipes-core/glibc/glibc_2.39.bb:do_populate_sysroot', '/var/tmp/user/x/yocto/sources/poky/meta/recipes-devtools/cmake/cmake-native_3.28.3.bb:do_populate_sysroot', '/var/tmp/user/x/yocto/sources/poky/meta/recipes-devtools/gcc/gcc-cross_13.3.bb:do_populate_sysroot', '/var/tmp/user/x/yocto/sources/poky/meta/recipes-devtools/gcc/gcc-runtime_13.3.bb:do_populate_sysroot', 'virtual:native:/var/tmp/user/x/yocto/sources/poky/../meta-ros/meta-ros2-jazzy/generated-recipes/ament-cmake/ament-cmake-vendor-package_2.5.2-1.bb:do_populate_sysroot', 'virtual:native:/var/tmp/user/x/yocto/sources/poky/../meta-ros/meta-ros2-jazzy/generated-recipes/ament-cmake/ament-cmake_2.5.2-1.bb:do_populate_sysroot', 'virtual:native:/var/tmp/user/x/yocto/sources/poky/meta/recipes-devtools/ninja/ninja_1.11.1.bb:do_populate_sysroot', 'virtual:native:/var/tmp/user/x/yocto/sources/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/var/tmp/user/x/yocto/sources/poky/meta/recipes-devtools/python/python3_3.12.6.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['python3-pybind11', 'openssl', 'expat', 'gettext-minimal-native', 'glibc', 'libxcrypt', 'ncurses', 'util-linux-libuuid', 'zlib', 'binutils-cross-aarch64', 'cmake-native', 'gcc-cross-aarch64', 'gcc-runtime', 'libgcc', 'libedit', 'libtool-native', 'm4-native', 'opkg-utils', 'python3', 'bzip2', 'libnsl2', 'libtirpc', 'texinfo-dummy-native', 'xz', 'linux-libc-headers', 'gdbm', 'libffi', 'sqlite3', 'python3-catkin-pkg-native', 'python3-vcstool-native', 'libyaml-native', 'ament-cmake-core-native', 'ament-cmake-export-definitions-native', 'ament-cmake-export-dependencies-native', 'ament-cmake-export-include-directories-native', 'ament-cmake-export-interfaces-native', 'ament-cmake-export-libraries-native', 'ament-cmake-export-link-flags-native', 'ament-cmake-export-targets-native', 'ament-cmake-gen-version-h-native', 'ament-cmake-include-directories-native', 'ament-cmake-libraries-native', 'ament-cmake-python-native', 'ament-cmake-target-dependencies-native', 'ament-cmake-test-native', 'ament-cmake-vendor-package-native', 'ament-cmake-version-native', 'ament-cmake-native', 'ament-package-native', 'openssl-native', 'expat-native', 'ncurses-native', 'util-linux-libuuid-native', 'zlib-native', 'flex-native', 'gnu-config-native', 'libedit-native', 'make-native', 'ninja-native', 'perl-native', 'pseudo-native', 'python3-build-native', 'python3-cython-native', 'python3-flit-core-native', 'python3-importlib-metadata-native', 'python3-installer-native', 'python3-more-itertools-native', 'python3-packaging-native', 'python3-pathlib2-native', 'python3-pip-native', 'python3-pyparsing-native', 'python3-pyproject-hooks-native', 'python3-pyyaml-native', 'python3-setuptools-scm-native', 'python3-setuptools-native', 'python3-six-native', 'python3-toml-native', 'python3-tomli-native', 'python3-typing-extensions-native', 'python3-wheel-native', 'python3-zipp-native', 'python3-native', 'bzip2-native', 'libnsl2-native', 'libtirpc-native', 'unzip-native', 'xz-native', 'zstd-native', 'attr-native', 'gdbm-native', 'gmp-native', 'libffi-native', 'libmpc-native', 'mpfr-native', 're2c-native', 'sqlite3-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing python function externalsrc_configure_prefunc
| DEBUG: Python function externalsrc_configure_prefunc finished
| DEBUG: Executing python function fetcher_hashes_dummyfunc
| DEBUG: Python function fetcher_hashes_dummyfunc finished
| DEBUG: Executing shell function do_configure
| -- The CXX compiler identification is GNU 13.3.0
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /var/tmp/user/x/yocto/build/tmp/work/cortexa72-cortexa53-xilinx-linux/pybind11-vendor/3.1.2-2/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Found ament_cmake: 2.5.2 (/var/tmp/user/x/yocto/build/tmp/work/cortexa72-cortexa53-xilinx-linux/pybind11-vendor/3.1.2-2/recipe-sysroot-native/opt/ros/jazzy/share/ament_cmake/cmake)
[... cut ...]
  /var/tmp/user/x/yocto/sources/poky/../meta-ros/meta-ros2-jazzy/generated-recipes/pybind11-vendor/pybind11-vendor_3.1.2-2.bb:do_configure

```

Just in case you need this:

```
user@server:/var/tmp/user/x/yocto$ find . -name pybind11Config.cmake
./build/tmp/work/cortexa72-cortexa53-xilinx-linux/pybind11-vendor/3.1.2-2/recipe-sysroot/usr/lib/python3.12/site-packages/pybind11/share/cmake/pybind11/pybind11Config.cmake
./build/tmp/sysroots-components/cortexa72-cortexa53/python3-pybind11/usr/lib/python3.12/site-packages/pybind11/share/cmake/pybind11/pybind11Config.cmake
```

closed 2025-01-23 by vermaete