ros/meta-ros #1280: {humble} backward_ros config causes build failures in other packages in scarthgap

robwoolley, 2025-01-20:
**Describe the bug**

**To Reproduce**
Using the kas configuration for:
* ROS Distro: ROS 2 Humble
* Yocto Release: Yocto Scarthgap

Running the command "bitbake clearpath-platform" causes the following build failure.

It is due to backward_ros containing an absolute path for the target that is checked with CMake set_and_check.  However, this path does not exist on the build machine so it fails and the configure stage for the controllers fail.

```
controller-manager
joint-trajectory-controller
diff-drive-controller
joint-state-broadcaster
```

The tail of the build log is as follows:
```
| -- The CXX compiler identification is GNU 13.3.0
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-g++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Found ament_cmake: 1.3.10 (/ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/recipe-sysroot/opt/ros/humble/share/ament_cmake/cmake)
| -- Found Python3: /ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/recipe-sysroot-native/usr/bin/python3-native/python3 (found version "3.12.6") found components: Interpreter
| CMake Error at /ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/recipe-sysroot/opt/ros/humble/share/backward_ros/cmake/backward_rosConfig.cmake:11 (message):
|   File or directory /opt/ros/humble/include referenced by variable
|   backward_ros_INCLUDE_DIRS does not exist !
| Call Stack (most recent call first):
|   /ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/recipe-sysroot/opt/ros/humble/share/backward_ros/cmake/backward_rosConfig.cmake:26 (set_and_check)
|   CMakeLists.txt:21 (find_package)
|
|
| -- Configuring incomplete, errors occurred!
| WARNING: /ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/temp/run.do_configure.1464830:172 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/git -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=sbin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DPYTHON_EXECUTABLE:PATH=/ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/recipe-sysroot-native/usr/bin/python3-native/python3 -DPython_EXECUTABLE:PATH=/ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/recipe-sysroot-native/usr/bin/python3-native/python3 -DPython3_EXECUTABLE:PATH=/ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/recipe-sysroot-native/usr/bin/python3-native/python3 -DLIB_SUFFIX= -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE:FILEPATH=/ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DFETCHCONTENT_FULLY_DISCONNECTED=ON -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=ON -DCMAKE_PREFIX_PATH='/ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/recipe-sysroot/opt/ros/humble;/ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/recipe-sysroot/usr' -DCMAKE_INSTALL_PREFIX:PATH='/opt/ros/humble' -DBUILD_TESTING=OFF -DPYTHON_SOABI=cpython-312-aarch64-linux-gnu -DAMENT_CMAKE_ENVIRONMENT_PARENT_PREFIX_PATH_GENERATION=OFF -Wno-dev'
| WARNING: Backtrace (BB generated script):
|       #1: cmake_do_configure, /ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/temp/run.do_configure.1464830, line 172
|       #2: do_configure, /ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/temp/run.do_configure.1464830, line 151
|       #3: main, /ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/work/cortexa72-oe-linux/joint-state-broadcaster/2.37.3-1/temp/run.do_configure.1464830, line 195
ERROR: Task (/ala-lpggp31/rwoolley/2025-01-16/build/../layers/meta-ros/meta-ros2-humble/generated-recipes/ros2-controllers/joint-state-broadcaster_2.37.3-1.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7096 tasks of which 7092 didn't need to be rerun and 4 failed.
NOTE: The errors for this build are stored in /ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/log/error-report/error_report_20250120144558.txt
You can send the errors to a reports server by running:
  send-error-report /ala-lpggp31/rwoolley/2025-01-16/build/tmp-glibc/log/error-report/error_report_20250120144558.txt [-s server]
NOTE: The contents of these logs will be posted in public if you use the above command with the default server. Please ensure you remove any identifying or proprietary information when prompted before sending.

Summary: 4 tasks failed:
  /ala-lpggp31/rwoolley/2025-01-16/build/../layers/meta-ros/meta-ros2-humble/generated-recipes/ros2-control/controller-manager_2.43.1-1.bb:do_configure
  /ala-lpggp31/rwoolley/2025-01-16/build/../layers/meta-ros/meta-ros2-humble/generated-recipes/ros2-controllers/joint-trajectory-controller_2.37.3-1.bb:do_configure
  /ala-lpggp31/rwoolley/2025-01-16/build/../layers/meta-ros/meta-ros2-humble/generated-recipes/ros2-controllers/diff-drive-controller_2.37.3-1.bb:do_configure
  /ala-lpggp31/rwoolley/2025-01-16/build/../layers/meta-ros/meta-ros2-humble/generated-recipes/ros2-controllers/joint-state-broadcaster_2.37.3-1.bb:do_configure
Summary: There was 1 WARNING message.
Summary: There were 4 ERROR messages, returning a non-zero exit code.
```

**Expected behavior**
The controllers should find backward_ros in the target recipe sysroot and build successfully.

graceagrace, 2025-01-23:
Hi @robwoolley, I actually encountered this same issue this morning. Patching in the commit on [this](https://github.com/pal-robotics/backward_ros/pull/23) PR on the backward_ros development repository resolved it for me.

sgstreet, 2025-02-06:
On rolling I have a commit like this:
'''
diff --git a/meta-ros2-rolling/recipes-bbappends/backward-ros/backward-ros_1.0.6-1.bbappend b/meta-ros2-rolling/recipes-bbappends/backward-ros/backward-ros_1.0.6-1.bbappend
index 7d47626a8..6f784d29f 100644
--- a/meta-ros2-rolling/recipes-bbappends/backward-ros/backward-ros_1.0.6-1.bbappend
+++ b/meta-ros2-rolling/recipes-bbappends/backward-ros/backward-ros_1.0.6-1.bbappend
@@ -3,3 +3,7 @@
 DEPENDS += "ament-cmake-native"
 
 BBCLASSEXTEND = "native nativesdk"
+
+FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"
+
+SRC_URI += "file://0001-Backport-make-backward_rosConfig-cmake-relocatable.patch"
'''

using a patch like this:
```
From e38d40f551eb668b527676fd3042e6fa3b555746 Mon Sep 17 00:00:00 2001
From: Stephen Street <stephen@redrocketcomputing.com>
Date: Tue, 4 Feb 2025 17:24:31 -0800
Subject: [PATCH] Backport make backward_rosConfig cmake relocatable

See https://github.com/pal-robotics/backward_ros/pull/23 for additional
information.

Upstream-Status: Pending

---
 backward_rosConfig.cmake.in | 10 ++++++----
 1 file changed, 6 insertions(+), 4 deletions(-)

diff --git a/backward_rosConfig.cmake.in b/backward_rosConfig.cmake.in
index 6713e91..bd9397a 100644
--- a/backward_rosConfig.cmake.in
+++ b/backward_rosConfig.cmake.in
@@ -1,9 +1,11 @@
 @PACKAGE_INIT@
-set_and_check(backward_ros_INCLUDE_DIRS "@BACKWARD_ROS_INSTALL_PREFIX@/include")
+set_and_check(backward_ros_INCLUDE_DIRS "@PACKAGE_BACKWARD_ROS_INSTALL_PREFIX@/include")
 if(WIN32)
-set_and_check(backward_ros_LIBRARIES "@BACKWARD_ROS_INSTALL_PREFIX@/lib/backward.lib")
+set_and_check(backward_ros_LIBRARIES "@PACKAGE_BACKWARD_ROS_INSTALL_PREFIX@/lib/backward.lib")
+elseif(APPLE)
+set_and_check(backward_ros_LIBRARIES "@PACKAGE_BACKWARD_ROS_INSTALL_PREFIX@/lib/libbackward.dylib")
 else()
-set_and_check(backward_ros_LIBRARIES "@BACKWARD_ROS_INSTALL_PREFIX@/lib/libbackward.so")
+set_and_check(backward_ros_LIBRARIES "@PACKAGE_BACKWARD_ROS_INSTALL_PREFIX@/lib/libbackward.so")
 endif()
 check_required_components(backward_ros)
-include(@BACKWARD_ROS_INSTALL_PREFIX@/share/@PROJECT_NAME@/cmake/BackwardConfigAment.cmake)
\ No newline at end of file
+include(@PACKAGE_BACKWARD_ROS_INSTALL_PREFIX@/share/@PROJECT_NAME@/cmake/BackwardConfigAment.cmake)
```

robwoolley, 2025-09-04:
Confirming that the merged fix resolved the issue.  Successfully built"
controller-manager
joint-trajectory-controller
diff-drive-controller
joint-state-broadcaster

on scarthgap + humble.

closed 2025-09-04 by robwoolley