ros/meta-ros #1304: Wiki -> OpenEmbedded: link to Discord is wrong

vermaete, 2025-02-09:
I did not found a way to send a PR
The link on the Home.md is correct. 

diff --git a/OpenEmbedded-Build-Instructions.md b/OpenEmbedded-Build-Instructions.md
index 47b5086..5c1feca 100644
--- a/OpenEmbedded-Build-Instructions.md
+++ b/OpenEmbedded-Build-Instructions.md
@@ -4,7 +4,7 @@
 >
 > For simplified instructions on how to get started please refer to these build instructions: https://github.com/ros/meta-ros/blob/build/kas/README.md
 >
-> To chat about meta-ros, come find us on the OSRF Discord [#cwg-openembedded channel](https://discord.com/channels/1077825543698927656/1202522247978557510).
+> To chat about meta-ros, come find us on the OSRF Discord [#cwg-openembedded channel](https://discourse.ros.org/c/openembedded).
 >
 > You are also welcome to join our biweekly meetings. Catch up on our meeting minutes here: [ROS OpenEmbedded Working Group Minutes - Google Docs](https://docs.google.com/document/d/1LqUjcu6vdlqVJO62SreCyjzddNDZhfO2n-7qYghY_cQ/edit)

robwoolley, 2025-02-10:
Hi @vermaete,

We use both Discord and Discourse.  The link you pointed out is for our Discord channel for anyone who wishes to chat in "real-time".

Regards,
Rob

vermaete, 2025-02-20:
sorry, my mistake

closed 2025-02-20 by vermaete