ros/meta-ros #1315: canopen-tests: compilation failed

vermaete, 2025-02-25:
**Describe the bug**
Could it be that the recipe canopen-tests if failing in the compilation step on Master(Yocto) and Rolling(Ros)?

**To Reproduce**

```
kas shell -k kas/oeros-master-rolling-raspberrypi4-64.yml
bitbake canopen-tests --continue
```

```
Build Configuration:
BB_VERSION           = "2.9.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "debian-12"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "raspberrypi4-64"
DISTRO               = "ros2"
DISTRO_VERSION       = "2.rolling.2025-01-17.3-0"
TUNE_FEATURES        = "aarch64 crc cortexa72"
TARGET_FPU           = ""
DISTRO_NAME          = "Robot Operating System 2 (ROS 2) with OpenEmbedded Linux"
ROS_DISTRO           = "rolling"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"
meta-filesystems     
meta-gnome           
meta-initramfs       
meta-multimedia      
meta-networking      
meta-oe              
meta-perl            
meta-python          
meta-webserver       
meta-xfce            = "master:8a29c46188b31851b32414d9d3dbb38b1b48a073"
meta                 = "master:6aab808bb9adee8e2409c23f49c6bc021007f1d8"
meta-raspberrypi     = "master:7494a378188f5d6330e038a5d6b9372b4aa232dd"
meta-ros-common      
meta-ros2            
meta-ros2-rolling    = "vermaete/license:c312eda71029b2852620d8739a87248c221685d8"
```

```
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: VERBOSE=1 cmake --build /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build --target all --
| Change Dir: '/home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build'
| 
| Run Build Command(s): ninja -v -j 4 all
| [1/20] cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build && rm -rf /opt/ros/rolling/share/canopen_tests/config/simple/* && rm -rf /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple/* && mkdir -p /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple
| [2/20] cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build && rm -rf /opt/ros/rolling/share/canopen_tests/config/canopen_system/* && rm -rf /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/canopen_system/* && mkdir -p /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/canopen_system
| [3/20] cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build && rm -rf /opt/ros/rolling/share/canopen_tests/config/cia402_namespaced_system/* && rm -rf /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/cia402_namespaced_system/* && mkdir -p /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/cia402_namespaced_system
| [4/20] cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build && rm -rf /opt/ros/rolling/share/canopen_tests/config/cia402_system/* && rm -rf /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/cia402_system/* && mkdir -p /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/cia402_system
| [5/20] cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build && rm -rf /opt/ros/rolling/share/canopen_tests/config/cia402/* && rm -rf /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/cia402/* && mkdir -p /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/cia402
| [6/20] cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build && rm -rf /opt/ros/rolling/share/canopen_tests/config/cia402_lifecycle/* && rm -rf /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/cia402_lifecycle/* && mkdir -p /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/cia402_lifecycle
| [7/20] cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build && rm -rf /opt/ros/rolling/share/canopen_tests/config/cia402_diagnostics/* && rm -rf /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/cia402_diagnostics/* && mkdir -p /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/cia402_diagnostics
| [8/20] cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build && rm -rf /opt/ros/rolling/share/canopen_tests/config/simple_lifecycle/* && rm -rf /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple_lifecycle/* && mkdir -p /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple_lifecycle
| [9/20] cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build && rm -rf /opt/ros/rolling/share/canopen_tests/config/simple_diagnostics/* && rm -rf /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple_diagnostics/* && mkdir -p /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple_diagnostics
| [10/20] cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build && rm -rf /opt/ros/rolling/share/canopen_tests/config/robot_control/* && rm -rf /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/robot_control/* && mkdir -p /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/robot_control
| [11/20] cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/git/config/simple && sed 's|@BUS_CONFIG_PATH@|/opt/ros/rolling/share/canopen_tests/config/simple|g' /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/git/config/simple/bus.yml > /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple/bus.yml && dcfgen -v -d /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple/ -rS /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple/bus.yml
| FAILED: CMakeFiles/simple.util
| cd /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/git/config/simple && sed 's|@BUS_CONFIG_PATH@|/opt/ros/rolling/share/canopen_tests/config/simple|g' /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/git/config/simple/bus.yml > /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple/bus.yml && dcfgen -v -d /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple/ -rS /home/jan/projects/meta-ros/build/tmp/work/cortexa72-oe-linux/canopen-tests/0.3.0-1/build/config/simple/bus.yml
| /bin/sh: 1: dcfgen: not found
```

So `defgen` is not found.  I assume this should be in some -native package?

Br

vermaete, 2025-02-27:
and same for cogen.  Seems to be also missing in the -native format.

vermaete, 2025-03-02:
@robwoolley It seems that 'cogen' is now found when running 'canopen-tests' with the current status of master.  But dcfgen not yet.
It looks like the package is rather empty.  Only some metadata of Python.  But the Python files itself are missing.

robwoolley, 2025-03-03:
@vermaete The upstream ROS builds lely_core inside the vendored package lely_core_libraries.   I opt'd to create a separate recipe for lely_core as it uses autoconf as its build system, where lely_core_libraries is a ROS package that uses ament/cmake and Python setup.py.  This also made it easier to build lely-core-native so the tools could be used at build-time.

I haven't tried using canopen at runtime.   lely-core-libraries should automatically pull in the lely-core package which provides /usr/bin/dcfgen.   If that is not the case or anything fails to work at runtime, please let me know what is broken and I'll try to fix it.

Thanks for the feedback, I appreciate that you are using it.  Glad to help anyway I can!

vermaete, 2025-03-09:
Some notes on my attempts to have canopen-tests working on RPi4-64 over vcan.

robwoolley, 2025-03-16:
Sorry for the delay, I was off last week.  I took a quick look at fixing it and came up with this: https://github.com/robwoolley/meta-ros/tree/fix-canopen-tests

I found a missing patch that resolved a problem with the dcfgen wheel not being installed.  I also had to remove some buildpaths in new Python bytecode in lely-core and make sure that canopen-tests was using the right binaries for a cross-platform build.

I will need to split the commits and write better commit messages before merging, but it should help unblock you.   I hope it works for you.  Let me know how it goes.