ros/meta-ros #1328: rosbag2-py humble compiled for native

chamois94, 2025-04-04:
**Describe the bug**
rosbag2-py is compiled and installed on rootfs with host librairies instead of target

**To Reproduce*
I created my own custom image with rosbag2 and so rosbag2-py.
All python librairies are generated for target and correctly installed except for rosbag2-py which are generated for host instead of target. 
Consequently ros bag can't find options (record, ..) and I can not use ros bag on thé target.

Any idea ?

chamois94, 2025-04-05:
Found that I have to create a .bbappend with inherit python3targetconfig

Bentis, 2025-04-09:
Encountered same issue, verified fix.
Suggest you make a PR and add it to [rosbag2-py_0.15.13-1.bbappend](https://github.com/ros/meta-ros/blob/master/meta-ros2-humble/recipes-bbappends/rosbag2/rosbag2-py_0.15.13-1.bbappend)