ros/meta-ros #1365: Inquiry about MoveIt support roadmap and dependency customization possibilities

quic-weijshen, 2025-04-23:
Hi meta-ros maintainers,

I noticed that MoveIt is currently listed in the blacklist and excluded from meta-ros testing infrastructure. Could you share:

- Whether there are plans to support MoveIt in future releases?

Additionally, I observed that core components like moveit-planners-ompl primarily require libomp without hard dependencies on qt5, opengl, or x11. For embedded/headless use cases where GUI dependencies are undesirable:

- Is there a recommended method to enable a minimal subset of MoveIt functionality while excluding non-essential dependencies?
- Would contributions towards modularizing the build (e.g., introducing PACKAGE_CONFIG flags for GUI vs core components) be welcomed?

I'm willing to collaborate on potential solutions if this aligns with the project's direction. Thank you for your guidance!

robwoolley, 2025-04-24:
Thanks for asking the questions.

> Whether there are plans to support MoveIt in future releases?

Supporting MoveIt is something I have looked into.  Back in March I tried building many of the metapackages.  This is what I reported in the minutes:

> Meta-packages
> - Search index.ros.org for “meta” or “metapackages” for a sample list
> - **Success**: ros-gz, gps-umd, geometry-tutorials, point-cloud-transport-plugins, ros2-controllers, ros2-control, rosbag2-storage-default-plugins, executive-smach, geometry2, geographic-info, rqt-common-plugins. flexbe-behavior-engine, ros2cli-common-extensions, rosbag2
> - **Failures**:
>    - moveit, moveit-plugins, moveit-planners, moveit-setup-core-plugins
>       - Miscellaneous packaging issues in dependencies
>    - lanelet2
>       - lanelet2_core: Boost deprecated robust_point_type• canopen
>    - fuse
>       - fuse-core Boost failure
>    - mola, mola-state-estimation, mrpt-sensors, mrpt-libbase,
>    - mrpt-libbase: use bitbake fetcher for dependencies
>    - navigation2 doesn’t generate recipes for rolling - TBD

I believe that it was a simple do_package_qa error related to buildpaths.  If you could help provide the fixes, I would be happy to support it.

Most of the buildpath items stem from ROS including a dependency by filename instead of by target (https://cmake.org/cmake/help/latest/manual/cmake-buildsystem.7.html#target-usage-requirements).  This causes the build paths to creep into the generated CMake Target files.   There are also cases where the compiled Python files (.pyc) or the QMake project files include buildpaths.  This requires a more complicated fix and if you run into it, I'm happy to point you to examples.

> Is there a recommended method to enable a minimal subset of MoveIt functionality while excluding non-essential dependencies?

Usually the PACKAGECONFIG variable (https://docs.yoctoproject.org/ref-manual/variables.html#term-PACKAGECONFIG) combined with checking for whether the DISTRO_FEATURE is the way to go.  Here is an example of a recipe that conditionally enabled a feature based on whether opengl was enabled: https://github.com/ros/meta-ros/blob/3b97e3bff051dfa4d536767c5a9842598689aaf0/meta-ros2-jazzy/recipes-bbappends/libg2o/libg2o_2020.5.29-6.bbappend#L5

> Would contributions towards modularizing the build (e.g., introducing PACKAGE_CONFIG flags for GUI vs core components) be welcomed?

I think that sounds great.  I think one of the challenges will be deciding how to set the default PACKAGECONFIGS in a way that builds the package in a way that resembles what users are expecting but also giving advanced users the ability to disable unnecessary features for minimizing footprint on a production embedded device.

quic-weijshen, 2025-04-25:
@robwoolley Thank you for the detailed explanation and actionable guidance! 
Based on your suggestions, I will proceed as follows:
1. Prioritize enabling a ​​minimal MoveIt configuration​​ and validate locally with fixes​​ and ​​dependency trimming ​​(e.g. qt/x11 exclusion).
2. Share verification results and submit a draft PR once core functionality passes QA.

Thanks!

closed 2025-04-26 by robwoolley